gem5/gem5 #943: gem5::statistics::Counter are 52-bit not 64-bit

yihuaeyang, 2024-03-18:
https://github.com/gem5/gem5/blob/e8bc4fc137a5a7e82b601432271a027b652ae69b/src/base/stats/types.hh#L45

As `double`, don't the counters really have only 52-bit precision?

andysan, 2024-03-18:
This is something that dates back to ~2004 when the statistic package was originally implemented by Nathan Binkert. Your observation is sort of correct. You won't get the full 64b precision for integer counts, but it is technically a 64b value.

You have highlighted an interesting and potentially important issue. Many stats are counters of integer events. Floating point semantics can be a bit weird for these sorts of things. One such problem is that stats will stop counting single events around 2^53 (I can't remember if this includes denormals or not). This could be a very real problem for high-frequency events in long-running simulations which can execute for 2^64 ticks. Another weird problem occurs if you subtract two large counters where the difference quickly ends up losing precision, this can end up being a problem when calculating averages.

yihuaeyang, 2024-03-18:
Thanks for your quick response. I agree with what you said. I have 2 further questions/comments:

1. Is there "advantage" to use `double` for counter? Perhaps it makes it easier to calculate things like averages and ratios?
2. Does the risk of having a limit of 2^53 events worth the effort of refactoring the counter from `double` to `int64_t`?

andysan, 2024-03-18:
I suspect the reason we use double everywhere might be that some derived counters (e.g., IPC) aren't integers. Having a separate type for event counters and derived stats would be complicated. Fixing that seems like _a lot_ of work. I suspect it isn't worth doing on its own, but it might be worth doing as a bigger stats overhaul at some point.