gem5/gem5 #960: Docker Image for Ubuntu 20.04 doesn't exist

Shreesh-Kulkarni, 2024-03-24:
The docker image for Ubuntu 20.04 doesn't exist. It redirects to 404 Page not found, and while pulling it via docker it says it doesn't exist.

Affects Ubuntu 20.04

Steps to reproduce the behavior. Please assume starting from a clean repository:

1. docker pull "ubuntu 20.04 image"
2. Error: image not found

ivanaamit, 2024-03-28:
Hi @Shreesh-Kulkarni, thank you for reporting this bug.

Could you please provide us with a link to the documentation that you are referring to? Also, could you share the specific commands that you used? Thank you.

EmilyG2413, 2024-04-26:
Hello! I was looking into this issue a bit more and I believe it is this [page](https://www.gem5.org/documentation/general_docs/building): https://www.gem5.org/documentation/general_docs/building
For this section: 
[image]

I might be mistaken but hope this helps!

closed 2024-05-26 by github-actions[bot]

ivanaamit, 2024-05-28:
@Shreesh-Kulkarni @EmilyG2413 This issue has been solved. There was a typo in the URL for 20.04.

Shreesh-Kulkarni, 2024-05-29:
@ivanaamit thank you. It works now.