gem5/gem5 #981: c.jr/c.jalr semantic wrong

Ye-Jinhong, 2024-04-01:
**Describe the bug**
As RISC-V unpriviledge isa doc says, `c.jr/c.jalr` expands to `jalr x0/x1 0(rs1)`. Since `jalr` should force the bit 0 to **zero**, I think `c.jr/c.jalr` should act the same way.

hnpl, 2024-04-01:
Hi @Ye-Jinhong, can you elaborate the issue more? (i.e., what was the unexpected output for c.jal/c.jalr?)

hnpl, 2024-04-03:
I see. If I'm not mistaken, that's a bug.

Interestingly, gem5 O3CPU does jump to the correct location due to branch prediction mechanism [1]. This is because c_jr's and c_jalr's branchTarget() function do generate the correct address, and if Rc1 is not 2-byte aligned, it is a branch misprediction (and if the branch predictor learned this, there'll be no misprediction later on).

 I'm not sure about the c_jr and c_jalr in other gem5 cpu types given that they call the instructions's execute() function directly to get the new PC.

```
    std::unique_ptr<PCStateBase>
    C_jr::branchTarget(ThreadContext *tc) const
    {
        PCStateBase *pc_ptr = tc->pcState().clone();
        pc_ptr->as<PCState>().set((tc->getReg(srcRegIdx(0)) + imm) & ~0x1);
        return std::unique_ptr<PCStateBase>{pc_ptr};
    }
```

[1] https://github.com/gem5/gem5/blob/ffd0680a2c3bcfb850da531f005381c94e311011/src/cpu/o3/decode.cc#L717-L723

@powerjg, @aarmejach: please correct if I'm wrong, my memory is rather vague on O3CPU branch prediction.

hnpl, 2024-04-29:
@Ye-Jinhong can you make a PR that fixes this?

closed 2024-06-06 by ivanaamit