gem5/gem5 #988: sim-se: gpu-tests are failing

ivanaamit, 2024-04-02:
**Describe the bug**
The daily tests, specifically the gpu-tests, are failing. This commit https://github.com/gem5/gem5/commit/00d4b6825c697a7c2b56d86ee54dc033c3c87794 appears to be causing the test failures. It is suspected that the older version of Ubuntu is incompatible with the changes introduced in this commit.

**Affects version**
current develop branch

**To Reproduce**
Build gem5 `scons build/VEGA_X86/gem5.opt -j $(nproc)` using this docker image `ghcr.io/gem5/gcn-gpu:latest`.

**Terminal Output**
This is the error message displayed in the terminal output during the build process.
```shell
src/sim/syscall_emul.hh:744:29: error: ‘struct statx’ has no member named ‘stx_mnt_id’; did you mean ‘stx_uid’?
  744 |     tgt->stx_mnt_id = host->stx_mnt_id;
      |                       ~~~~~~^~~~~~~~~~
      |                       stx_uid
```

ivanaamit, 2024-04-02:
@nmosier, could you please take a look at this issue? The commit that was merged yesterday is causing tests to fail. We tried to fix it but didn't find the appropriate solution. Thank you.

nmosier, 2024-04-02:
Oh no! I think the problem is because the statx struct is only defined on Linux hosts since v4.11. Are the GPU tests running on older Linux hosts (pre 4.11) or a different os (e.g., macOS)?

BobbyRBruce, 2024-04-02:
> Oh no! I think the problem is because the statx struct is only defined on Linux hosts since v4.11. Are the GPU tests running on older Linux hosts (pre 4.11) or a different os (e.g., macOS)?
> 
> It could also be the members in struct statx has changed over time, and stx_mnt_id was added in a more recent version of the kernel than the one the GPU tests are running on.

I believe these tests are running on 5.14. They are running in the they are running on Ubuntu 20.04. According to this: https://ubuntu.com/about/release-cycle#ubuntu-kernel-release-cycle, should be running with Linux 5.14.

However, I think this solution/patch would be worth checking. We have a way to reproduce this error so should be able to quickly verify if this fix works.

nmosier, 2024-04-02:
I submitted a patch. Let me know if that fixes the issue!

BobbyRBruce, 2024-04-02:
I've verified that this patch appears to fix this issue. I use `docker run -u ${UID}:${GID} -v $(pwd):/gem5 -w /gem5 -it --rm ghcr.io/gem5/gcn-gpu:latest scons build/VEGA_X86/gem5.opt -j8` : on the develop branch it fails with the error outlined, but completes fine with #989.

powerjg, 2024-04-02:
Thanks for the quick fix!!

PR #989 (2024-04-02): sim-se: Fix copyOutStatxBuf compile error
Fix #988. Rewrite statxFunc and copyOutStatxBuf to use platform-agnostic stat system call, not Linux-specific statx system call.

Change-Id: I3d17e14684e9cd77cdbfd0141b93c3bcbd27dbeb
commit 32ee09df (2024-04-02): sim-se: Fix copyOutStatxBuf compile error (#989)

Fix #988. Rewrite statxFunc and copyOutStatxBuf to use platform-agnostic
stat system call, not Linux-specific statx system call.

Change-Id: I3d17e14684e9cd77cdbfd0141b93c3bcbd27dbeb