gem5/gem5 #1044: mem-ruby: Ruby functional read failures with MESI Three-Level caches

nmosier, 2024-04-20:
**Describe the bug**
When running the O3 processor under various configurations with MESI Three-Level Ruby caches, I've encountered frequent Ruby functional read errors, causing a panic [here](https://github.com/gem5/gem5/blob/e578f83739e54ed0fdbb7a68dfce688a4080760e/src/mem/ruby/system/RubyPort.cc#L431).

**Affects version**
develop @ [e578f83](https://github.com/gem5/gem5/commit/e578f83739e54ed0fdbb7a68dfce688a4080760e)

**gem5 Modifications**
None

**To Reproduce**
1. Create a build options file for X86 MESI Three-Level: `printf 'RUBY=y\nNUMBER_OF_BITS_PER_SET=128\nRUBY_PROTOCOL_MESI_THREE_LEVEL=y\nBUILD_ISA=y\nUSE_X86_ISA=y\n' > build_opts/X86_MESI_Three_Level`
1. Compile gem5: `scons build/X86_MESI_Three_Level/gem5.opt`
2. Compile a simple program: `echo 'int main() {}' > test.c && cc test.c -o test`
3. Run two copies of the program on O3 under SE with SMT: `./build/X86_MESI_Three_Level/gem5.opt configs/deprecated/example/se.py --cpu-type=X86O3CPU --smt --ruby -c './test;./test'`

**Terminal Output**
```
$ ./build/X86_MESI_Three_Level/gem5.opt configs/deprecated/example/se.py --cpu-type=X86O3CPU --ruby --smt -c './test;./test' --interp-dir=/usr
gem5 Simulator System.  https://www.gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version DEVELOP-FOR-24.0
gem5 compiled Apr 19 2024 23:18:59
gem5 started Apr 20 2024 15:06:02
gem5 executing on cafe-cet, pid 1286658
command line: ./build/X86_MESI_Three_Level/gem5.opt configs/deprecated/example/se.py --cpu-type=X86O3CPU --ruby --smt -c './test;./test' --interp-dir=/usr

warn: The se.py script is deprecated. It will be removed in future releases of  gem5.

Global frequency set at 1000000000000 ticks per second
warn: No dot file generated. Please install pydot to generate the dot file and pdf.
src/mem/dram_interface.cc:690: warn: DRAM device capacity (8192 Mbytes) does not match the address range assigned (512 Mbytes)
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
system.remote_gdb: Listening for connections on port 7000
src/sim/power_state.cc:105: warn: PowerState: Already in the requested power state, request ignored
**** REAL SIMULATION ****
src/sim/simulate.cc:199: info: Entering event queue @ 0.  Starting simulation...
src/mem/ruby/system/Sequencer.cc:680: warn: Replacement policy updates recently became the responsibility of SLICC state machines. Make sure to setMRU() near callbacks in .sm files!
src/sim/mem_state.cc:448: info: Increasing stack size by one page.
src/sim/mem_state.cc:448: info: Increasing stack size by one page.
src/sim/syscall_emul.cc:74: warn: ignoring syscall mprotect(...)
src/sim/syscall_emul.cc:74: warn: ignoring syscall mprotect(...)
src/sim/syscall_emul.cc:74: warn: ignoring syscall set_robust_list(...)
src/sim/syscall_emul.cc:74: warn: ignoring syscall set_robust_list(...)
src/sim/syscall_emul.cc:74: warn: ignoring syscall rseq(...)
src/sim/syscall_emul.cc:74: warn: ignoring syscall rseq(...)
src/sim/syscall_emul.cc:74: warn: ignoring syscall mprotect(...)
src/sim/syscall_emul.cc:74: warn: ignoring syscall mprotect(...)
src/sim/syscall_emul.cc:74: warn: ignoring syscall mprotect(...)
src/sim/syscall_emul.cc:74: warn: ignoring syscall mprotect(...)
src/sim/syscall_emul.cc:74: warn: ignoring syscall mprotect(...)
src/sim/syscall_emul.cc:74: warn: ignoring syscall mprotect(...)
src/mem/ruby/system/RubyPort.cc:431: fatal: Ruby functional read failed for address 0x40a10
Memory Usage: 791320 KBytes
```

**Expected behavior**
The two copies of `test` should exit cleanly.

**Host Operating System**
Ubuntu 22.04

**Host ISA**
X86

**Compiler used**
GCC 11.4.0

**Additional information**
I will submit a patch shortly to fix this

nmosier, 2024-04-23:
Some further relevant details.

The functional read fails for 0x40a10, which belongs the cache line 0x40a00:
```
src/mem/ruby/system/RubyPort.cc:431: fatal: Ruby functional read failed for address 0x40a10
```

To figure out where 0x40a00 might be in the cache hierarchy, I dumped Ruby debug info as follows:
```sh
$ ./build/X86_MESI_Three_Level/gem5.opt --debug-flag=Ruby --debug-file=dbgout configs/deprecated/example/se.py --cpu-type=X86O3CPU --smt --ruby -c './hello;./hello'
...
$ grep '40a00' m5out/dbgout | tail -10
226651500: system.ruby.l0_cntrl0: MESI_Three_Level-L0cache.sm:554: address: 0x40a00, destination: L1Cache-0
226651500: system.ruby.l0_cntrl0.bufferToL1: Enqueue arrival_time: 226652500, Message: [CoherenceMsg: addr = [0x40a00, line 0x40a00] Class = GETS AccessMode = Supervisor Sender = L0Cache-0 Dest = L1Cache-0 MessageSize = Control DataBlk = [ 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 ] Dirty = 0 Prefetch = No ]
226651500: system.ruby.l0_cntrl0.mandatoryQueue: Message: [RubyRequest: LineAddress = 0x40a00 PhysicalAddress = 0x40a08 Type = LD ProgramCounter = 0x7ffff7ed62ec AccessMode = Supervisor Size = 8 Prefetch = No isGLCSet = 0isSLCSet = 0]
226651500: system.ruby.l0_cntrl0.Dcache: address: 0x40a00 found
226651500: system.ruby.l0_cntrl0.Icache: No tag match for address: 0x40a00
226652500: system.ruby.l0_cntrl0.bufferToL1: Message: [CoherenceMsg: addr = [0x40a00, line 0x40a00] Class = GETS AccessMode = Supervisor Sender = L0Cache-0 Dest = L1Cache-0 MessageSize = Control DataBlk = [ 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 ] Dirty = 0 Prefetch = No ]
226652500: system.ruby.l1_cntrl0: [L1Cache_Controller 0], Time: 453305, state: MM, event: Load, addr: 0x40a00
226652500: system.ruby.l0_cntrl0.bufferFromL1: Enqueue arrival_time: 226653500, Message: [CoherenceMsg: addr = [0x40a00, line 0x40a00] Class = DATA_EXCLUSIVE AccessMode = User Sender = L1Cache-0 Dest = L0Cache-0 MessageSize = Response_Data DataBlk = [ 0xf8 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x6c 0x1b 0xf3 0xf7 0xff 0x7f 0x0 0x0 0xe0 0xa2 0x3 0xf8 0xff 0x7f 0x0 0x0 0x2 0x11 0x0 0xf8 0xff 0x7f 0x0 0x0 0x40 0x97 0xdb 0xf7 0xff 0x7f 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xd6 0xb3 0x6 0x0 0x0 0x0 0x0 0x0 0x34 0x32 0x2 0xf8 0xff 0x7f 0x0 0x0 ] Dirty = 1 Prefetch = No ]
226652500: system.ruby.dir_cntrl0.directory: Looking up address: 0x40a00
226652500: system.ruby.dir_cntrl0.directory: Looking up address: 0x40a00
```

Note the third-to-last line:
```
226652500: system.ruby.l0_cntrl0.bufferFromL1: Enqueue arrival_time: 226653500, Message: [CoherenceMsg: addr = [0x40a00, line 0x40a00] Class = DATA_EXCLUSIVE AccessMode = User Sender = L1Cache-0 Dest = L0Cache-0 MessageSize = Response_Data DataBlk = [ 0xf8 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x6c 0x1b 0xf3 0xf7 0xff 0x7f 0x0 0x0 0xe0 0xa2 0x3 0xf8 0xff 0x7f 0x0 0x0 0x2 0x11 0x0 0xf8 0xff 0x7f 0x0 0x0 0x40 0x97 0xdb 0xf7 0xff 0x7f 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xd6 0xb3 0x6 0x0 0x0 0x0 0x0 0x0 0x34 0x32 0x2 0xf8 0xff 0x7f 0x0 0x0 ] Dirty = 1 Prefetch = No ]
```
It appears that the reason the ruby functional read fails is because a DATA_EXCLUSIVE message for 0x40a00 is on the network at that moment.

powerjg, 2024-04-23:
Just to give a bit of context to this issue...

Function reads in Ruby are **hard**. In fact, there are times that it could be *impossible* to know what the correct data to read is, depending on the protocol. Functional writes are even worse! The problem is that at some point in time, it maybe not be clear what the *logical* time of the system is, as different caches have different views of the logical time at any particular point in "physical" time.

In general, rather than relying on the brittle logical reads/writes in Ruby, I would strongly encourage people to use full system mode. Most functional accesses occur because of syscall emulation. So, a "simple" way around a lot of these issues is to use FS mode.

ivanaamit, 2024-04-29:
Since #1045 has been merged, I am closing this issue. @nmosier, please reopen it if I am mistaken. Thanks.

PR #1045 (2024-04-25): mem-ruby: Fix functional reads for MESI Three-Level messages
Fix #1044. This patch adds checks for message types (PUTX_COPY, DATA, DATA_EXCLUSIVE) that contain data blocks but were missing from the original `functionalRead` method in MESI Three-Level messages.

Change-Id: I0cedc314166c9cc037bf20f5b7fef5552dd1253c
commit 66decb2e (2024-04-25): mem-ruby: Fix functional reads for MESI Three-Level messages (#1045)

Fix #1044. This patch adds checks for message types (PUTX_COPY, DATA,
DATA_EXCLUSIVE) that contain data blocks but were missing from the
original `functionalRead` method in MESI Three-Level messages.

Change-Id: I0cedc314166c9cc037bf20f5b7fef5552dd1253c
closed 2024-04-29 by ivanaamit