gem5/gem5 #1050: Indirect Memory Prefetcher only performs a single indirect prefetch even if the prefetch distance is bigger than 1

sapphi-red, 2024-04-21:
**Describe the bug**
Indirect Memory Prefetcher does not work as expected.
Even if `max_prefetch_distance` is bigger than 1, it only performs a single indirect prefetch.

**Affects version**
Since the version Indirect Memory Prefetcher was added,
to the current develop branch (c54039da5bfe9d2101bfa3889252a15230053b11).

**gem5 Modifications**
None

**To Reproduce**
https://github.com/gem5/gem5/blob/c54039da5bfe9d2101bfa3889252a15230053b11/src/mem/cache/prefetch/indirect_memory.cc#L150-L154
In this for-loop, the same address is pushed to `addresses`.

**Expected behavior**

If I understand the paper correctly, the prefetcher should perform multiple indirect prefetch.
The code would be something like:
```cpp
 for (int delta = 1; delta < distance; delta += 1) {
     Addr index = pfi.getData(b_base_address + b_item_size * delta) // value of B[i + delta]
     Addr pf_addr = pt_entry->baseAddr + 
         (index << pt_entry->shift); 
     addresses.push_back(AddrPriority(pf_addr, 0)); 
 } 
```

At least, I guess it doesn't make sense to push the same address to `addresses`.

**Host Operating System**
N/A

**Host ISA**
N/A

**Compiler used**
N/A

**Additional information**
None

anooprac, 2024-05-01:
My partner and I are UT Ausitn students working on open source contributions for a class project. Could I be assigned to this to test fixes for this bug?

anooprac, 2024-05-02:
I read through the paper also and beleive the fix to pushing different addresses should be 

`for (int delta = 0; delta < distance; delta += 1) {
       Addr pf_addr = pt_entry->baseAddr + ((index + delta) << pt_entry->shift); 
       addresses.push_back(AddrPriority(pf_addr, 0)); 
}`

because this lines up with the breakdown for the formula from the papers, while following what is given in the IndirectMemory class.

anooprac, 2024-05-03:
Instead of `pt_entry->index`, it should be `B[pt_entry->index + delta]`, but how can I access the index array B?