gem5/gem5 #1087: global instruction counter

studyztp, 2024-04-30:
Currently, if we want to schedule an exit event after a certain number of instructions have been executed, we need to use either [BaseCPU::scheduleInstStopAnyThread](https://github.com/gem5/gem5/blob/stable/src/cpu/base.cc#L818) or [this](https://github.com/gem5/gem5/blob/stable/src/cpu/base.cc#L326C18-L326C40). Both methods require scheduling the exit event in advance. However, the current method has two major disadvantages:

1. It is difficult to achieve a precise global instruction count-based exit for multi-core simulations because we can only schedule these events per core. As a result, the instruction counting is not done as a whole. [related mailing list thread](https://www.mail-archive.com/gem5-users@gem5.org/msg21842.html)
2. The checkpoint might capture the future exit event and cause an incorrect exit event when restoring the checkpoint. Related mailing list thread. [related mailing list thread](https://www.mail-archive.com/gem5-users@gem5.org/msg22383.html)

Therefore, I believe it is necessary to have a global instruction counter that can trigger an exit event based on actual instruction counting, and the counting should be across multiple cores. This is easily achievable with the probing feature in gem5.  I can create a pull request for it, but I still want to discuss about if it is needed since there could also be risks associated with continually adding features.

powerjg, 2024-05-09:
If you do this with an optional probe, I think it's a great idea. Please open a PR.

BobbyRBruce, 2024-08-07:
@studyztp L was this ever done?

studyztp, 2024-08-07:
@powerjg @BobbyRBruce Sorry, I don't know why I missed the message. I'll create a PR today!

studyztp, 2024-08-08:
pull request created https://github.com/gem5/gem5/pull/1433

closed 2024-12-06 by erin-le