gem5/gem5 #1119: Build ROCm disk image

ivanaamit, 2024-05-09:
Build ROCm disk image following these instructions: https://github.com/gem5/gem5-resources/pull/29.

Harshil2107, 2024-05-24:
@abmerop I am trying to run the new gpu diskimage with the example script mi200.py as shows in the readme but it leads to the following error. Does this look like a kvm bug or a config issue by me?

```
./build/VEGA_X86/gem5.opt configs/example/gpufs/mi200.py --disk-image /home/harshilp/gem5-resources-worktrees/gem5-resources-repo/src/x86-ubuntu-gpu-ml/disk-image/x86-ubuntu-gpu-ml --kernel  /home/harshilp/gem5-resources-worktrees/gem5-resources-repo/src/x86-ubuntu-gpu-ml/vmlinux-gpu-ml --app ./pytorch_test.py
gem5 Simulator System.  https://www.gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version DEVELOP-FOR-24.0
gem5 compiled May 23 2024 13:08:48
gem5 started May 24 2024 14:15:07
gem5 executing on COE-CS-sterling, pid 1157351
command line: ./build/VEGA_X86/gem5.opt configs/example/gpufs/mi200.py --disk-image /home/harshilp/gem5-resources-worktrees/gem5-resources-repo/src/x86-ubuntu-gpu-ml/disk-image/x86-ubuntu-gpu-ml --kernel /home/harshilp/gem5-resources-worktrees/gem5-resources-repo/src/x86-ubuntu-gpu-ml/vmlinux-gpu-ml --app ./pytorch_test.py

warn: Physical memory size specified is 8GB which is greater than 3GB.  Twice the number of memory controllers would be created.
Global frequency set at 1000000000000 ticks per second
warn: system.workload.acpi_description_table_pointer.rsdt adopting orphan SimObject param 'entries'
warn: No dot file generated. Please install pydot to generate the dot file and pdf.
src/mem/dram_interface.cc:690: warn: DRAM device capacity (8192 Mbytes) does not match the address range assigned (4096 Mbytes)
src/sim/kernel_workload.cc:46: info: kernel located at: /home/harshilp/gem5-resources-worktrees/gem5-resources-repo/src/x86-ubuntu-gpu-ml/vmlinux-gpu-ml
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
src/mem/dram_interface.cc:690: warn: DRAM device capacity (128 Mbytes) does not match the address range assigned (16384 Mbytes)
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
      0: system.pc.south_bridge.cmos.rtc: Real-time clock set to Sun Jan  1 00:00:00 2012
system.pc.com_1.device: Listening for connections on port 3456
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
system.remote_gdb: Listening for connections on port 7000
src/dev/intel_8254_timer.cc:128: warn: Reading current count from inactive timer.
Running the simulation
src/cpu/kvm/base.cc:169: info: KVM: Coalesced MMIO disabled by config.
src/sim/simulate.cc:199: info: Entering event queue @ 0.  Starting simulation...
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x3a) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x48) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0xe1) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x12) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x11) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x4b564d01) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x4b564d00) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x40000000) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x40000001) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x40000020) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x40000021) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x40000022) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x40000023) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x40000100) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x40000101) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x40000102) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x40000103) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x40000104) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x40000105) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x40000003) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x40000002) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x40000010) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x40000080) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x400000b0) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x40000073) unsupported by gem5. Skipping.
src/arch/x86/kvm/x86_cpu.cc:1688: warn: kvm-x86: MSR (0x40000106) unsupported by gem5. Skipping.
[... cut ...]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9d68)[0x71035c87b9c8]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af)[0x71035c9c23af]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x69de)[0x71035c87863e]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af)[0x71035c9c23af]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyEval_EvalCode+0xbe)[0x71035c9bd3de]
--- END LIBC BACKTRACE ---
For more info on how to address this issue, please visit https://www.gem5.org/documentation/general_docs/common-errors/ 

Aborted (core dumped)
```

abmerop, 2024-05-24:
Looking into it.  I have seen that panic before

abmerop, 2024-05-24:
I think the application run might be a KVM issue. I am not seeing this.

For this issue in particular, the disk image does build, correct?

If we think it's a KVM issue, maybe we could start making a list of processors KVM works on.  My dev machine is:

$ cat /proc/cpuinfo  | grep "model name" | head -n 1
model name      : AMD EPYC 7601 32-Core Processor

Harshil2107, 2024-05-25:
I am running it in intel 13700 cpu

I will run it on an AMD machine, to see what happenes.

Just to double check, I dont need any specific dependency for this to work right?

abmerop, 2024-05-25:
nothing special as long as KVM works..

Harshil2107, 2024-05-30:
Kernel link: https://storage.googleapis.com/dist.gem5.org/dist/v24-0/gpu-fs/kernel/vmlinux-gpu-ml.gz
Diskimage link: https://storage.googleapis.com/dist.gem5.org/dist/v24-0/gpu-fs/diskimage/x86-ubuntu-gpu-ml.gz

Harshil2107, 2024-06-03:
Closing this issue as the diskimage and kernel have been built and uploaded. An issue representing the error have been made at #1195

closed 2024-06-03 by Harshil2107