gem5/gem5 #1121: Weekly tests failure

ivanaamit, 2024-05-10:
Weekly tests are failing https://github.com/gem5/gem5/actions/runs/8948827382. 

We switched to Ubuntu 24.04, and it uses GCC 13 which has some compatibility issues with `dramsys`.

```
src/systemc/ext/systemc_home/include/../../tlm_core/2/sockets/target_socket.hh:95:5: error: ‘void tlm::
tlm_base_target_socket<BUSWIDTH, FW_IF, BW_IF, N, POL>::bind(base_initiator_socket_type&) [with unsigned int 
BUSWIDTH = 32; FW_IF = tlm::tlm_fw_transport_if<>; BW_IF = tlm::tlm_bw_transport_if<>; int N = 0; sc_core::
sc_port_policy POL = sc_core::SC_ONE_OR_MORE_BOUND; base_initiator_socket_type = tlm::tlm_base_initiator_socket_b
<32, tlm::tlm_fw_transport_if<>, tlm::tlm_bw_transport_if<> >]’ was hidden [-Werror=overloaded-virtual=]
```

derchr, 2024-09-16:
This is due to more extensive checks in new compiler versions that now issue a `overloaded-virtual` warning in the SystemC TLM 2.0 header file. The warning then gets treated by an error due to the `-Werror` flag in `SConstruct` of gem5.

This is what the SystemC release notes says about this issue:

>   - On some recent compilers (e.g. Clang, Solaris Studio), warnings are
>     generated about the "hidden overloaded virtual function" bind of the
>     sc_port(_b) class templates in the 'specialized_signals' example.
>     These warnings are caused by the introduction of the virtual bind
>     implementation in IEEE 1666-2011.
>     As a workaround, check your compiler documentation how to suppress
>     the warning (e.g. 'CXXFLAGS=-Wno-overloaded-virtual') or mark
>     the SystemC include directory as "system directory" by using
>     '-isystem ${SYSTEMC_HOME}/include' (or equivalent) instead of
>     the usual '-I'.
> 
>     This also affects the TLM-2.0 sockets, see Section 9.3.

So would be `-Wno-overloaded-virtual` an option for gem5?

BobbyRBruce, 2024-10-07:
Fixex by https://github.com/gem5/gem5/pull/1576

PR #1576 (2024-09-26): ext: Fix GCC v13+ comp of systemc due to problematic overloaded-virtual warn
Fixes #1121 in line with the following suggesting:

https://github.com/gem5/gem5/issues/1121#issuecomment-2352743409
commit 054790ad (2024-09-26): ext: Fix GCC v13+ comp of systemc due to problematic overloaded-virtual warn (#1576)

Fixes #1121 in line with the following suggesting:

https://github.com/gem5/gem5/issues/1121#issuecomment-2352743409
closed 2024-10-07 by BobbyRBruce