gem5/gem5 #1126: kvm full system workloads fail to run on x86

Harshil2107, 2024-05-13:
**Describe the bug**
This bug was introduced in #1065 
KVM full system workload fail with the following error:
```
build/X86/gem5.opt configs/example/gem5_library/test_script.py 
gem5 Simulator System.  https://www.gem5.org/
gem5 is copyrighted software; use the --copyright option for details.
 
gem5 version DEVELOP-FOR-24.0
gem5 compiled May 10 2024 11:06:09
gem5 started May 10 2024 11:10:58
gem5 executing on COE-CS-sterling, pid 65871
command line: build/X86/gem5.opt configs/example/gem5_library/test_script.py
 
info: Using default config
Resource 'x86-ubuntu-22.04-img' was not found locally. Downloading to '/home/harshilp/.cache/gem5/x86-ubuntu-22.04-img.gz'...
Finished downloading resource 'x86-ubuntu-22.04-img'.
Decompressing resource 'x86-ubuntu-22.04-img' ('/home/harshilp/.cache/gem5/x86-ubuntu-22.04-img.gz')...
Finished decompressing resource 'x86-ubuntu-22.04-img'.
warn: <orphan X86Board>.kvm_vm already has parent not resetting parent.
        Note: kvm_vm is not a parameter of X86Board
warn: (Previously declared as <orphan X86Board>.processor.kvm_vm
Global frequency set at 1000000000000 ticks per second
warn: No dot file generated. Please install pydot to generate the dot file and pdf.
src/mem/dram_interface.cc:690: warn: DRAM device capacity (8192 Mbytes) does not match the address range assigned (4096 Mbytes)
src/sim/kernel_workload.cc:46: info: kernel located at: /home/harshilp/.cache/gem5/x86-linux-kernel-5.4.0-105-generic
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
      0: board.pc.south_bridge.cmos.rtc: Real-time clock set to Sun Jan  1 00:00:00 2012
board.pc.com_1.device: Listening for connections on port 3456
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
src/dev/intel_8254_timer.cc:128: warn: Reading current count from inactive timer.
board.remote_gdb: Listening for connections on port 7000
src/cpu/kvm/base.cc:170: info: KVM: Coalesced MMIO disabled by config.
src/cpu/kvm/base.cc:170: info: KVM: Coalesced MMIO disabled by config.
src/sim/simulate.cc:199: info: Entering event queue @ 0.  Starting simulation...
src/cpu/kvm/perfevent.cc:217: panic: PerfKvmCounter::attach failed (2)
Memory Usage: 3793480 KBytes
Program aborted at tick 0
src/cpu/kvm/perfevent.cc:217: panic: PerfKvmCounter::attach failed (2)
Memory Usage: 3793488 KBytes
Aborted (core dumped)
```

The test config used is the following:
```
from gem5.coherence_protocol import CoherenceProtocol
from gem5.components.boards.x86_board import X86Board
from gem5.components.memory.single_channel import SingleChannelDDR3_1600
from gem5.components.processors.cpu_types import CPUTypes
from gem5.components.processors.simple_processor import (
    SimpleProcessor
)
from gem5.components.processors.simple_switchable_processor import (
    SimpleSwitchableProcessor,
)
from gem5.isas import ISA
from gem5.resources.resource import obtain_resource
from gem5.simulate.exit_event import ExitEvent
from gem5.simulate.simulator import Simulator
from gem5.utils.requires import requires

# This runs a check to ensure the gem5 binary is compiled to X86 and to the
# MESI Two Level coherence protocol.
requires(
    isa_required=ISA.X86,
    coherence_protocol_required=CoherenceProtocol.MESI_TWO_LEVEL,
)

from gem5.components.cachehierarchies.ruby.mesi_two_level_cache_hierarchy import (
    MESITwoLevelCacheHierarchy,
)

# Here we setup a MESI Two Level Cache Hierarchy.
cache_hierarchy = MESITwoLevelCacheHierarchy(
    l1d_size="16kB",
    l1d_assoc=8,
    l1i_size="16kB",
    l1i_assoc=8,
    l2_size="256kB",
    l2_assoc=16,
    num_l2_banks=1,
)

# Setup the system memory.
memory = SingleChannelDDR3_1600(size="3GB")

# Here we setup the processor. This is a special switchable processor in which
# a starting core type and a switch core type must be specified. Once a
# configuration is instantiated a user may call `processor.switch()` to switch
# from the starting core types to the switch core types. In this simulation
# we start with KVM cores to simulate the OS boot, then switch to the Timing
# cores for the command we wish to run after boot.
processor = SimpleProcessor(
    cpu_type=CPUTypes.KVM,
    isa=ISA.X86,
    num_cores=2,
)

# Here we setup the board. The X86Board allows for Full-System X86 simulations.
board = X86Board(
    clk_freq="3GHz",
    processor=processor,
    memory=memory,
    cache_hierarchy=cache_hierarchy,
)

# Here we set the Full System workload.
# The `set_kernel_disk_workload` function for the X86Board takes a kernel, a
# disk image, and, optionally, a command to run.

workload = obtain_resource("x86-ubuntu-22.04-boot-with-systemd")
# command = (
#     "m5 exit;"
# )
# workload.set_parameter("readfile_contents", command)

board.set_workload(workload)

def exit_event_handler():
    print("first exit event")
    yield False
    print("second exit event")
    yield False
    print("third exit event")
    yield False
    print("fourth exit event")
    yield False
    print("fifth exit event")
    yield True

simulator = Simulator(
    board=board,
    on_exit_event={
        # Here we want override the default behavior for the first m5 exit
        # exit event. Instead of exiting the simulator, we just want to
        # switch the processor. The 2nd m5 exit after will revert to using
        # default behavior where the simulator run will exit.
        ExitEvent.EXIT: exit_event_handler()
    },
)
simulator.run()
```

The above is an test script that should get 3 exits.
**Affects version**
develop for 24.0

**To Reproduce**
Steps to reproduce the behavior. Please assume starting from a clean repository:

1. Compile gem5 with command `scons build/X86/gem5.opt`
2. Run the test config above or any fs simulation the uses x86 kvm

**Expected behavior**
We should get 3 m5 exit events.

powerjg, 2024-05-13:
Have you tried setting "allow_hybrid" to false on the KVM object?

Harshil2107, 2024-05-13:
No, I have not tried that

Harshil2107, 2024-05-13:
Changing https://github.com/gem5/gem5/blob/10b24dc9a477d9c3b82c1a4c86c7462d65d3a64d/src/cpu/kvm/BaseKvmCPU.py#L76

to false allows me to run simulation again.

Harshil2107, 2024-05-15:
I will keep updating this message as I get updates

Intel with P/E cores:
stable: KVM works
Current Develop: KVM Doesnt work as highlighted above
Cherry picked 1065 on stable: running
Cherry picked 1040 on stable: 
Cherry picked  1065 + 1040 on stabe: 

Intel without P/E cores:
stable: KVM works
develop: running
Cherry picked 1065 on stable: running
Cherry picked 1040 on stable: 
Cherry picked  1065 + 1040 on stabe:

abmerop, 2024-05-15:
#1138 fixes "Error 2".  Not sure about the other error, have not seen it.

powerjg, 2024-05-15:
@Harshil2107 can you run with Matt's new PR and see what happens?

If all cases fail with #1065, I think we should revert it until @nmosier can resubmit.

Harshil2107, 2024-05-15:
I will run #1138 on the AMD machine as KVM works with #1065 on that macine

Harshil2107, 2024-05-15:
The script with KVM runs on AMD when I checkedout #1138

closed 2024-06-06 by ivanaamit