gem5/gem5 #1130: cpu: control hazard in MinorCPU

robhau, 2024-05-14:
**Bug description**
```asm
L1:
    ....
    j L3
L2:
    lw a1, 0(a0)
    ....
L3:
    ....
```

The program above can lead to an error when using MinorCPU. The reason is, that the memory access is started after the unconditional jump. If `a0` does not point to a valid memory region, an error is thrown in `BaseXBar::findPort`.

https://github.com/gem5/gem5/blob/b279e40cb788087bab9926fa92484f57ab1e8978/src/mem/xbar.cc#L333-L370

```txt
src/mem/xbar.cc:368: fatal: Unable to find destination for [0:0x4] on system.membus
Use --debug-flags=PortTrace to see the port trace of the packet.
```

**Affects version**
Gem5 development branch, Commit SHA: b279e40cb788087bab9926fa92484f57ab1e8978

**gem5 Modifications**
No modifications.

**To Reproduce**
I pushed a minimal working example in https://github.com/robhau/gem5/tree/minor_cpu_bug_minimal_working_example

1. Set path to multilib RISC-V compiler in `firmware/Makefile`
2. `bash execute.sh`

Firmware code:

```asm
.global start
.type start,@function
start:
    .option push
    .option norelax
    la gp, __global_pointer
    .option pop
    la sp, __stack_top
    li a0, 0
    add a0, a1, a2
    add a0, a1, a2
    add a0, a1, a2
    add a0, a1, a2
    add a0, a1, a2
    add a0, a1, a2
    add a0, a1, a2
    add a0, a1, a2
    add a0, a1, a2
    add a0, a1, a2
    add a0, a1, a2
    li a1, 1
    j L2
L1:
    lw a2, 0(a0)
    lw a3, 4(a0)
    lw a4, 8(a0)
    j exit
L2:
    li a0, 0
    li a0, 0
    li a0, 0
    li a0, 0
    li a0, 0
    li a0, 0
[... 261 lines cut ...]
 734375: system.cpu.execute.lsq: Received translation response for request: 0/1.1/2/19.19 pc: 0x80000042 (c_lw) delayed:0 
 734375: system.cpu.execute.lsq: Setting state from InTranslation to Translated for request: 0/1.1/2/19.19 pc: 0x80000042 (c_lw)
 734375: system.cpu.execute.lsq: No forwardable data from store buffer
 734375: system.cpu.execute.lsq: Trying to send request: 0/1.1/2/19.19 pc: 0x80000042 (c_lw) addr: 0x0
src/mem/xbar.cc:368: fatal: Unable to find destination for [0:0x4] on system.membus
Use --debug-flags=PortTrace to see the port trace of the packet.
Memory Usage: 448876 KBytes
```
**Expected behavior**
The memory access should be canceled after the unconditional jump.

If i remove `li a1, 1` from `start:`, it works.

**Host Operating System**
Ubuntu 22.04.4 LTS

**Host ISA**
X86

**Compiler used**
for gem5:
g++ version 11.4.0
linker: mold
for firmware:
RISC-V GNU Compiler Toolchain (riscv64-unknown-elf-gcc version 13.2.0, GNU objdump version 2.42)

robhau, 2024-05-14:
> If i remove `li a1, 1` from `start:`, it works.

Comment because it get lost in the text wall.

giactra, 2024-05-30:
This is a weird issue to happen on a in-order CPU. Is it possible this is a just RISCV issue (eg the instruction wrongly mislabelled) ?

I am looking at the line:

` 703125: system.cpu.execute.lsq: No matching memory response for inst: 0/1.1/2/18.18 pc: 0x80000040 (c_j)`

And I don't understand why a jump instruction should be sent to the LSQ

robhau, 2024-05-30:
Hi, thank you for your comment 😄.
> Is it possible this is a just RISCV issue (eg the instruction wrongly mislabelled) ?

I do not think that the instruction is mislabelled, because at `0x80000040` there is also a jump instruction in the disassembly.

> I am looking at the line:
> 
> ` 703125: system.cpu.execute.lsq: No matching memory response for inst: 0/1.1/2/18.18 pc: 0x80000040 (c_j)`
> 
> And I don't understand why a jump instruction should be sent to the LSQ

In `getCommitingThread()`, for `head_inflight_inst` it is checked if the instruction is not in the LSQ or if there is a response for the instruction. This results in the debug message.

https://github.com/gem5/gem5/blob/65b86cfac98338153936b6b308368f8edf961d3d/src/cpu/minor/execute.cc#L1686-L1751

The `c_j` lies at address `0x80000040`. The program counter is set from `0x80000042` to `0x80000044`, I guess because it increases before we reach the jump instruction.

 734375: global: ExecContext setting PC: (0x80000042=>0x80000044).(0=>1)

Can this cause problems?

giactra, 2024-05-31:
Hi @robhau 

> Hi, thank you for your comment 😄.
> 
> > Is it possible this is a just RISCV issue (eg the instruction wrongly mislabelled) ?
> 
> I do not think that the instruction is mislabelled, because at `0x80000040` there is also a jump instruction in the disassembly.
> 
> `80000040: a029 j 8000004a <L2>`
> 

By mislabelled I don't mean mis-decoded. Basically in gem5 an instruction is defined (labelled) with some StaticInstFlag which are used by the CPU model to understand which kind of instruction it is dealing with (see https://github.com/gem5/gem5/blob/stable/src/cpu/StaticInstFlags.py). Is the instruction defined with **IsUncondControl**?

> > I am looking at the line:
> > ` 703125: system.cpu.execute.lsq: No matching memory response for inst: 0/1.1/2/18.18 pc: 0x80000040 (c_j)`
> > And I don't understand why a jump instruction should be sent to the LSQ
> 
> In `getCommitingThread()`, for `head_inflight_inst` it is checked if the instruction is not in the LSQ or if there is a response for the instruction. This results in the debug message (line 1712)
> 
> https://github.com/gem5/gem5/blob/65b86cfac98338153936b6b308368f8edf961d3d/src/cpu/minor/execute.cc#L1686-L1751

You are getting the condition wrong:
`                 (!inst->inLSQ || (lsq.findResponse(inst) != NULL)); `

Checks if there is a response in the LSQ only if the instruction **is** in the LSQ. So for whatever reason the Jump instruction gets added to the LSQ. This looks wrong to me. I would suggest you to debug the matter with GDB

robhau, 2024-06-03:
Hi @giactra 
> Hi @robhau
>
>
> By mislabelled I don't mean mis-decoded. Basically in gem5 an instruction is defined (labelled) with some StaticInstFlag which are used by the CPU model to understand which kind of [...]

Thank you. The instructions are not mislabed.

**Jumps (32 Bit format):**

https://github.com/gem5/gem5/blob/3b7307182f272756bd805ca2d5f4af79efa639a0/src/arch/riscv/isa/decoder.isa#L4878-L4881
https://github.com/gem5/gem5/blob/3b7307182f272756bd805ca2d5f4af79efa639a0/src/arch/riscv/isa/decoder.isa#L4872-L4876

**Compressed Jumps (16 bit format, used here):**

https://github.com/gem5/gem5/blob/3b7307182f272756bd805ca2d5f4af79efa639a0/src/arch/riscv/isa/decoder.isa#L351-L353
https://github.com/gem5/gem5/blob/3b7307182f272756bd805ca2d5f4af79efa639a0/src/arch/riscv/isa/decoder.isa#L215-L218
https://github.com/gem5/gem5/blob/3b7307182f272756bd805ca2d5f4af79efa639a0/src/arch/riscv/isa/decoder.isa#L451-L457
https://github.com/gem5/gem5/blob/3b7307182f272756bd805ca2d5f4af79efa639a0/src/arch/riscv/isa/decoder.isa#L471-L474
> You are getting the condition wrong: ` (!inst->inLSQ || (lsq.findResponse(inst) != NULL));`
>
> Checks if there is a response in the LSQ only if the instruction **is** in the LSQ. So for whatever [...]

Thank you for the clarification ☺️.

I quoted the wrong code section. `findResponse` is called in `evaluate()`:

https://github.com/gem5/gem5/blob/3b7307182f272756bd805ca2d5f4af79efa639a0/src/cpu/minor/execute.cc#L1554-L1571

robhau, 2024-06-03:
Could the branch target predictor be the problem?

The branch predictor predicts taken, because it is an unconditional branch.

https://github.com/gem5/gem5/blob/3b7307182f272756bd805ca2d5f4af79efa639a0/src/cpu/pred/bpred_unit.cc#L120-L163

After that, the branch target buffer is looked up. However, there is a miss, because we this is the first time this jump is executed
and the branch type is not considered.

https://github.com/gem5/gem5/blob/3b7307182f272756bd805ca2d5f4af79efa639a0/src/cpu/pred/bpred_unit.cc#L164-L190

https://github.com/gem5/gem5/blob/3b7307182f272756bd805ca2d5f4af79efa639a0/src/cpu/pred/simple_btb.cc#L129-L144

Therefore, `hist->targetProvider` stays `TargetProvider::NoTarget`. This results in an untaken branch.

https://github.com/gem5/gem5/blob/3b7307182f272756bd805ca2d5f4af79efa639a0/src/cpu/pred/bpred_unit.cc#L289-L293

giactra, 2024-06-03:
> Could the branch target predictor be the problem?
>
> The branch predictor predicts taken, because it is an unconditional branch. [...]

I thought about this at the beginning (wondering why an unconditional branch was predicted as not-taken). However the following reasoning convinced me there's something more to it:

1) If we are missing in the BTB, there are only two things we can do. One is stalling the front-end, the other is to predict a NT branch. I think always going for the second (even in the counter intuitive case of an unconditional branch) is a reasonable approach.

2) A CPU should recover anyway from a misprediction as early as possible. In the O3CPU this happens after decode for a direct unconditional branch: https://github.com/gem5/gem5/blob/stable/src/cpu/o3/decode.cc#L712. This doesn't seem to be handled in MinorCPU. Instead this seems to be solved altogether when committing the branch: https://github.com/gem5/gem5/blob/stable/src/cpu/minor/execute.cc#L284C16-L284C27

My suspicion is that the load is issued before the branch commit, therefore before it can resteer the pipeline.

While we should solve it (maybe be adding a similar check as the one we have in the O3CPU?) it seems to me a "niche" corner case. It only happens when virtual memory is disabled (at boot) and therefore we are allowed to send an invalid request down the memory subsystem. Normally (with virtual memory on) the MMU would return an exception to the translation request, and no memory reference would go out of the LSQ.

Let me know if it makes sense to you

robhau, 2024-06-04:
Thank you @giactra, makes sense to me.

> A CPU should recover anyway from a misprediction as early as possible. In the O3CPU this happens after decode for a direct unconditional branch: [...]

Adapting the handling from O3CPU sounds good. Can we add it right after inserting a decoded instruction into the output buffer of the decode stage?

https://github.com/gem5/gem5/blob/dad5c7b6f7434ec7668192edffad83bc31a1d5f7/src/cpu/minor/decode.cc#L247C1-L248C32

> While we should solve it (maybe be adding a similar check as the one we have in the O3CPU?) it seems to me a "niche" corner case. It only happens when virtual memory is disabled (at boot) and [...]

Since we only carry out bare metal simulations, we run into this problem quite often 😅.

erin-le, 2025-04-17:
@robhau Hello, thank you for opening this issue! Did you ever get around to making the change you described above?

robhau, 2025-04-25:
Hi @erin-le. Unfortunately, I did not have time to work on this issue.