gem5/gem5 #1133: cpu: Abort attempt to rename a pinned register

vsoriap, 2024-05-14:
**Description of the bug**
Running simple SVE code with gather operations, I get the following message:

```
gem5.opt: build/ARM/cpu/o3/rename_map.cc:85: gem5::o3::SimpleRenameMap::RenameInfo gem5::o3::SimpleRenameMap::rename(const gem5::RegId&): Assertion `arch_reg.getNumPinnedWrites() == 0' failed.
Program aborted at tick 257229480756
```

Digging in the bug, I've generated a simple Exec trace. What I've observed is that after executing the instruction `ld1   {z21}, p0/z, [x4, z32]`, the kernel launches a page fault. After the page fault, the program recovers and tries to execute the affected instruction again. This second time, gem5 arises the above abort. 

**Explanation of the Bug**

When Gather SVE instructions arrive to Rename, they use pinned registers to avoid renaming the same register for each muop. This code can be found [here](https://github.com/gem5/gem5/blob/115322319c709a6e2a12e8b5c5299aa813d94633/src/cpu/o3/rename_map.cc#L72).

```
SimpleRenameMap::rename(const RegId& arch_reg)
{
    PhysRegIdPtr renamed_reg;
    // Record the current physical register that is renamed to the
    // requested architected register.
    PhysRegIdPtr prev_reg = map[arch_reg.index()];

    if (arch_reg.is(InvalidRegClass)) {
        assert(prev_reg->is(InvalidRegClass));
        renamed_reg = prev_reg;
    } else if (prev_reg->getNumPinnedWrites() > 0) {
        // Do not rename if the register is pinned
        assert(arch_reg.getNumPinnedWrites() == 0);  // Prevent pinning the
                                                     // same register twice
        DPRINTF(Rename, "Renaming pinned reg, numPinnedWrites %d\n",
                prev_reg->getNumPinnedWrites());
        renamed_reg = prev_reg;
        renamed_reg->decrNumPinnedWrites();
    } else {
        renamed_reg = freeList->getReg();
        map[arch_reg.index()] = renamed_reg;
        renamed_reg->setNumPinnedWrites(arch_reg.getNumPinnedWrites());
        renamed_reg->setNumPinnedWritesToComplete(
            arch_reg.getNumPinnedWrites() + 1);
    }
```
I've checked in the Rename trace to see that the register was correctly renamed for the first time. It executed for the 16 muops that compose the vector instruction, decrementing the NumPinnedWrites from 16 to 0. However, it raises a page fault when it executes the 11th muop. After the exception, the remaining muops are squashed from the pipeline. Then gem5 executes the following [code](https://github.com/gem5/gem5/blob/115322319c709a6e2a12e8b5c5299aa813d94633/src/cpu/o3/dyn_inst.cc#L323):

```
void
DynInst::setSquashed()
{
    status.set(Squashed);

    if (!isPinnedRegsRenamed() || isPinnedRegsSquashDone())
        return;

    // This inst has been renamed already so it may go through rename
    // again (e.g. if the squash is due to memory access order violation).
    // Reset the write counters for all pinned destination register to ensure
    // that they are in a consistent state for a possible re-rename. This also
    // ensures that dest regs will be pinned to the same phys register if
    // re-rename happens.
    for (int idx = 0; idx < numDestRegs(); idx++) {
        PhysRegIdPtr phys_dest_reg = renamedDestIdx(idx);
        if (phys_dest_reg->isPinned()) {
            phys_dest_reg->incrNumPinnedWrites();
            if (isPinnedRegsWritten())
                phys_dest_reg->incrNumPinnedWritesToComplete();
        }
    }
    setPinnedRegsSquashDone();
}
```
This code increases the number of numPinnedWrites to 4. Expecting that the instruction would be executed from the failing muop.

The second time the SVE gather is executed, the gem5 execution gets in the second case (else if getNumPinnedWrites() > 0) rather than getting inside the final else. Since the first muop has the `arch_reg.getNumPinnedWrites() ` set to 16, an abort arises.

**Affects version**
Affected in gem5 v22.0.0, but same code exists in gem5 v23.0.0

**gem5 Modifications**
No modification

**To Reproduce**
Steps to reproduce the behavior. Please assume starting from a clean repository:

1. Compile gem5 with ARM ISA
2. Execute the simulation with fs mode and run an SPMV code vectorized with SVE and a vector length of 2048 bits.

**Host Operating System**
Ubuntu 22.04

**Host ISA**
X86

**Compiler used**
GCC-10

giactra, 2024-05-20:
I need to look at this more carefully.
An initial solution (a bit of a hack), would be to handle squashing due to faulting differently from simple re-execution (in the first case we flush the pipeline completely). For example in:

https://github.com/gem5/gem5/blob/stable/src/cpu/o3/dyn_inst.cc#L405

We could add the register pinning handling code. Rather than incrementing the number of pinned writes with

`phys_dest_reg->incrNumPinnedWrites()`

we could reset it completely. Something like:

    for (int idx = 0; idx < numDestRegs(); idx++) {
        PhysRegIdPtr phys_dest_reg = renamedDestIdx(idx);
        if (phys_dest_reg->isPinned()) {
            phys_dest_reg->resetPinnedWrites();
        }
    }
    setPinnedRegsSquashDone();

Let me know if this works

Note: I might be missing something in the snippet above

erin-le, 2025-03-20:
@vsoriap Hello, thank you for reporting this issue. Do you know if this is still an ongoing bug?

vsoriap, 2025-03-21:
Hi, @erin-le. We didn't devise a solution for this problem, as I could not replicate the error. I would close it and re-open it if somebody else finds a similar problem.

powerjg, 2025-03-21:
Thanks for the info @vsoriap!

mayagokhale, 2025-04-25:
The same error occurs in riscv: 
gem5.opt: src/cpu/o3/[rename_map.cc:84](http://rename_map.cc:84/): gem5::o3::SimpleRenameMap::RenameInfo gem5::o3::SimpleRenameMap::rename(const gem5::RegId&): Assertion `arch_reg.getNumPinnedWrites() == 0' failed.

This one runs the barcelona benchmark canneal with "vector, large" options and config for the core of L1D = 16KiB, L2 = 256KiB, vlen = 128 elen = 64 
[https://github.com/RALC88/riscv-vectorized-benchmark-suite/tree/master/_canneal](url)

[riscv-rvv-se.py.txt](https://github.com/user-attachments/files/19902408/riscv-rvv-se.py.txt) (just delete the .txt part)

command: /gem5/build/RISCV/gem5.opt /.../riscv-rvv-se.py -d 16KiB -2 256KiB -v 128 -e 64 bin/canneal_vector.exe -p="1 100 300 input/100.nets 8"

It runs a long time before it finally gives the assertion violation.

We have several other examples showing the same error.

erin-le, 2025-05-07:
@mayagokhale Hello, I tried reproducing the error on current develop and current stable (v24.1.0.3) using the instructions you gave. I did encounter an error, but it wasn't the assert that others have encountered. The error message is below:
```
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
```
Could you let me know what version(s) of gem5 you are using?

mayagokhale, 2025-05-07:
@erin-le thanks so much for looking at this issue. 
 was using the develop branch eec9622e5738920087cd2e57b80d9036a9971b71.

polpetras, 2025-09-18:
Hello all (@vsoriap , @powerjg @giactra). I hope you are doing well.

We have tested with the following patch, which indeed fixes the issue, and we have successful runs of HPCG in ARM+O3 CPU+SVE setup.

`diff --git a/src/cpu/o3/dyn_inst.cc b/src/cpu/o3/dyn_inst.cc
index 0f500d8923..ababe74e59 100644
--- a/src/cpu/o3/dyn_inst.cc
+++ b/src/cpu/o3/dyn_inst.cc
@@ -324,6 +324,22 @@ DynInst::setSquashed()
{
status.set(Squashed);

+    // Handle fault-induced squashes
+    // (e.g., from page faults in SVE gathers)
+    // If the instruction has pinned destination registers and is squashed
+    // due to a fault, reset the pinned writes to prevent assertion failures
+    if (isPinnedRegsRenamed() && !isPinnedRegsSquashDone() && fault != NoFault) {
+        for (int idx = 0; idx < numDestRegs(); ++idx) {
+            auto phys_dest_reg = renamedDestIdx(idx);
+            if (phys_dest_reg && phys_dest_reg->isPinned()) {
+                phys_dest_reg->setNumPinnedWrites(0);
+                phys_dest_reg->setNumPinnedWritesToComplete(0);
+            }
+        }
+        setPinnedRegsSquashDone();
+        return; // Prevent non-fault squash logic from executing
+    }
+
if (!isPinnedRegsRenamed() || isPinnedRegsSquashDone())
return;`

Would you consider incorporating this patch in the current version of gem5? Shall I do a pull request?

polpetras, 2025-09-26:
@mayagokhale Does it crash in the exact same instruction/location? have you looked into the instruction sequence that leads to the new error? (https://www.gem5.org/documentation/general_docs/cpu_models/visualization/).

mayagokhale, 2025-09-29:
Here are the instructions leading up to it:
578323000: system.cpu: T0 : 0x10dee @_ZN15annealer_thread3RunEv+332. 3 : vlse32_v_micro v10, (a0), zero, zero : SimdStridedLoad :  D=[02000000_02000000_02000000_ffffffff_ffffffff_ffffffff_ffffffff_ffffffff] A=0xac5a4
578323500: system.cpu: T0 : 0x10dee @_ZN15annealer_thread3RunEv+332. 4 : vlse32_v_micro v10, (a0), zero, zero : SimdStridedLoad :  D=[02000000_02000000_02000000_02000000_ffffffff_ffffffff_ffffffff_ffffffff] A=0xac5a4
578324000: system.cpu: T0 : 0x10dee @_ZN15annealer_thread3RunEv+332. 5 : vlse32_v_micro v10, (a0), zero, zero : SimdStridedLoad :  D=[02000000_02000000_02000000_02000000_02000000_ffffffff_ffffffff_ffffffff] A=0xac5a4
578324500: system.cpu: T0 : 0x10dee @_ZN15annealer_thread3RunEv+332. 6 : vlse32_v_micro v10, (a0), zero, zero : SimdStridedLoad :  D=[02000000_02000000_02000000_02000000_02000000_02000000_ffffffff_ffffffff] A=0xac5a4
578325000: system.cpu: T0 : 0x10dee @_ZN15annealer_thread3RunEv+332. 7 : vlse32_v_micro v10, (a0), zero, zero : SimdStridedLoad :  D=[02000000_02000000_02000000_02000000_02000000_02000000_02000000_ffffffff] A=0xac5a4
578325500: system.cpu: T0 : 0x10dee @_ZN15annealer_thread3RunEv+332. 8 : vlse32_v_micro v10, (a0), zero, zero : SimdStridedLoad :  D=[02000000_02000000_02000000_02000000_02000000_02000000_02000000_02000000] A=0xac5a4
578326500: system.cpu: T0 : 0x10df2 @_ZN15annealer_thread3RunEv+336    : vlse32_v v10, (a3), zero, v0.t
578326500: system.cpu: T0 : 0x10df2 @_ZN15annealer_thread3RunEv+336. 0 : vpinvd_v_micro v10, v10    : SimdMisc :  D=[02000000_02000000_02000000_02000000_02000000_02000000_02000000_02000000]
gem5 has encountered a segmentation fault!

polpetras, 2025-09-29:
Can you post the Exec debug trace (similar to "578323000: system.cpu: T0 : 0x10dee...."), just before the assert error ?

mayagokhale, 2025-09-30:
535151000: board.processor.cores.core: T0 : 0x118e8 @_ZN7netlist18get_random_elementEPllP3Rng+30    : bne a0, s1, 18             : IntAlu : 
535152000: board.processor.cores.core: T0 : 0x118fa @_ZN7netlist18get_random_elementEPllP3Rng+48    : ld a1, 16(s3)              : MemRead :  D=0x00000000000a5fa0 A=0x7ffffffffffff
df8
535153000: board.processor.cores.core: T0 : 0x118fe @_ZN7netlist18get_random_elementEPllP3Rng+52    : addi a2, zero, 112         : IntAlu :  D=0x0000000000000070
535153000: board.processor.cores.core: T0 : 0x11902 @_ZN7netlist18get_random_elementEPllP3Rng+56    : mul a2, a0, a2             : IntMult :  D=0x0000000000002220
535153000: board.processor.cores.core: T0 : 0x11906 @_ZN7netlist18get_random_elementEPllP3Rng+60    : c_add a1, a2               : IntAlu :  D=0x00000000000a81c0
535153000: board.processor.cores.core: T0 : 0x11908 @_ZN7netlist18get_random_elementEPllP3Rng+62    : sd a0, 0(s2)               : MemWrite :  D=0x000000000000004e A=0x7fffffffffff
fcc8
535153000: board.processor.cores.core: T0 : 0x1190c @_ZN7netlist18get_random_elementEPllP3Rng+66    : c_mv a0, a1                : IntAlu :  D=0x00000000000a81c0
535153000: board.processor.cores.core: T0 : 0x1190e @_ZN7netlist18get_random_elementEPllP3Rng+68    : c_ldsp ra, 40(sp)          : MemRead :  D=0x0000000000010d7e A=0x7ffffffffffff
ca8
535153000: board.processor.cores.core: T0 : 0x11910 @_ZN7netlist18get_random_elementEPllP3Rng+70    : c_ldsp s0, 32(sp)          : MemRead :  D=0x0000000000000000 A=0x7ffffffffffff
ca0
535153000: board.processor.cores.core: T0 : 0x11912 @_ZN7netlist18get_random_elementEPllP3Rng+72    : c_ldsp s1, 24(sp)          : MemRead :  D=0x00000000000a8b60 A=0x7ffffffffffff
c98
535154000: board.processor.cores.core: T0 : 0x11914 @_ZN7netlist18get_random_elementEPllP3Rng+74    : c_ldsp s2, 16(sp)          : MemRead :  D=0x7ffffffffffffdc8 A=0x7ffffffffffff
c90
535154000: board.processor.cores.core: T0 : 0x11916 @_ZN7netlist18get_random_elementEPllP3Rng+76    : c_ldsp s3, 8(sp)           : MemRead :  D=0x0000000000000000 A=0x7ffffffffffff
c88
535154000: board.processor.cores.core: T0 : 0x11918 @_ZN7netlist18get_random_elementEPllP3Rng+78    : c_addi16sp sp, 48          : IntAlu :  D=0x7ffffffffffffcb0
535154000: board.processor.cores.core: T0 : 0x1191a @_ZN7netlist18get_random_elementEPllP3Rng+80    : c_jr ra                    : IntAlu : 
535155000: board.processor.cores.core: T0 : 0x10d7e @_ZN15annealer_thread3RunEv+220    : c_mv s9, a0                : IntAlu :  D=0x00000000000a81c0
535156000: board.processor.cores.core: T0 : 0x10d80 @_ZN15annealer_thread3RunEv+222    : c_ld a0, 40(s1)            : MemRead :  D=0x0000000000000000 A=0xa8b88
535156000: board.processor.cores.core: T0 : 0x10d82 @_ZN15annealer_thread3RunEv+224    : c_ld a1, 32(s1)            : MemRead :  D=0x0000000000000000 A=0xa8b80
535156000: board.processor.cores.core: T0 : 0x10d84 @_ZN15annealer_thread3RunEv+226    : c_ld a2, 64(s1)            : MemRead :  D=0x0000000000000000 A=0xa8ba0
535156000: board.processor.cores.core: T0 : 0x10d86 @_ZN15annealer_thread3RunEv+228    : c_ld a3, 56(s1)            : MemRead :  D=0x0000000000000000 A=0xa8b98
535156000: board.processor.cores.core: T0 : 0x10d88 @_ZN15annealer_thread3RunEv+230    : c_sub a0, a1               : IntAlu :  D=0x0000000000000000
535156000: board.processor.cores.core: T0 : 0x10d8a @_ZN15annealer_thread3RunEv+232    : srli s0, a0, 3             : IntAlu :  D=0x0000000000000000
535156000: board.processor.cores.core: T0 : 0x10d8e @_ZN15annealer_thread3RunEv+236    : sub a1, a2, a3             : IntAlu :  D=0x0000000000000000
gem5.debug: src/cpu/o3/rename_map.cc:84: gem5::o3::SimpleRenameMap::RenameInfo gem5::o3::SimpleRenameMap::rename(const gem5::RegId&): Assertion `arch_reg.getNumPinnedWrites() == 0'
 failed.
Program aborted at tick 535191000

mayagokhale, 2025-09-30:
@polpetras I'm not sure what you are comparing to. This is a different core than the one that generated a segmentation fault, and the error is an assertion violation not segmentation fault.
It's possible this might be only RISC-V vector and the patch isn't applicable. We haven't looked at it more closely since @erin-le  was investigating it.