gem5/gem5 #1204: RISCV SE mode: pthread_create() leads to a problem with clone3()

DiaZixX, 2024-06-06:
**Describe the bug**
I want to do multi-threading in SE mode on RISCV. To isolate the problem, I'm executing a statically compiled binary that creates a thread (I've disassembled the binary to check that all the labels are present). Using Exec debug-flag, I discover that when clone3() calls ecall, a fatal is raised: `src/sim/syscall_desc.hh:209: fatal: Syscall 435 out of range` and the thread cannot be created.

**Affects version**
I'm on version 23.1.0.0 (The latest)

**gem5 Modifications**
No modification.

**To Reproduce**
Steps to reproduce the behavior. Please assume starting from a clean repository:

1. Compile gem5 with command : scons build/RISCV/gem5.opt
2. Execute the simulation with this python script (using option --nCPU 2):
```python

import m5

m5.util.addToPath("../../")
from m5.objects import *
from caches import *

import argparse

parser = argparse.ArgumentParser(
    description="A simple system with 2-level cache to execute binary riscv code."
)
parser.add_argument(
    "binary",
    default="",
    nargs="?",
    type=str,
    help="Path to the binary to execute."
)

#########################################################################################
#                                        CACHES                                         #
#########################################################################################

# L1 Caches
parser.add_argument(
    "--l1i-size", help="L1 inscrution cache size. Default value: 16kB."
)
parser.add_argument(
    "--l1i-assoc", help="L1 instruction cache associativity. Default value: 8."
)
parser.add_argument(
    "--l1d-size", help="L1 data cache size. Default value: 16kB."
)
parser.add_argument(
    "--l1d-assoc", help="L1 data cache associativity. Default value: 8."
)

# L2 Caches
parser.add_argument(
    "--l2-cache", action="store_true", help="Enable L2 cache"
)
parser.add_argument(
    "--l2-size", help="L2 cache size. Default value: 64kB."
)
parser.add_argument(
    "--l2-assoc", help="L2 cache associativity. Default value: 16"
)
parser.add_argument(
    "--l2-num-cache",
    default=1,
    type=int,
    help="Number of L2 cache. Need to divide the number of CPU. Default value: 1"
)
parser.add_argument(
    "--l2-clusivity",
    help="Clusivity of L2 caches. Default mode: mostly inclusive.",
    choices=["mostly_incl", "mostly_excl"]
)
# parser.add_argument(
#     "--l2-writeback-clean",
#     help="See documentation from Cache.py (writeback_clean parameter).",
#     action="store_true",
# )

# Others
parser.add_argument(
    "--cache-line-size",
    default=64,
    type=int,
    help="Line size in bytes. Default value: 64B"
)

#########################################################################################
#                                        OTHERS                                         #
#########################################################################################

parser.add_argument(
    "--nCPU", default=1, type=int, help="Number of CPU. Default value: 1"
)
parser.add_argument(
    "--mem-size", default="16GB", help="RAM size. Default value: 16GB"
)
parser.add_argument(
    "--mem-type", 
    default="DDR4_2400_8x8", 
    help="Memory component. Default: DDR4_2400_8x8", 
    choices=["DDR3_1600_8x8", "DDR3_2133_8x8", "DDR4_2400_16x4", "DDR4_2400_8x8", "DDR4_2400_4x16", "DDR5_4400_4x8", "DDR5_6400_4x8", "DDR5_8400_4x8"]
) # For more choices, look into src/mem/DRAMInterface.py
parser.add_argument(
    "--clk-speed", default="2GHz", help="System clock speed. Default value: 2GHz"
)
parser.add_argument(
    "--argv", default="", type=str, help="Binary's arguments."
)

options = parser.parse_args()

#########################################################################################
#                                        SYSTEM                                         #
#########################################################################################

nb_cores = options.nCPU

# Verif on L2 Cache
if nb_cores % options.l2_num_cache != 0 :
    fatal("Le nombre de cache L2 ne divise pas le nombre de CPU")

system = System(
    cpu=[RiscvTimingSimpleCPU(cpu_id=i, numThreads=1) for i in range(nb_cores)],
    mem_mode="timing",
    mem_ranges=[AddrRange(options.mem_size)],
    cache_line_size=options.cache_line_size,
    multi_thread=True
)

system.clk_domain = SrcClockDomain()
system.clk_domain.clock = options.clk_speed
system.clk_domain.voltage_domain = VoltageDomain()
system.membus = SystemXBar()
system.membus.badaddr_responder = BadAddr()
system.membus.default = system.membus.badaddr_responder.pio

system.workload = SEWorkload.init_compatible(options.binary)

# The process to be launched
process = Process()
process.cmd = [options.binary, *options.argv.split(" ")]

# Configure all cpus
for cpu in system.cpu:
    # Decoder isa
    cpu.isa = RiscvISA()

    cpu.workload = process
    cpu.createThreads()
    cpu.createInterruptController()

    # Create a memory bus, a coherent crossbar, in this case
    cpu.l2bus = L2XBar()

    # Create an L1 instruction and data cache
    cpu.icache = L1ICache(options)
    cpu.dcache = L1DCache(options)

    # Connect the instruction and data caches to the CPU
    cpu.icache.connectCPU(cpu)
    cpu.dcache.connectCPU(cpu)

    # Hook the CPU ports up to the l2bus
    cpu.icache.connectBus(cpu.l2bus)
    cpu.dcache.connectBus(cpu.l2bus)

    # Create an L2 cache and connect it to the l2bus
    cpu.l2cache = L2Cache()
    cpu.l2cache.connectCPUSideBus(cpu.l2bus)

    # Connect the L2 cache to the L3 bus
    cpu.l2cache.connectMemSideBus(system.membus)

system.system_port = system.membus.cpu_side_ports

system.mem_ctrl = MemCtrl()

# Options for the memory
memory_table = {
    "DDR3_1600_8x8" : DDR3_1600_8x8, 
    "DDR3_2133_8x8" : DDR3_2133_8x8, 
    "DDR4_2400_16x4" : DDR4_2400_16x4, 
    "DDR4_2400_8x8" : DDR4_2400_8x8, 
    "DDR4_2400_4x16" : DDR4_2400_4x16, 
    "DDR5_4400_4x8" : DDR5_4400_4x8, 
    "DDR5_6400_4x8" : DDR5_6400_4x8, 
    "DDR5_8400_4x8" : DDR5_8400_4x8
}
system.mem_ctrl.dram = memory_table[options.mem_type]()
system.mem_ctrl.dram.range = system.mem_ranges[0]
system.mem_ctrl.port = system.membus.mem_side_ports

root = Root(full_system=False, system=system)
m5.instantiate()

print("Beginning simulation!")
exit_event = m5.simulate()
print("Exiting @ tick {} because {}".format(m5.curTick(), exit_event.getCause()))

```
3. The binary is just a main with a call to pthread_create()

**Terminal Output**
```shell
src/arch/riscv/isa.cc:275: info: RVV enabled, VLEN = 256 bits, ELEN = 64 bits
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
src/arch/riscv/isa.cc:275: info: RVV enabled, VLEN = 256 bits, ELEN = 64 bits
system.remote_gdb: Listening for connections on port 7000
src/sim/simulate.cc:199: info: Entering event queue @ 0.  Starting simulation...
src/sim/syscall_emul.cc:85: warn: ignoring syscall set_robust_list(...)
      (further warnings will be suppressed)
src/sim/syscall_emul.hh:1014: warn: readlink() called on '/proc/self/exe' may yield unexpected results in various settings.
      Returning '/home/.../Documents/.../programs/bin/launch_thread'
src/sim/mem_state.cc:448: info: Increasing stack size by one page.
src/sim/syscall_emul.cc:74: warn: ignoring syscall mprotect(...)
src/sim/syscall_emul.cc:85: warn: ignoring syscall rt_sigaction(...)
      (further warnings will be suppressed)
src/sim/syscall_emul.cc:85: warn: ignoring syscall rt_sigprocmask(...)
      (further warnings will be suppressed)
src/sim/syscall_emul.cc:74: warn: ignoring syscall mprotect(...)
src/sim/syscall_desc.hh:209: fatal: Syscall 435 out of range
```

With Exec debug-flag the last 4 lines are :
```shell
291542000: system.cpu: T0 : 0x1d414 @clone3    : c_beqz a0, 20              : IntAlu : 
291543000: system.cpu: T0 : 0x1d416 @clone3+2    : c_beqz a2, 18              : IntAlu : 
291544000: system.cpu: T0 : 0x1d418 @clone3+4    : addi a7, zero, 435         : IntAlu :  D=0x00000000000001b3
291545000: system.cpu: T0 : 0x1d41c @clone3+8    : ecall      
```

**Host Operating System**
Ubuntu 22.04

**Host ISA**
RISC-V

**Compiler used**
Riscv compiler from the toolchain

qqli579, 2024-06-11:
Clone3 is not implemented under the riscv architecture of gem5. You can refer to clone3 under x86 and implement it yourself under the riscv architecture

DiaZixX, 2024-06-13:
In the end, I solved my problem by downgrading the crosscompiler (I'm now on version 11.4.0) to the Ubuntu repos instead of the toolchain available on github.

But if I have time before the end of my internship, I'll try to implement clone3 for Riscv directly.

ivanaamit, 2024-06-14:
I am closing this issue. If I am mistaken and there is a real bug that needs to be fixed, please let me know, and I will reopen it. Thanks.

closed 2024-06-14 by ivanaamit

TonyXinPlus, 2024-09-20:
How do you use the Riscv compiler in the riscv toolchain to cross-compile? My riscv64-unknown-elf-gcc does not recognize the functions pthread_create and pthread_join

DiaZixX, 2024-09-20:
Could you give me the version of your elf ?

TonyXinPlus, 2024-09-21:
OK
riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/opt/riscv64/libexec/gcc/riscv64-unknown-elf/13.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/shz37/yiz75/riscv-gnu-toolchain/gcc/configure --target=riscv64-unknown-elf --prefix=/opt/riscv64 --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion=gc891d8dc23e --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv64/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --disable-multilib --with-abi=lp64 --with-arch=rv64imc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 13.2.0 (gc891d8dc23e)

TonyXinPlus, 2024-09-24:
> Could you give me the version of your elf ?

OK
riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/opt/riscv64/libexec/gcc/riscv64-unknown-elf/13.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/riscv-gnu-toolchain/gcc/configure --target=riscv64-unknown-elf --prefix=/opt/riscv64 --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion=gc891d8dc23e --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv64/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --disable-multilib --with-abi=lp64 --with-arch=rv64imc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 13.2.0 (gc891d8dc23e)

DiaZixX, 2024-09-26:
Try by downgrading to version 11.4 like I did. It worked for me.

The bug is due to the fact that clone3() is a "recent" way to manage multi thread but Gem5 still not handle clone3()

TonyXinPlus, 2024-09-27:
> Try by downgrading to version 11.4 like I did. It worked for me.
> 
> The bug is due to the fact that clone3() is a "recent" way to manage multi thread but Gem5 still not handle clone3()

Can your riscv64-unknown-elf-gcc be configured enable-threads=posix?