gem5/gem5 #1209: Failure to bringup 2+ KVM CPUs on X86

abmerop, 2024-06-08:
**Describe the bug**
When attempting to create a system with 2 or more KVM CPUs, Linux fails to initialize the other CPUs nearly always.  Without looking at the console printout, this is somewhat of a silent error and you end up in a simulation with just 1 CPU.  This can be checked by running `nproc` in the simulated system.

**Affects version**
Develop:  ec5881ec4e (github/develop) arch-arm: avoid using an uninitialized variable use in MMU walks (#1198)

**gem5 Modifications**
None

**To Reproduce**
Steps to reproduce the behavior:

1. Compile gem5 with command

`scons -j$(nproc) build/ALL/gem5.opt`

2. Execute the simulation with...

`build/ALL/gem5.opt configs/example/gem5_library/x86-ubuntu-run-with-kvm.py`

Optionally, the problem can occur faster by removing the switch to timing CPU and verified by printing the number of processors.  I used this to run it in a loop and see how many times I actually got the requested number of CPUs.

```diff
diff --git a/configs/example/gem5_library/x86-ubuntu-run-with-kvm.py b/configs/example/gem5_library/x86-ubuntu-run-with-kvm.py
index b9d035757c..b55465a9cb 100644
--- a/configs/example/gem5_library/x86-ubuntu-run-with-kvm.py
+++ b/configs/example/gem5_library/x86-ubuntu-run-with-kvm.py
@@ -110,12 +110,7 @@ board = X86Board(
 # then, again, call `m5 exit` to terminate the simulation. After simulation
 # has ended you may inspect `m5out/system.pc.com_1.device` to see the echo
 # output.
-command = (
-    "m5 exit;"
-    + "echo 'This is running on Timing CPU cores.';"
-    + "sleep 1;"
-    + "m5 exit;"
-)
+command = 'echo "There are `nproc` CPUs";' + "sleep 1;" + "m5 exit;"

 workload = obtain_resource("x86-ubuntu-18.04-boot", resource_version="2.0.0")
 workload.set_parameter("readfile_contents", command)
@@ -123,12 +118,12 @@ board.set_workload(workload)

 simulator = Simulator(
     board=board,
-    on_exit_event={
-        # Here we want override the default behavior for the first m5 exit
-        # exit event. Instead of exiting the simulator, we just want to
-        # switch the processor. The 2nd m5 exit after will revert to using
-        # default behavior where the simulator run will exit.
-        ExitEvent.EXIT: (func() for func in [processor.switch])
-    },
 )
 simulator.run()
```

**Terminal Output**
During Linux boot:

```shell
...
smpboot: do_boot_cpu failed(-1) to wakeup CPU#1
...
```

If the optional diff above is applied `There are 8 CPUs` is printed on success and `There are 1 CPUs` on failure.

**Expected behavior**
`nproc` running in full system reports >1 CPU.

**Host Operating System**
Ubuntu 22.04

**Host ISA**
X86

**Compiler used**
gcc 13.1.0

**Additional information**
Reverting #1039 fixes the issue for me.

I don't think we should revert the patch entirely as it seems useful.  Something specific to KVM would be better, but I don't have any non-intrusive suggestions.

KVM also brings up the correct number of CPUs on stable and v22.0.0.0 so this is only on develop as of now.

abmerop, 2024-06-09:
I've been experimenting on the side running gem5 on my windows 11 laptop (which now supports nested virtualization in WSL2).  I was experiencing complete hangs using the KVM cpu, but reverting #1039 also fixes that problem (i.e., can't even boot up 1 KVM CPU).

Since I don't know what that change intended to fix, my thought is to create a "KvmThread" class inheriting SimpleThread and override the suspend() method.  I wanted to start a discussion before I write any code though.

abmerop, 2024-06-10:
Adding milestone as I feel this is a significant regression.

powerjg, 2024-06-10:
Two quick things:
1. cc: @rogerchang23424 
2. Related? #1195 

I don't think #1195 is closely related. However, it seems that we are having many different issues with x86 KVM. I don't know if we want to hold up the staging branch for this, but I agree that we need to figure it out before the release.

One other note... we're generally having issues with x86 boot right now. It seems that some change in the past month or so has caused a significant performance regression (maybe it was switching to 24.04?). This also needs to be investigated before we release.

powerjg, 2024-06-11:
Let's revert #1039. It causes CPUs to not come up in atomic and timing as well. All of our daily tests have been failing for 3 weeks because of this change.

rogerchang23424, 2024-06-12:
Thanks for point out the issue. It seems the x86 cpu requires set thread contexts status to suspend at initial stage. I don't understand gem5 x86 much.

https://github.com/gem5/gem5/blob/8fc4d3f793dc7f1837b2507398c423665d8c9908/src/arch/x86/fs_workload.cc#L111-L122

https://github.com/gem5/gem5/blob/8fc4d3f793dc7f1837b2507398c423665d8c9908/src/arch/x86/bare_metal/workload.cc#L46-L67

powerjg, 2024-06-12:
I'm not sure how it works, but we need to more carefully test it if we want to revive #1039. It would be good to explain what problem #1039 was trying to solve as well. If there's a bug that needed to be fixed via #1039, let's focus on that bug. If it's just a generic "improvement" to the code, then maybe it's better to drop it.

abmerop, 2024-06-14:
Bug is fixed by revert, free to close unless you are keeping it open for the discussion

powerjg, 2024-06-14:
Fixed by #1225

closed 2024-06-14 by powerjg