gem5/gem5 #1210: Ruby Protocol: unhandled event type in MI_example-dir.sm

hahaxxz, 2024-06-09:
**Describe the bug**

I am reading MI_example protocol in Ruby system and I found two potential bugs in `MI_example-dir.sm`.

**Case 1**

I found that when the Directory is in the `M_DRDI` state, events `DMA_WRITE` and `DMA_READ` cannot be processed. This is because, during the transition `(M_DRD, PUTX, M_DRDI)`, the DMA data has already been sent to the DMA by the `drp_sendDMAData` action. Consequently, when the DMA is in the `READY` state and send `DMA_READ` or `DMA_WRITE` requests to the Directory, the Directory is unable to handle these requests while it is in the `M_DRDI` state.

I think the following transition

```
transition({IM, MI, ID, ID_W}, {DMA_READ, DMA_WRITE} ) {
  y_recycleDMARequestQueue;
}
```

should be modified to 

```
transition({IM, MI, ID, ID_W, M_DRDI}, {DMA_READ, DMA_WRITE} ) {
  y_recycleDMARequestQueue;
}
```

**Case 2**

Directory cannot handle `PUTX_NotOwner` event when it is in `{M_DWR, M_DRD, M_DRDI, M_DWRI}` state.

The following transition should be added to `MI_example-dir.sm`.

```
 transition({M_DWR, M_DRD, M_DRDI, M_DWRI}, PUTX_NotOwner) {
    b_sendWriteBackNack;
    i_popIncomingRequestQueue;
  }
}
```

I am extremely grateful to have experts to answer the two questions mentioned above for me.

**Affects version**
the latest commit in stable branch (Change-Id: I33bca345d985618e3fca62e9ddd5bcc3ad8226a3)

**To Reproduce**
I did not construct a test case to trigger this bug; it was discovered while reviewing the code. Constructing such a test case would be challenging for me.

powerjg, 2024-06-10:
I wouldn't be surprised if these were bugs. DMA is not heavily used, and MI_example is not often tested. It's really just an "example" protocol, and it's not meant for use in realistic simulations.

I would hesitate to make these changes without example of protocol failures. I wonder if you could get this to trigger with DMA accesses in the RubyRandomTester?

hahaxxz, 2024-06-10:
> I wouldn't be surprised if these were bugs. DMA is not heavily used, and MI_example is not often tested. It's really just an "example" protocol, and it's not meant for use in realistic simulations. [...]

Sure. I will try to make them triggered.

hahaxxz, 2024-06-12:
@powerjg 

Finally, two cases are triggered by RubyRandomTester.

Case 2 was first triggered after just 1000 DMA read requests.

```
./build/X86_MI_example/gem5.opt ./configs/example/ruby_random_test.py  -n 2 --maxloads 100000 --num-dmas 1  --mem-size 6400000
gem5 Simulator System.  https://www.gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version 23.1.0.0
gem5 compiled Jun 12 2024 15:45:07
gem5 started Jun 12 2024 15:47:24
gem5 executing on lenovo, pid 615061
command line: ./build/X86_MI_example/gem5.opt ./configs/example/ruby_random_test.py -n 2 --maxloads 100000 --num-dmas 1 --mem-size 6400000

...

system.dma_devices: completed 1000 read, 597 write, 0 atomic accesses @75689

build/X86_MI_example/mem/ruby/protocol/Directory_Transitions.cc:269: panic: Invalid transition
system.ruby.dir_cntrl0 time: 79088 addr: 0x3c0 event: PUTX_NotOwner state: M_DRD

Memory Usage: 133056 KBytes
Program aborted at tick 79088
--- BEGIN LIBC BACKTRACE ---
./build/X86_MI_example/gem5.opt(_ZN4gem515print_backtraceEv+0x30)[0x55c929f83b90]
./build/X86_MI_example/gem5.opt(_ZN4gem512abortHandlerEi+0x4c)[0x55c929fa859c]
/lib/x86_64-linux-gnu/libc.so.6(+0x42520)[0x7711a9042520]
...
```

Then I fixed it as mentioned above:

```
  // MI_example-dir.sm
  
  + transition({M_DWR, M_DRD, M_DRDI, M_DWRI}, PUTX_NotOwner) {
  +  b_sendWriteBackNack;
  +  i_popIncomingRequestQueue;
  +}
```

After fixing Case2, Case1 was triggered after 184,000 DMA read requests.

```
./build/X86_MI_example/gem5.opt ./configs/example/ruby_random_test.py  -n 2 --maxloads 100000000 --num-dmas 1  --mem-size 6400000
gem5 Simulator System.  https://www.gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version 23.1.0.0
gem5 compiled Jun 12 2024 15:48:49
gem5 started Jun 12 2024 15:52:23
gem5 executing on lenovo, pid 615478
command line: ./build/X86_MI_example/gem5.opt ./configs/example/ruby_random_test.py -n 2 --maxloads 100000000 --num-dmas 1 --mem-size 6400000

system.dma_devices: completed 1000 read, 597 write, 0 atomic accesses @75689
system.dma_devices: completed 2000 read, 1153 write, 0 atomic accesses @148610
system.dma_devices: completed 3000 read, 1717 write, 0 atomic accesses @221879
system.dma_devices: completed 4000 read, 2278 write, 0 atomic accesses @293690
...
system.dma_devices: completed 178000 read, 98451 write, 0 atomic accesses @12881460
system.dma_devices: completed 179000 read, 98951 write, 0 atomic accesses @12949551
system.dma_devices: completed 180000 read, 99536 write, 0 atomic accesses @13023101
system.dma_devices: completed 181000 read, 100040 write, 0 atomic accesses @13092960
system.dma_devices: completed 182000 read, 100602 write, 0 atomic accesses @13163898
system.dma_devices: completed 183000 read, 101202 write, 0 atomic accesses @13237218
system.dma_devices: completed 184000 read, 101736 write, 0 atomic accesses @13308749
build/X86_MI_example/mem/ruby/protocol/Directory_Transitions.cc:273: panic: Invalid transition
[... 61 lines cut ...]
            stats.numReads++;
```

```
src/cpu/testers/rubytest/Check.cc:316
    } else if (m_status == ruby::TesterStatus_Check_Pending) {
        DPRINTF(RubyTest, "Check callback\n");
        // Perform load/check
        // for (int byte_number=0; byte_number<CHECK_SIZE; byte_number++) {
        //     if (uint8_t(m_value + byte_number) != data->getByte(byte_number)) {
        //         panic("Action/check failure: proc: %d address: %#x data: %s "
        //               "byte_number: %d m_value+byte_number: %d byte: %d %s"
        //               "Time: %d\n",
        //               proc, address, data, byte_number,
        //               (int)m_value + byte_number,
        //               (int)data->getByte(byte_number), *this, curTime);
        //     }
        // }
        DPRINTF(RubyTest, "Action/check success\n");
        debugPrint();
```

The python script is attached. 

[ruby_random_test.zip](https://github.com/user-attachments/files/15804424/ruby_random_test.zip)

powerjg, 2024-06-12:
Awesome! Thanks for digging into this and producing the errors!

Feel free to open a PR with these fixes :)

BobbyRBruce, 2024-06-17:
Fiied by #1236

closed 2024-06-17 by BobbyRBruce