gem5/gem5 #1219: square can not run under the gpufs

Yongze-zzz, 2024-06-11:
```
docker pull ghcr.io/gem5-test/gcn-gpu:v24-0
docker run --net host -itd --name gcn3-gpu -v /home/yanyz/db/gem5space/:/root/gem5space  ghcr.io/gem5/gcn-gpu:v23-0 bash
//enter to docker container
docker exec -w /root/gem5space -it gcn3-gpu bash
the file structure likes below:
gem5space
        |--gem5(stable branch)
        |--gem5-resources
cd gem5
scons build/GCN3_X86/gem5.opt -j 8
scons build/VEGA_X86/gem5.opt -j 8

//test quare
cd gem5-resources/src/gpu/square
make
gem5/build/GCN3_X86/gem5.opt gem5/configs/example/apu_se.py -n 3 -c gem5-resources/src/gpu/square/bin/square
//output
info: check result
PASSED!
Ticks: 139906310500
Exiting because  exiting with last active thread context

//test gpufs square
vim square.cpp
--------------------------
#include <stdio.h>
#include "hip/hip_runtime.h"
#include <pthread.h>
#define CHECK(cmd) \
{\
    hipError_t error  = cmd;\
    if (error != hipSuccess) { \
      fprintf(stderr, "error: '%s'(%d) at %s:%d\n", hipGetErrorString(error), error,__FILE__, __LINE__); \
    exit(EXIT_FAILURE);\
    }\
}

/*
 *  Thread function for printing "Hello, World!".
 */
void* print_hello(void* arg) {
    int thread_id = *(int*)arg;
    printf("Hello, World! from thread %d\n", thread_id);
    pthread_exit(NULL);
}

/*
 * Square each element in the array A and write to array C.
 */
template <typename T>
__global__ void
vector_square(T *C_d, const T *A_d, size_t N)
{
    size_t offset = (hipBlockIdx_x * hipBlockDim_x + hipThreadIdx_x);
    size_t stride = hipBlockDim_x * hipGridDim_x ;

    for (size_t i=offset; i<N; i+=stride) {
        C_d[i] = A_d[i] * A_d[i];
    }
}

int main(int argc, char *argv[])
{
    float *A_h, *C_h;
    size_t N = 1000000;
    size_t Nbytes = N * sizeof(float);

    hipDeviceProp_t props;
    CHECK(hipGetDeviceProperties(&props, 0/*deviceID*/));
[... 181 lines cut ...]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x2a8748)[0x7ff14033d748]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x8dd8)[0x7ff140112f48]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x8fb)[0x7ff14025fe4b]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyFunction_Vectorcall+0x94)[0x7ff14033d124]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x74d6d)[0x7ff140109d6d]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x7d86)[0x7ff140111ef6]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x8006b)[0x7ff14011506b]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x74d6d)[0x7ff140109d6d]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x12fd)[0x7ff14010b46d]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x8fb)[0x7ff14025fe4b]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(PyEval_EvalCodeEx+0x42)[0x7ff1402601d2]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(PyEval_EvalCode+0x1f)[0x7ff1402605bf]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x1cfc01)[0x7ff140264c01]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x25f547)[0x7ff1402f4547]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x74d6d)[0x7ff140109d6d]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x12fd)[0x7ff14010b46d]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x8006b)[0x7ff14011506b]
/lib/x86_64-linux-gnu/libpython3.8.so.1.0(PyVectorcall_Call+0x60)[0x7ff14033d840]
/root/gem5space/gem5/build/VEGA_X86/gem5.opt(+0x1946e84)[0x55a8ef92fe84]
/root/gem5space/gem5/build/VEGA_X86/gem5.opt(main+0x213)[0x55a8eecf16d3]
--- END LIBC BACKTRACE ---
For more info on how to address this issue, please visit https://www.gem5.org/documentation/general_docs/common-errors/ 

Aborted (core dumped)
```

Yongze-zzz, 2024-06-11:
/root/gem5space/gem5/build/GCN3_X86/gem5.opt /root/gem5space/gem5/configs/example/apu_se.py -n 3 -c /root/gem5space/gem5-resources/src/matrix-multiply-omp/matrix-omp. I try to run multi thread with apu_se mode, but  it report :Thread creation failed: Resource temporarily unavailable. I just need a mode which can run cpu multi-thread and gpu kernel function at the same time.

mattsinc, 2024-06-11:
I have never tried running multi-threaded CPU programs so unfortunately I cannot provide any advice there -- maybe @BobbyRBruce or @powerjg can.

In terms of the GPU portion, I am confused by what you posted.  I see the following:

...
info: check result
PASSED!
Ticks: 139906310500
Exiting because  exiting with last active thread context

This seems to imply that square passed.  However, I later see:

...
src/mem/ruby/system/Sequencer.cc:240: panic: Possible Deadlock detected. Aborting!
 version: 0 request.paddr: 0x70000 m_readRequestTable: 1 current time: 250000000 issue_time: 0 difference: 250000000

So I'm confused how you could have had the test both pass and fail.  Are you including two separate runs -- one of which passed and one of which failed?

My guess from trying to parse the text in your message is that a) square is working fine when run in the provided format in SE mode, b) you did not try square in the provided format in FS mode, and c) when you added pthreads to it for a CPU portion and ran in FS mode it has the error you later posted?  If so, I think this is not a problem with the GPU support, but rather a question for the CPU folks.  But let @abmerop and I know if we're missing something.

Yongze-zzz, 2024-06-11:
I did two separate runs, first executing square using apu_se.py mode, and it passed. But when I added a separate pthread for square, and executed it in runfs.py mode, it failed.
Indeed, as you might guess, a) square works fine when running in SE mode in the provided format. b) square did not work when tried in FS mode in the provided format, and c) when adding pthreads for the CPU portion and running in FS mode, it gave the error posted later

Also, I tried running square in FS mode in the provided format for the last few hours, and it gave the same problem as adding pthreads

abmerop, 2024-06-12:
Timing CPUs do not work with GPUFS due to the fundamental design of Ruby (TimingSimpleCPU / O3CPU).  Please use the top level script `configs/example/gpufs/vega10_kvm.py` which sets the known working parameters for you.

Yongze-zzz, 2024-06-13:
Thank you ! I run `/root/gem5space/gem5/build/VEGA_X86/gem5.opt /root/gem5space/gem5/configs/example/gpufs/vega10_kvm.py --disk-image /root/gem5space/gem5-resources/src/x86-ubuntu-gpu-ml/disk-image/x86-ubuntu-gpu-ml --kernel /root/gem5space/gem5-resources/src/x86-ubuntu-gpu-ml/vmlinux-gpu-ml --gpu-mmio-trace /root/gem5space/gem5-resources/src/gpu-fs/vega_mmio.log --mem-size=8GB -a /root/gem5space/gem5-resources/src/gpu/square/bin/square`.

Then it report src/cpu/kvm/perfevent.cc:217: panic: PerfKvmCounter::attach failed (1).

I try to run "echo "-1" | sudo tee /proc/sys/kernel/perf_event_paranoid".  But it stilll report panic: PerfKvmCounter::attach failed (1).

abmerop, 2024-06-13:
You could try adding `--no-kvm-perf` to the command.  If KVM does not work at all, there is also a `configs/example/gpufs/vega10_atomic.py`.  With atomic CPU however, your simulation is going to take hours instead of minutes and you might have to rely on things like checkpointing

Yongze-zzz, 2024-06-14:
`src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
src/base/stats/storage.hh:278: warn: Bucket size (10000) does not divide range [1:1.6e+06] into equal-sized buckets. Rounding up.
[... cut ...]
src/dev/x86/pc.cc:117: warn: Don't know what interrupt to clear for console.
src/arch/x86/interrupts.cc:530: hack: Assuming logical destinations are 1 << id.
src/dev/amdgpu/amdgpu_vm.hh:254: warn: Accessing unsupported MMIO aperture! Assuming NBIO
Exiting @ tick 6061858002000 because m5_exit instruction encountered
src/cpu/kvm/base.cc:591: hack: Pretending totalOps is equivalent to totalInsts()`

I try to add `--no-kvm-perf` , then run square. The script seems like execute successful, but there are no application's output like "printf ("AMDGPU PASSED!\n");"

abmerop, 2024-06-14:
You will need to build square for ROCm 6.1 (the version on the disk image).  There is a docker provided to help with that here: `docker run ghcr.io/gem5/gpu-fs:latest`

ivanaamit, 2024-06-14:
This seems to be a discussion rather than a gem5 bug. Please let me know if that is the case so that I can convert it into discussion. Thanks.