gem5/gem5 #1227: subtraction of two long numbers simulated wrongly

karineek, 2024-06-12:
**Describe the bug**
Subtraction of two long numbers simulated wrongly, giving 0-2 = 0, instead of -2.

**Affects version**
State which version of gem5 this bug was found in. If on the develop branch state the Commit revision ID you are working.

I have tested two versions:

- gem5 version 23.1.0.0; gem5 compiled Jun 11 2024 11:23:46
- gem5 version 23.0.0.1--SSBSE Challeng Track; gem5 compiled Aug 13 2023 00:24:12

**gem5 Modifications**
If you have modified gem5 in some way please state, to the best of your ability, how it has been modified.
No modifications. I used the script hello-custom-binary-Ex.py modified from this script https://github.com/BobbyRBruce/gem5-ssbse-challenge-2023, to allow input. The script is available here: https://github.com/karineek/SearchGEM5/blob/main/hello-custom-binary-Ex.py.

**To Reproduce**
Steps to reproduce the behavior. Please assume starting from a clean repository:

1. Built gem5 from source:
```
git clone https://github.com/gem5/gem5.git
cd gem5/
pip install -r requirements.txt
scons build/X86/gem5.opt -j 4
```

2. Compile the code:
```
gcc-11 -O3 test_missim_01.c -o test_missim_01.c.o
```
and create a text file to contain input:
```
:~$ more test_missim_01.txt
/home/ubuntu/test_missim_01.c.o
0
```

3. Run:
```
./gem5/build/X86/gem5.opt ssbse-challenge-examples/hello-custom-binary-Ex.py --isa X86 --input test_missim_01.txt
```

The code is:
```C
#include <stdio.h>
#include <stdlib.h>
int main(int argc, char *argv[]) {
  if (argc != 2) {
    printf("Usage: %s <value>\n", argv[0]);
    return 1;
  }
  int x = atoi(argv[1]);
  long double result  = (long double)(x) - 2;
  long double result2 = (long double)(x) - 1;
  printf(">> Result: x - 2 = %.2Lf\n", result);
  printf(">> Result: x - 1 = %.2Lf\n", result2);
  return 0;
}
```
The result of the first print is wrong:
```
gem5 Simulator System.  https://www.gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version 23.1.0.0
gem5 compiled Jun 11 2024 11:23:46
gem5 started Jun 12 2024 11:23:13
gem5 executing on fuzzing-05, pid 704980
command line: /home/ubuntu/gem5-Jun-11/gem5/build/X86/gem5.opt /home/ubuntu/gem5-ssbse-challenge-2023/ssbse-challenge-examples/hello-custom-binary-Ex.py --isa X86 --input test_missim_01.txt

Global frequency set at 1000000000000 ticks per second
warn: No dot file generated. Please install pydot to generate the dot file and pdf.
src/mem/dram_interface.cc:690: warn: DRAM device capacity (8192 Mbytes) does not match the address range assigned (32 Mbytes)
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
board.remote_gdb: Listening for connections on port 7000
src/sim/simulate.cc:199: info: Entering event queue @ 0.  Starting simulation...
src/sim/mem_state.cc:448: info: Increasing stack size by one page.
src/sim/syscall_emul.cc:74: warn: ignoring syscall mprotect(...)
src/sim/syscall_emul.cc:74: warn: ignoring syscall mprotect(...)
src/sim/syscall_emul.cc:74: warn: ignoring syscall mprotect(...)
>> Result: x - 2 = -0.00
>> Result: x - 1 = -1.00
Exiting @ tick 11927511549 because exiting with last active thread context.
```

**Terminal Output**
If applicable, add the terminal output here. If long, only include the relevant lines.
Please put the terminal output in code blocks. I.e.:

```shell
:~$ ./test_missim_01.c.o 0
>> Result: x - 2 = -2.00
>> Result: x - 1 = -1.00
```

**Expected behavior**
To compute 0 - 2 as -2.

**Host Operating System**
Ubuntu 20.04, x86_64-pc-linux-gnu

**Host ISA**
X86

**Compiler used**
GCC-11 with scons:
```
gcc-11 (Ubuntu 11.4.0-2ubuntu1~20.04) 11.4.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
and scons:
```
SCons by Steven Knight et al.:
	script: v3.1.2.bee7caf9defd6e108fc2998a2520ddb36a967691, 2019-12-17 02:07:09, by bdeegan on octodog
	engine: v3.1.2.bee7caf9defd6e108fc2998a2520ddb36a967691, 2019-12-17 02:07:09, by bdeegan on octodog
	engine path: ['/usr/lib/scons/SCons']
Copyright (c) 2001 - 2019 The SCons Foundation
```

**Additional information**
Add any other information which does not fit in the previous sections but may be of use in fixing this bug.

karineek, 2024-06-12:
I also found another example atof instead of atoi:

```
#include <stdio.h>
#include <stdlib.h>
#include <float.h>

int main(int argc, char *argv[]) {
    if (argc != 2) {
        printf("Usage: %s <value>\n", argv[0]);
        return 1;
    }

    long double x = atof(argv[1]);
    long double result = 2 * x - 2;

    printf("Result: %Lf\n", result);

    return 0;
}
```
where the bug disappeared when changing -2 to -1, in long double result = 2 * x - 2;.

andysan, 2024-06-13:
If memory serves me right, `long double` is treated as an 80-bit float on x86. My suspicion is that this bug is related to gem5's x87 implementation which has never been reliable (I wasted a great deal of time fixing some of the most glaring issues many years ago when I was a PhD student). As far as I know, it still implements 80-bit floating point arithmetic in x87 using 64-bit double precision and then expanding that to the 80-bit format when written to memory. 

It would be helpful if you could provide the disassembly if the main function in the bug report, that would let whoever picks this up narrow it down to specific instructions.

karineek, 2024-06-13:
Thanks!

If I build it with -O0 then the simulation runs correctly; however, with -O1 (and -O2, -O3...) the bug appears also with -g. Here are the two versions:

O0:
```
0000000000001169 <main>:
    1169:	f3 0f 1e fa          	endbr64 
    116d:	55                   	push   rbp
    116e:	48 89 e5             	mov    rbp,rsp
    1171:	48 83 ec 40          	sub    rsp,0x40
    1175:	89 7d cc             	mov    DWORD PTR [rbp-0x34],edi
    1178:	48 89 75 c0          	mov    QWORD PTR [rbp-0x40],rsi
    117c:	83 7d cc 02          	cmp    DWORD PTR [rbp-0x34],0x2
    1180:	74 25                	je     11a7 <main+0x3e>
    1182:	48 8b 45 c0          	mov    rax,QWORD PTR [rbp-0x40]
    1186:	48 8b 00             	mov    rax,QWORD PTR [rax]
    1189:	48 89 c6             	mov    rsi,rax
    118c:	48 8d 05 7d 0e 00 00 	lea    rax,[rip+0xe7d]        # 2010 <_IO_stdin_used+0x10>
    1193:	48 89 c7             	mov    rdi,rax
    1196:	b8 00 00 00 00       	mov    eax,0x0
    119b:	e8 c0 fe ff ff       	call   1060 <printf@plt>
    11a0:	b8 01 00 00 00       	mov    eax,0x1
    11a5:	eb 6f                	jmp    1216 <main+0xad>
    11a7:	48 8b 45 c0          	mov    rax,QWORD PTR [rbp-0x40]
    11ab:	48 83 c0 08          	add    rax,0x8
    11af:	48 8b 00             	mov    rax,QWORD PTR [rax]
    11b2:	48 89 c7             	mov    rdi,rax
    11b5:	e8 b6 fe ff ff       	call   1070 <atoi@plt>
    11ba:	89 45 dc             	mov    DWORD PTR [rbp-0x24],eax
    11bd:	db 45 dc             	fild   DWORD PTR [rbp-0x24]
    11c0:	db 2d 9a 0e 00 00    	fld    TBYTE PTR [rip+0xe9a]        # 2060 <_IO_stdin_used+0x60>
    11c6:	de e9                	fsubp  st(1),st
    11c8:	db 7d e0             	fstp   TBYTE PTR [rbp-0x20]
    11cb:	db 45 dc             	fild   DWORD PTR [rbp-0x24]
    11ce:	d9 e8                	fld1   
    11d0:	de e9                	fsubp  st(1),st
    11d2:	db 7d f0             	fstp   TBYTE PTR [rbp-0x10]
    11d5:	ff 75 e8             	push   QWORD PTR [rbp-0x18]
    11d8:	ff 75 e0             	push   QWORD PTR [rbp-0x20]
    11db:	48 8d 05 41 0e 00 00 	lea    rax,[rip+0xe41]        # 2023 <_IO_stdin_used+0x23>
    11e2:	48 89 c7             	mov    rdi,rax
    11e5:	b8 00 00 00 00       	mov    eax,0x0
    11ea:	e8 71 fe ff ff       	call   1060 <printf@plt>
    11ef:	48 83 c4 10          	add    rsp,0x10
    11f3:	ff 75 f8             	push   QWORD PTR [rbp-0x8]
    11f6:	ff 75 f0             	push   QWORD PTR [rbp-0x10]
    11f9:	48 8d 05 3d 0e 00 00 	lea    rax,[rip+0xe3d]        # 203d <_IO_stdin_used+0x3d>
    1200:	48 89 c7             	mov    rdi,rax
    1203:	b8 00 00 00 00       	mov    eax,0x0
    1208:	e8 53 fe ff ff       	call   1060 <printf@plt>
    120d:	48 83 c4 10          	add    rsp,0x10
    1211:	b8 00 00 00 00       	mov    eax,0x0
    1216:	c9                   	leave  
    1217:	c3                   	ret    
    1218:	0f 1f 84 00 00 00 00 	nop    DWORD PTR [rax+rax*1+0x0]
    121f:	00 
```

and O1:
```
0000000000001169 <main>:
    1169:	f3 0f 1e fa          	endbr64 
    116d:	48 83 ec 18          	sub    rsp,0x18
    1171:	83 ff 02             	cmp    edi,0x2
    1174:	74 23                	je     1199 <main+0x30>
    1176:	48 8b 16             	mov    rdx,QWORD PTR [rsi]
    1179:	48 8d 35 84 0e 00 00 	lea    rsi,[rip+0xe84]        # 2004 <_IO_stdin_used+0x4>
    1180:	bf 01 00 00 00       	mov    edi,0x1
    1185:	b8 00 00 00 00       	mov    eax,0x0
    118a:	e8 e1 fe ff ff       	call   1070 <__printf_chk@plt>
    118f:	b8 01 00 00 00       	mov    eax,0x1
    1194:	48 83 c4 18          	add    rsp,0x18
    1198:	c3                   	ret    
    1199:	48 8b 7e 08          	mov    rdi,QWORD PTR [rsi+0x8]
    119d:	ba 0a 00 00 00       	mov    edx,0xa
    11a2:	be 00 00 00 00       	mov    esi,0x0
    11a7:	e8 b4 fe ff ff       	call   1060 <strtol@plt>
    11ac:	89 04 24             	mov    DWORD PTR [rsp],eax
    11af:	db 04 24             	fild   DWORD PTR [rsp]
    11b2:	d9 c0                	fld    st(0)
    11b4:	db 3c 24             	fstp   TBYTE PTR [rsp]
    11b7:	d8 25 8f 0e 00 00    	fsub   DWORD PTR [rip+0xe8f]        # 204c <_IO_stdin_used+0x4c>
    11bd:	48 8d 64 24 f0       	lea    rsp,[rsp-0x10]
    11c2:	db 3c 24             	fstp   TBYTE PTR [rsp]
    11c5:	48 8d 35 4b 0e 00 00 	lea    rsi,[rip+0xe4b]        # 2017 <_IO_stdin_used+0x17>
    11cc:	bf 01 00 00 00       	mov    edi,0x1
    11d1:	b8 00 00 00 00       	mov    eax,0x0
    11d6:	e8 95 fe ff ff       	call   1070 <__printf_chk@plt>
    11db:	d9 e8                	fld1   
    11dd:	db 6c 24 10          	fld    TBYTE PTR [rsp+0x10]
    11e1:	de e1                	fsubrp st(1),st
    11e3:	48 8d 64 24 f0       	lea    rsp,[rsp-0x10]
    11e8:	db 3c 24             	fstp   TBYTE PTR [rsp]
    11eb:	48 8d 35 3f 0e 00 00 	lea    rsi,[rip+0xe3f]        # 2031 <_IO_stdin_used+0x31>
    11f2:	bf 01 00 00 00       	mov    edi,0x1
    11f7:	b8 00 00 00 00       	mov    eax,0x0
    11fc:	e8 6f fe ff ff       	call   1070 <__printf_chk@plt>
    1201:	48 83 c4 20          	add    rsp,0x20
    1205:	b8 00 00 00 00       	mov    eax,0x0
    120a:	eb 88                	jmp    1194 <main+0x2b>
    120c:	0f 1f 40 00          	nop    DWORD PTR [rax+0x0]
```
From the first 3 lines, it seems like an address that is treated differently with the sub command.

andysan, 2024-06-13:
I'm nor an expert in x86 assembly, but I think the main difference is that the `fsub` in the O0 example operates on the x87 stack (confusingly, these are actually registers) only while the O1 example seems to use one operand from the stack and one from memory. I wouldn't be surprised if that's where it goes wrong. This seems plausible since gem5 internally represents floating point registers as 64b doubles but it supports 80b floats in-memory. I wouldn't be surprised if you have a similar bug in any instruction that takes one 80b operand from a register (x87 stack) and the other one from memory.