gem5/gem5 #1235: The riscv virtual address translation process may not work under non 4KiB page sizes

qqli579, 2024-06-13:
**Describe the bug**
The process of converting virtual addresses into physical addresses under the riscv architecture is as follows:
```
1. Let a be satp:ppn × PAGESIZE, and let i = LEVELS − 1. (For Sv32, PAGESIZE=2^12 and
LEVELS=2.)
2. Let pte be the value of the PTE at address a+va:vpn[i]×PTESIZE. (For Sv32, PTESIZE=4.)
If accessing pte violates a PMA or PMP check, raise an access exception.
3. If pte:v = 0, or if pte:r = 0 and pte:w = 1, stop and raise a page-fault exception.
4. Otherwise, the PTE is valid. If pte:r = 1 or pte:x = 1, go to step 5. Otherwise, this PTE is a
pointer to the next level of the page table. Let i = i − 1. If i < 0, stop and raise a page-fault
exception. Otherwise, let a = pte:ppn × PAGESIZE and go to step 2.
5. A leaf PTE has been found. Determine if the requested memory access is allowed by the
pte:r, pte:w, pte:x, and pte:u bits, given the current privilege mode and the value of the SUM
and MXR fields of the mstatus register. If not, stop and raise a page-fault exception.
6. If i > 0 and pa:ppn[i − 1 : 0] 6= 0, this is a misaligned superpage; stop and raise a page-fault
exception.
7. If pte:a = 0, or if the memory access is a store and pte:d = 0, either raise a page-fault
exception or:
• Set pte:a to 1 and, if the memory access is a store, also set pte:d to 1.
• If this access violates a PMA or PMP check, raise an access exception.
• This update and the loading of pte in step 2 must be atomic; in particular, no intervening
store to the PTE may be perceived to have occurred in-between.
8. The translation is successful. The translated physical address is given as follows:
• pa.pgoff = va.pgoff.
• If i > 0, then this is a superpage translation and pa:ppn[i − 1 : 0] = va:vpn[i − 1 : 0].
• pa:ppn[LEVELS − 1 : i] = pte:ppn[LEVELS − 1 : i]
```
In step 8, if i > 0, then：
```
 If i > 0, then this is a superpage translation and pa:ppn[i − 1 : 0] = va:vpn[i − 1 : 0].
```
But the implementation in the doTranslate function in src/arch/riscv/tlb.cc of gem5 is like this:
```
Addr paddr = e->paddr << PageShift | (vaddr & mask(e->logBytes));
```
There is no problem when the page size is 4kiB, but if the page size is not 4kiB, this implementation will have problems.

**Affects version**
gem5 version 23.1.0.0

**gem5 Modifications**
There are two places in gem5 that need to be modified.One is 
```
Addr paddr = e->paddr << PageShift | (vaddr & mask(e->logBytes));
```
in the doTranslate() function in the src/arch/riscv/tlb.cc file, which should be modified as follows:
```
Addr paddr = ((e->paddr >> (e->logBytes - PageShift)) << e->logBytes) | (vaddr & mask(e->logBytes));
```
Another place is that the translateWithTLB function in the file needs to be modified as follows:
```
Addr
TLB::translateWithTLB(Addr vaddr, uint16_t asid, BaseMMU::Mode mode)
{
    TlbEntry *e = lookup(vaddr, asid, mode, false);
    assert(e != nullptr);
    return ((e->paddr >> (e->logBytes - PageShift)) << e->logBytes) | (vaddr & mask(e->logBytes));
}
```

**To Reproduce**
Steps to reproduce the behavior. Please assume starting from a clean repository:

1. Compile gem5 with command ...
2. Execute the simulation with...

If writing code, or a terminal command, use code blocks. Either an inline code block, `scons build/ALL/gem5.opt` (enclosed in two '`') or a multi-line codeblock:

```python
int x=2
int y=3
print(x+y)
```

If possible, please include the Python configuration script used and state clearly any parameters passed.

**Terminal Output**
If applicable, add the terminal output here. If long, only include the relevant lines.
Please put the terminal output in code blocks. I.e.:

```shell
#Terminal output here#
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Host Operating System**
Ubuntu 22.04,

**Host ISA**
X86,

**Compiler used**
State which compiler was used to compile gem5. Please include the compiler version.

**Additional information**
Add any other information which does not fit in the previous sections but may be of use in fixing this bug.

hnpl, 2024-06-14:
@qqli579 Thanks for reporting this problem. Are you working on a fix for this?

qqli579, 2024-06-14:
I modified it locally and ran some simple tests without encountering any problems.

hnpl, 2024-06-14:
Do you plan on making a PR for this soon? I think supporting big pages is important.

Regarding testing, have you verified that the translation is correct when there is a big page?

powerjg, 2024-06-19:
I think there's some deeper problems with the address translation in the RISC-V TLB. I'm currently digging into this.

I have encountered an error when running a newer kernel (6.6) where the following happens:
1. TLB miss
2. PTW finds a valid PTE
3. The TLB entry is inserted
4. The subsequent lookup fails

The lookup after inserting the new entry seems to fail because the ASID doesn't match. 

This code seems odd to me. It seems to me that "somebody beat us to" inserting into the TLB shouldn't happen. If this code *is* right, we should probably update the ASID too?
https://github.com/gem5/gem5/blob/e88f0944e309e04800eefe4f03ce23e35a3c7fe5/src/arch/riscv/tlb.cc#L148-L155

I don't fully understand what's going on, yet. But I'm looking into it. My goal is to get this solved before the 24.0 release.

If anyone has pointers or ideas, I would love to hear them.

powerjg, 2024-06-20:
After continuing to dig on this, I believe that the suggested fix is just masking (pun intended) the underlying issue that we were not using the right bits to insert into the TLB. I have a draft PR that I've put together that might fix things. I'm still testing it as of now, though. #1264

PR #1264 (2024-06-20): arch-riscv: Fix TLB lookup with vaddrs
Previously, all of the TLB lookup/insert functions were using the full virtual addresses even though the variables in the functions said "vpn." This change explicitly converts the virtual address to the VPN without any least significant zeros for the offset. I.e., vpn >> page_size.

The main bug solved in this changeset is the asid was |'d with the upper bits of the virtual address, but sometimes there were all 1's. Therefore, you could get a TLB hit even if the ASID was different. Interestingly, the page that seemed to cause these issues was a 1 GiB page [... cut]
commit 013f773d (2024-06-20): arch-riscv: Fix TLB lookup with vaddrs (#1264)

Previously, all of the TLB lookup/insert functions were using the full
virtual addresses even though the variables in the functions said "vpn."
This change explicitly converts the virtual address to the VPN without
any least significant zeros for the offset. I.e., vpn >> page_size.

The main bug solved in this changeset is the asid was |'d with the upper
bits of the virtual address, but sometimes there were all 1's.
Therefore, you could get a TLB hit even if the ASID was different.
Interestingly, the page that seemed to cause these issues was a 1 [... cut]
closed 2024-06-28 by BobbyRBruce