gem5/gem5 #1239: Cannot build ALL architectures without SystemC (USE_SYSTEMC=0) in v23.1

voertler, 2024-06-14:
I have Problems how to build gem5 with KConfig for all architectures, both for gem5.opt and libgem5_opt.so

Before v23.1 we could build gem5 and the library version just like this:

```scons-3 --with-cxx-config --without-tcmalloc USE_SYSTEMC=0 --verbose build/ALL/gem5.opt ```
```scons-3 --with-cxx-config --without-python --without-tcmalloc USE_SYSTEMC=0 --verbose build/ALL/libgem5_opt.so```

However this leads to compile errors as described in https://github.com/gem5/gem5/issues/591 when using v23.1

So I tried as was proposed for ARM architectures in the ticket for ALL architectures

```scons-3 setconfig build/ALL USE_SYSTEMC=n```
```scons-3 --with-cxx-config --without-tcmalloc --verbose  build/gem5.opt ```

However this didn't work as the gem5.opt is missing all ISA specific Simobjects.

When doing the official KConfig way from the documentation, the options --with-cxx-config --without-tcmalloc --verbose are not supported , Also I find no documentation how to build libgem5 with KConfig

```scons-3 defconfig build_gem5 build_opts/ALL```
```scons-3 setconfig build_gem5 USE_SYSTEMC=n```
```scons-3 --with-cxx-config --without-tcmalloc --verbose  build_gem5/gem5.opt```

So what is the official way to build gem5.opt without integrated systemc and the corresponding libgem5 ?

**Affects version**
v23.1

voertler, 2024-06-14:
Just an addtitional thing. I use the --verbose option to create a compile_commands.json based on the compile commands executed, as scons seems not be able to create one

rogerchang23424, 2024-06-14:
The build way with cutomized option is change in the https://github.com/gem5/gem5/pull/69

> So I tried as was proposed for ARM architectures in the ticket for ALL architectures
> 
> `scons-3 setconfig build/ALL USE_SYSTEMC=n` `scons-3 --with-cxx-config --without-tcmalloc --verbose  build/gem5.opt `
> 

It should be
> `scons-3 defconfig build/ALL build_opts/ALL` `scons-3 setconfig build/ALL USE_SYSTEMC=n` `scons-3 --with-cxx-config --without-tcmalloc --verbose  build/ALL/gem5.opt `

> However this didn't work as the gem5.opt is missing all ISA specific Simobjects.
>
> When doing the official KConfig way from the documentation, the options --with-cxx-config --without-tcmalloc [...]

For building `libgem5.so, just simply change to `build_gem5/gem5.opt` to `build_gem5/libgem5_opt.so`

voertler, 2024-06-14:
> It should be
> 
> > `scons-3 defconfig build/ALL build_opts/ALL` `scons-3 setconfig build/ALL USE_SYSTEMC=n` `scons-3 --with-cxx-config --without-tcmalloc --verbose  build/ALL/gem5.opt `

Have you tried this v23.1 or the head version?

The documentation says https://www.gem5.org/documentation/general_docs/kconfig_build_system/ 
`Note: In order to maintain backward compatibility with the old build scheme, users need to avoid using the “build” directory for Kconfig builds.`

Also I got errors that the options  --with-cxx-config --without-tcmalloc --verbose are not supported, but I didn't use the build directory and rather the gem5_build directory as the documentation says to use

rogerchang23424, 2024-06-14:
@voertler 

> Have you tried this v23.1 or the head version?

Yes, we can build `gem5.opt` without systemc by running three commands. You can checkout the Kconfig options in the `build/ALL/gem5.build/config` before running third option.

> The documentation says https://www.gem5.org/documentation/general_docs/kconfig_build_system/
> `Note: In order to maintain backward compatibility with the old build scheme, users need to avoid using the “build” directory for Kconfig builds.`

It means when you want to simple build gem5 with given ISA option in the `build_opts`. We can just build gem5 instance with the command `scons build/{ISA}/gem5.opt` without kconfig command. If the folder `build/{ISA}` is not exist, the scons will run defconfig to initial the kconfig `build_opts/{ISA}` options to the build dir `build/{ISA}` then impicitly before building the gem5 binary. So if the user want to use kconfig commands specify the build directory. The user should not use "build" as build directoy. However , `build/foo`, `build/foo/var` work in the case.

> Also I got errors that the options --with-cxx-config --without-tcmalloc --verbose are not supported, but I didn't use the build directory and rather the gem5_build directory as the documentation says to use

These options should be used in the thiry command(compile gem5 to binary command).

Feel free to leave comments if any concert.