gem5/gem5 #1317: dacapo (java) benchmark suite encounters "SIGSEGV"  during fs mode

Xuefeng-2000, 2024-07-02:
**Describe the bug**
When I attempt to run the dacapo benchmark(e.g. cassandra), gem5 fails to execute the Java benchmark, resulting in the following error. 

After restoring the checkpoint, YCSB did not send any requests and reported the following error. Despite the error, the Java program and gem5 did not stop, but no requests were sent or processed.

```shell
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f51053ca4f4, pid=1332, tid=1333
#
# JRE version: Java(TM) SE Runtime Environment 18.9 (11.0.22+9) (build 11.0.22+9-LTS-219)
# Java VM: Java HotSpot(TM) 64-Bit Server VM 18.9 (11.0.22+9-LTS-219, mixed mode, tiered, compressed oops, g1 gc, linux-amd64)
# Problematic frame:
# J 1516 c2 jdk.internal.org.objectweb.asm.ClassWriter.newUTF8(Ljava/lang/String;)I java.base@11.0.22 (70 bytes) @ 0x00007f51053ca4f4 [0x00007f51053ca300+0x00000000000001f4]
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /dacapobench/benchmarks/hs_err_pid1332.log

```

Upon inspecting the commit logs, it appears that '0x00007f51053ca4f4' is associated with a system call, as indicated by the message: "Committing instruction with PC (0x7f51053ca4f4=>0xffffffff81601000)(32951=>32952)".

**Affects version**

**My environment setup is as follows:**

host ISA:x86
host OS:Ubuntu 20.04.6 LTS
gem5 ISA: x86
Gem5: v23.0.0.1
VM Kernel: vmlinux-5.4.49 (gem5-resources)
JDK in image : Java 11.0.22 (2024-01-16 LTS)

**To Reproduce**

The execution method involves using the Dacapo-provided benchmarks/harness/src/ExampleCallback.java, with checkpoints triggered in requestsStarting(). 

```java
dacapobench/benchmarks/harness/src/ExampleCallback.java:
public void requestsStarting() { 
        System.err.println("Request begin!!!!!!!\n");
        String command = "bash /ckpt.sh";
        try {
          Process process = Runtime.getRuntime().exec(command);
          process.waitFor();
        } catch (Exception e) {
            e.printStackTrace();
        }
}

/ckpt.sh:

echo "======ckpt start====="

m5 exit  # switch from KVM to O3
m5 checkpoint

echo "=====ckpt finished===="
```

The script for running the image is:

```shell
echo -1 > /proc/sys/kernel/sched_rt_runtime_us
echo -1 > /proc/sys/kernel/sched_rt_period_us
cd /dacapobench/benchmarks
chrt -r 90 java -jar dacapo-evaluation-git-b281fb21.jar -c ExampleCallback cassandra
[... 483 lines cut ...]
    100%
Starting 200000 requests...
Request begin !!!!!

bash /ckpt started !!!!!
```
```shell
 A fatal error has been detected by the Java Runtime Environment:

  SIGSEGV (0xb) at pc=0x00007f51053ca4f4, pid=1332, tid=1333

 JRE version: Java(TM) SE Runtime Environment 18.9 (11.0.22+9) (build 11.0.22+9-LTS-219)
 Java VM: Java HotSpot(TM) 64-Bit Server VM 18.9 (11.0.22+9-LTS-219, mixed mode, tiered, compressed oops, g1 gc, linux-amd64)
 Problematic frame:
 J 1516 c2 jdk.internal.org.objectweb.asm.ClassWriter.newUTF8(Ljava/lang/String;)I java.base@11.0.22 (70 bytes) @ 0x00007f51053ca4f4 [0x00007f51053ca300+0x00000000000001f4]

 No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again

 An error report file with more information is saved as:
 /dacapobench/benchmarks/hs_err_pid1332.log
```

The gem5 running window does not report any errors.

Any guidance, steps, or methods on how to run Java benchmarks (e.g., dacapo, Renaissance) on gem5 would be incredibly helpful.

Xuefeng-2000, 2024-07-15:
Here are the tests I conducted:

1. Take checkpoint in KVM mode, restore with O3CPU: SIGSEGV
2. Take checkpoint in KVM mode, restore with TimingSimpleCPU: No Fault
3. Take checkpoint in KVM mode, restore with O3CPU with JIT disabled: No Fault
4. Take checkpoint in KVM mode, restore with O3CPU with COW layer disabled: SIGSEGV
5. Switch from KVM to O3CPU without using checkpoint: SIGSEGV
6. Switch from KVM to TimingCPU without using checkpoint: No Fault

All benchmarks result in a segmentation fault in JIT mode. If disabling JIT might affect the realistic behavior of the simulated programs, I am unsure how to proceed to resolve this issue.

erin-le, 2025-04-17:
@Xuefeng-2000 Hello, thanks for opening this issue! The response I got from @powerjg was as follows: JIT compilation results in self modifying code, which is where code is loaded and stored, and instruction fetch reads from stores. This is not well tested in gem5, so this could be the problem. 

Also, does this work if you run it with only atomic or only timing?

Xuefeng-2000, 2025-04-18:
> [@Xuefeng-2000](https://github.com/Xuefeng-2000) Hello, thanks for opening this issue! The response I got from [@powerjg](https://github.com/powerjg) was as follows: JIT compilation results in self [...]

Hi @erin-le ,thanks for your reply. Segmentation fault is reported only in O3CPU + JIT mode, it can be executed normally in atomic or timing mode.

powerjg, 2025-04-18:
> Segmentation fault is reported only in O3CPU + JIT mode, it can be executed normally in atomic or timing mode.

Is this with KVM fast forwarding/checkpointing or not?

A few thoughts:
1. Jitting and self-modifying code has not been tested in gem5 (especially with the O3 CPU). It is possible the pipeline is not being flushed correctly or that the data and instruction caches are not coherent.
2. KVM is flakey. It's possible there's a bug in the information that KVM is saving in the checkpoint.
3. When using KVM the guest system sees information about the host system (i.e., not all CPUID-like functionality is emulated fully). So, the guest could see a different machine type (and choose to compile to a different ISA variant) by booting in KVM than is supported by gem5. For instance, gem5 doesn't support all AVX instructions. So, if in KVM mode the jit emits AVX instructions, when these instructions are run in gem5 you can get incorrect results.

No matter what the underlying cause, I would call this a missing feature in gem5. This is a workload that we have not tested in the past, and it's not surprising that it fails.

If you find the cause and can devise a solution, the community would greatly appreciate the contribution!