gem5/gem5 #1334: ARM KVM : Failed to set register

Xuefeng-2000, 2024-07-07:
**Describe the bug**
i am using gem5's KVM to accelerate my simulation, I am currently using the latest release version of gem5, v24.0.0.0. I am encountering the same issue as mentioned in my previous email(https://www.mail-archive.com/gem5-users@gem5.org/msg21860.html).  

Here is the command I am using to run gem5:

```shell
sudo build/ARM/gem5.fast  -d $CKPT_DIR \
configs/deprecated/example/fs.py  --machine-type VExpress_GEM5_V2  \
--bootloader=${PATH_TO_BINARARY}/boot.arm64 \
--kernel=${KERNEL_PATH} \
--disk-image=${DISK_PATH} \
--script=${SCRIPT_PATH} \
--cpu-type AtomicSimpleCPU \
--fast-forward=999999999999 \
--num-cpus=1 \
--checkpoint-dir=${CKPT_DIR} \
--mem-type=SimpleMemory \
--mem-size=16GB 
```

During execution, I received the following error:

```shell
warn: The fs.py script is deprecated. It will be removed in future releases of  gem5.
Global frequency set at 1000000000000 ticks per second
warn: No dot file generated. Please install pydot to generate the dot file and pdf.
src/sim/kernel_workload.cc:46: info: kernel located at: /home/feixiang/zxf/BTB/img/binaries/binaries/arm64-vmlinux-5.10.110
src/base/loader/symtab.cc:95: warn: Cannot insert a new symbol table due to name collisions. Adding prefix to each symbol's name can resolve this issue.
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
system.vncserver: Listening for connections on port 5900
system.terminal: Listening for connections on port 3456
system.realview.uart1.device: Listening for connections on port 3457
system.realview.uart2.device: Listening for connections on port 3458
system.realview.uart3.device: Listening for connections on port 3459
system.remote_gdb: Listening for connections on port 7000
src/mem/coherent_xbar.cc:140: warn: CoherentXBar system.membus has no snooping ports attached!
src/arch/arm/fs_workload.cc:121: info: Using bootloader at address 0x10
src/arch/arm/fs_workload.cc:139: info: Using kernel entry physical address at 0x80000000
src/arch/arm/linux/fs_workload.cc:111: info: Loading DTB file: ../../ckpt/dacapo/dacapo_CPU4/cassandra_2/system.dtb at address 0x88000000
Switch at instruction count:999999999999
src/cpu/kvm/base.cc:169: info: KVM: Coalesced MMIO disabled by config.
regId:1074835116 id:6931039826524487685 addr:ffffe25a83b0
src/dev/arm/energy_ctrl.cc:252: warn: Existing EnergyCtrl, but no enabled DVFSHandler found.
src/sim/simulate.cc:199: info: Entering event queue @ 0.  Starting simulation...
regId:1074835116 id:6931039826524241986 addr:fffb7c614498
src/cpu/kvm/base.cc:894: panic: KVM: Failed to set register (0x6030000000100042) value (errno: 22)
Memory Usage: 17576124 KBytes
Program aborted at tick 0
--- BEGIN LIBC BACKTRACE ---
build/ARM/gem5.fast(_ZN4gem515print_backtraceEv+0x3c)[0xaaaac69cae9c]
build/ARM/gem5.fast(_ZN4gem512abortHandlerEi+0x5c)[0xaaaac69e2fe0]
linux-vdso.so.1(__kernel_rt_sigreturn+0x0)[0xffff9f7f37dc]
/lib/aarch64-linux-gnu/libc.so.6(+0x7f200)[0xffff9e6ff200]
/lib/aarch64-linux-gnu/libc.so.6(raise+0x1c)[0xffff9e6ba67c]
/lib/aarch64-linux-gnu/libc.so.6(abort+0xe4)[0xffff9e6a7130]
build/ARM/gem5.fast(+0xe2d068)[0xaaaac607d068]
build/ARM/gem5.fast(_ZNK4gem510BaseKvmCPU9getOneRegEmPv+0x0)[0xaaaac680a014]
build/ARM/gem5.fast(_ZN4gem511ArmV8KvmCPU14updateKvmStateEv+0xf4)[0xaaaac7d267b4]
build/ARM/gem5.fast(_ZN4gem510BaseKvmCPU4tickEv+0x1d8)[0xaaaac6808478]
build/ARM/gem5.fast(_ZN4gem510EventQueue10serviceOneEv+0xa8)[0xaaaac69d7d48]
build/ARM/gem5.fast(_ZN4gem59doSimLoopEPNS_10EventQueueE+0x8c)[0xaaaac69f972c]
build/ARM/gem5.fast(_ZNSt6thread11_State_implINS_8_InvokerISt5tupleIJZN4gem516SimulatorThreads17runUntilLocalExitEvEUlPNS3_10EventQueueEE_S6_EEEEE6_M_runEv+0xac)[0xaaaac69fa85c]
/lib/aarch64-linux-gnu/libstdc++.so.6(+0xd31fc)[0xffff9e9d31fc]
/lib/aarch64-linux-gnu/libc.so.6(+0x7d5c8)[0xffff9e6fd5c8]
/lib/aarch64-linux-gnu/libc.so.6(+0xe5edc)[0xffff9e765edc]
--- END LIBC BACKTRACE ---
For more info on how to address this issue, please visit 
https://www.gem5.org/documentation/general_docs/common-errors/ 
```

**configuration**

Arch: AArch64

gem5: v24.0.0.0

Linux:. Ubuntu 5.15.0-113-generic

-----

Could you please assist me with this matter?

ritaru, 2024-07-10:
Try VExpress_GEM5_V1 with --param 'system.realview.gic.simulate_gic=True'. I mostly had success with this method. If this is not working, try fs_bigLITTLE.py in configs/example/arm. Otherwise you might have to write a custom fs.py for your setup.

Xuefeng-2000, 2024-07-10:
It doesn't work when I use VExpress_GEM5_V1 with --param 'system.realview.gic.simulate_gic=True'. Then, I try  fs_bigLITTLE.py, KVM simulation can start but stalls on the boot process. Do you know how I could solve it? Thank you.

```shell
sudo build/ARM/gem5.fast  -d $CKPT_DIR \
configs/example/arm/fs_bigLITTLE.py --machine-type VExpress_GEM5_V1 \
 --param 'system.realview.gic.simulate_gic=True' \
--bootloader=${PATH_TO_BINARARY}/boot.arm64 \
--kernel=${KERNEL_PATH} \
--disk ${DISK_PATH} \
--cpu-type kvm \
--mem-size=16GB 

-----
gem5 :

Global frequency set at 1000000000000 ticks per second
info: Simulated platform: VExpress_GEM5_V1
info: Running in PDES mode with a 1ms simulation quantum.
warn: No dot file generated. Please install pydot to generate the dot file and pdf.
src/sim/kernel_workload.cc:46: info: kernel located at: /home/feixiang/zxf/BTB/img/binaries/binaries/arm64-vmlinux-5.10.110
src/base/loader/symtab.cc:95: warn: Cannot insert a new symbol table due to name collisions. Adding prefix to each symbol's name can resolve this issue.
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
system.vncserver: Listening for connections on port 5900
system.terminal: Listening for connections on port 3456
system.realview.uart1.device: Listening for connections on port 3457
system.realview.uart2.device: Listening for connections on port 3458
system.realview.uart3.device: Listening for connections on port 3459
system.remote_gdb: Listening for connections on port 7000
src/mem/coherent_xbar.cc:140: warn: CoherentXBar system.membus has no snooping ports attached!
src/arch/arm/fs_workload.cc:121: info: Using bootloader at address 0x10
src/arch/arm/fs_workload.cc:139: info: Using kernel entry physical address at 0x80000000
src/arch/arm/linux/fs_workload.cc:111: info: Loading DTB file: ../../ckpt/dacapo/dacapo_CPU4/tomcat_2/system.dtb at address 0x88000000
src/cpu/kvm/base.cc:169: info: KVM: Coalesced MMIO disabled by config.
src/cpu/kvm/base.cc:169: info: KVM: Coalesced MMIO disabled by config.
src/dev/arm/energy_ctrl.cc:252: warn: Existing EnergyCtrl, but no enabled DVFSHandler found.
src/sim/simulate.cc:199: info: Entering event queue @ 0.  Starting simulation...
src/dev/arm/rv_ctrl.cc:176: warn: SCReg: Access to unknown device dcc0:site0:pos0:fn7:dev0
src/dev/arm/rv_ctrl.cc:122: warn: Tried to read RealView I/O at offset 0x48 that doesn't exist
src/dev/arm/rv_ctrl.cc:122: warn: Tried to read RealView I/O at offset 0x8 that doesn't exist
src/dev/arm/rv_ctrl.cc:122: warn: Tried to read RealView I/O at offset 0x48 that doesn't exist
src/dev/arm/rv_ctrl.cc:122: warn: Tried to read RealView I/O at offset 0x60 that doesn't exist
src/dev/arm/rv_ctrl.cc:198: warn: Tried to write RVIO at offset 0xa8 (data 0) that doesn't exist

-------
terminal:

==== m5 terminal: Terminal 0 ====
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x481fd010]
[    0.000000] Linux version 5.10.110 (kaustavg@citra) (aarch64-linux-gnu-gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #3 SMP PREEMPT Wed Apr 27 17:07:00 PDT 2022
[    0.000000] Machine model: V2P-CA15
[    0.000000] earlycon: pl11 at MMIO 0x000000001c090000 (options '')
[    0.000000] printk: bootconsole [pl11] enabled
[    0.000000] Memory limited to 16384MB
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000080000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x000000047fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080000000-0x000000047fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080000000-0x000000047fffffff]
[    0.000000] On node 0 totalpages: 4194304
[    0.000000]   DMA zone: 8192 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 524288 pages, LIFO batch:63
[    0.000000]   Normal zone: 57344 pages used for memmap
[    0.000000]   Normal zone: 3670016 pages, LIFO batch:63
[    0.000000] cma: Reserved 16 MiB at 0x00000000ff000000
[    0.000000] percpu: Embedded 22 pages/cpu s52568 r8192 d29352 u90112
[    0.000000] pcpu-alloc: s52568 r8192 d29352 u90112 alloc=22*4096
[... 137 lines cut ...]
[    0.228269] pci_bus 0000:00: root bus resource [bus 00-ff]
[    0.228895] pci_bus 0000:00: root bus resource [io  0x0000-0xffff]
[    0.230627] pci_bus 0000:00: root bus resource [mem 0x40000000-0x7fffffff] (bus address [0x00000000-0x3fffffff])
[    0.231787] pci_bus 0000:00: scanning bus
[    0.232334] pci 0000:00:01.0: [1af4:1000] type 00 class 0xff0000
[    0.233065] pci 0000:00:01.0: reg 0x10: [io  0x0000-0x001f]
[    0.233813] pci 0000:00:01.0: reg 0x30: [mem 0x40000000-0x400007ff pref]
[    0.234934] pci_bus 0000:00: fixups for bus
[    0.235418] pci_bus 0000:00: bus scan returning with max=00
[    0.236100] pci 0000:00:01.0: BAR 6: assigned [mem 0x40000000-0x400007ff pref]
[    0.236941] pci 0000:00:01.0: BAR 0: assigned [io  0x1000-0x101f]
[    0.238609] virtio-pci 0000:00:01.0: assign IRQ: got 31
[    0.239466] virtio-pci 0000:00:01.0: enabling device (0000 -> 0001)
[    0.240209] virtio-pci 0000:00:01.0: virtio_pci: leaving for legacy driver
[    0.241015] virtio-pci 0000:00:01.0: enabling bus mastering
[    0.242564] Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled
[    0.246593] cacheinfo: Unable to detect cache hierarchy for CPU 0
[    0.250277] virtio_blk virtio0: [vda] 16514064 512-byte logical blocks (8.46 GB/7.87 GiB)
[    0.251286] vda: detected capacity change from 0 to 8455200768
[    0.258310]  vda: vda1
[    0.259635] basic-mmio-gpio: Failed to locate of_node [id: -2]
[    0.260411] basic-mmio-gpio: Failed to locate of_node [id: -2]
[    0.261178] basic-mmio-gpio: Failed to locate of_node [id: -2]

```

giactra, 2024-07-10:
Hi @Xuefeng-2000 ; related to the initial issue (VExpress_GEM5_V2), have you tried commenting out the following line of code?

https://github.com/gem5/gem5/blob/stable/src/dev/arm/gic_v3_cpu_interface.cc#L2604

I remember some issues when trying to set some values from gem5 to KVM. Most likely we fail when we encounter non-matching configurations between guest and host. If commenting ICC_CTLR does not work, try debugging with gdb and let me know which copy fails

hnpl, 2024-08-08:
@giactra Do you think it is a good idea to make copying the ICC_CTRL_EL1 register from guest to host optional? I was able to use ARM KVM when I removed this line as well.

Xuefeng-2000, 2024-08-12:
> https://github.com/gem5/gem5/blob/stable/src/dev/arm/gic_v3_cpu_interface.cc#L2604

@giactra    It still doesn't work. I try debugging with gdb and find that this copy is from this line:
https://github.com/gem5/gem5/blob/b1a44b89c7bae73fae2dc547bc1f871452075b85/src/arch/arm/kvm/armv8_cpu.cc#L240
setOneReg(INT_REG(regs.pstate), static_cast<uint64_t>(cpsr));

nkrim, 2024-09-10:
I may have isolated the initial issue to the fact that `fs.py` does not set `release = ArmDefaultRelease.for_kvm()` and instead seems to just use the default release in `ArmSystem`:
https://github.com/gem5/gem5/blob/b1a44b89c7bae73fae2dc547bc1f871452075b85/src/arch/arm/ArmSystem.py#L302

While working with `ArmBoard` for KVM, omitting `.for_kvm` will cause the exact same error with the exact same register ID. However, I am a bit unfamiliar with `fs.py` and I couldn't get a simulation running immediately, so I was unable to test. Anyway, I think this is ultimately a consequence of `fs.py`'s deprecation as the other `fs_*` scripts explicitly set their release as above.