gem5/gem5 #1349: Multiple RubySystem objects

abmerop, 2024-07-13:
[Description WIP] Allow for multiple RubySystem objects. Not to be confused with multiple Ruby **protocols**.  This has been a topic in the past for having one System object with two nodes communicating, but has not made it into gem5. Now, we have a PCI device with a unique Ruby protocol (amdgpu). This would allow for more modular GPUs in gem5 and much easier integration with stdlib. For example, there could be an X86PciBoard (~name) and you could "plug in" any PCI device, e.g., an AMD gpu. 

## Challenges

RubySystem has multiple static members - i.e., is a singleton.  These would have to be removed. The biggest challenge would be the makeLineAddress function. This requires either passing a pointer to RubySystem or defining it elsewhere. The last developer meeting proposes the latter.

## Opportunity

The AMD GPU object right now uses a "disjoint" Ruby network which works for dGPU simulation.  There is a design issue with Ruby's network that causes problems with GPUFS because both share the same directory SLICC machine.  This causes bad NoC routes and immediately deadlocks simulation if a timing CPU is used (O3, Minor, Simple).  Mutiple RubySystems would remove this problem and would allow for **APU** simulations in GPUFS.

abmerop, 2024-08-10:
Had some time to look into this today. Here is what I am suggesting:

- AbstractController gets a RubySystem pointer. This will require changing some configs to make everything inheriting AbstractController set a RubySystem pointer.
   - Alternate: The sequencer already has a RubySystem poiner and most SLICC state machines (files ending with .sm) have a sequencer. Not all do which will require changing those SMs, so this is less ideal. However, all config files except stdlib set the ruby_system pointer when creating a RubySequencer SimObject

- Modify SLICC parser to set sizes for DataBlock and WriteMask. These use RubySystem static members to alloc memory. This is going to be very difficult to modify in a non-intrusive way otherwise (i.e., would probably requiring changing every single protocol to allocate data blocks manually).
   - Currently I have modified the SLICC parser to find any SLICC object defined using `structure(...)` which contains a `DataBlock` member. I removed the default constructor from DataBlock and forced a constructor containing the block size to be used.
   - This worked until the point where an `AbstractCacheEntry` needs to be allocated. The SLICC parser does not allow you to call a non-default constructor on Entry (or at least doesn't understand parenthesis within parenthesis?). I've attempted to instead call a function to allocate an Entry with a block size and assign to a variable which can be passed to `CacheMemory::allocate`, but there is no way to pass the type as SLICC also does not understand C++ templates.
   - Currently stuck on the above issue and focusing only on DataBlock for now. Similar changes will be needed for WriteMask and other things in `src/mem/ruby/structures/`.

powerjg, 2024-08-12:
> AbstractController gets a RubySystem pointer. This will require changing some configs to make everything inheriting AbstractController set a RubySystem pointer.

This should be easy. You can just add `ruby_system = Param.RubySystem(Parent.any, "...")` and as long as the controller is a child of  `RubySystem` it will work. This might require changing the config scripts and some small changes to the stdlib, though.

> Modify SLICC parser to set sizes for DataBlock and WriteMask. These use RubySystem static members to alloc memory. 

Could you post your current WIP code? I would be happy to take a quick look and see if I can come up with any solutions.

abmerop, 2024-08-12:
> This should be easy. You can just add `ruby_system = Param.RubySystem(Parent.any, "...")` and as long as the controller is a child of `RubySystem` it will work. This might require changing the config scripts and some small changes to the stdlib, though.

I want to avoid using `Parent.any` since it might pick the wrong RubySystem. I am not an expert on how the python proxies work so maybe it's OK if it is a child object? My current plan is to set this explicitly. I think most configs and stdlib set this explicitly already when I did a cursory check.

> Could you post your current WIP code? I would be happy to take a quick look and see if I can come up with any solutions.

Posted draft here: https://github.com/gem5/gem5/pull/1453

abmerop, 2024-08-28:
@powerjg Update: Brandon Potter still has his patches from 2016 (? .. guessing).  The solution to DataBlock / WriteMask was to modify *every* protocol.  I am hesitant to do that, even though it makes sense the protocol should have to do it.  My thought is to add some SLICC parser magic to init the DataBlock, but I am not a huge fan of things being hidden from the user/developer and I am not sure if that is possible.

In either case might be a good idea to rope in the other ISA devs to this conversation. Let me know what your thoughts are as we have some different goals here.

giactra, 2024-10-03:
A side comment: it would be nice if after the RubySystem has been reworked we would specify in the documentation what the RubySystem actually is... And when a user is supposed to use multiple RubySystems instead of one

abmerop, 2024-10-03:
> A side comment: it would be nice if after the RubySystem has been reworked we would specify in the documentation what the RubySystem actually is... And when a user is supposed to use multiple RubySystems instead of one

Sure, I can write up something for the website to be queued up for next release.

powerjg, 2024-10-28:
It looks like the only outstanding to do item on this issue is documentation. Is that correct?

abmerop, 2024-10-30:
> It looks like the only outstanding to do item on this issue is documentation. Is that correct?

Yes, i'll start with #1735

closed 2025-01-09 by erin-le