gem5/gem5 #1355: arch-riscv: Potential bug in wfi implementation

h0lyalg0rithm, 2024-07-17:
**Describe the bug**
WFI instruction implementation in gem5 does not match the specification.

**Affects version**
Commit ID in develop - 9b8c84cb5d2f71b5af4329f9ff6cdad380a6225c

**Expected behavior**
According to the spec https://github.com/riscv/riscv-isa-manual/releases/tag/20240411 on page 51 and 52 it mentions the following

> The WFI instruction can also be executed when interrupts are disabled. The operation of WFI must be unaffected by the global interrupt bits in mstatus (MIE and SIE) and the delegation register mideleg (i.e., the hart must resume if a locally enabled interrupt becomes pending, even if it has been delegated to a less-privileged mode), but should honor the individual interrupt enables (e.g, MTIE) (i.e.,
implementations should avoid resuming the hart if the interrupt is pending but not individually enabled). WFI is also required to resume execution for locally enabled interrupts pending at any privilege level, regardless of the global interrupt enable at each privilege level.

**Current Implementation**
In the code here https://github.com/gem5/gem5/blob/9b8c84cb5d2f71b5af4329f9ff6cdad380a6225c/src/arch/riscv/isa/decoder.isa#L4883
I checks for pending interrupt  and masked interrupt but we dont check if the interrupt is enabled.This results in the hart being resumed instead.

4883   │                             if (ic->readIP() == 0
4884   │                                 && xc->readMiscReg(MISCREG_NMIP) == 0) {
4885   │                                 tc->quiesce();
4886   │                             }

**Suggested fix**
Add new method to the https://github.com/gem5/gem5/blob/9b8c84cb5d2f71b5af4329f9ff6cdad380a6225c/src/arch/riscv/interrupts.hh 
```
bool checkInterruptsWfi() const {
     return checkNonMaskableInterrupt() || (ip & ie).any());
}
```
The method is similar to the existing `checkInterrupts` method but it doesnt take the globalMask() into account, as mentioned in the spec.

If someone who has more knowledge about the spec could validate if this approach is correct.

**Host Operating System**
Ubuntu 22.04, Mac OS X, etc.

**Host ISA**
RISC-V

cc @adriaarmejach

robhau, 2024-07-19:
Hi @h0lyalg0rithm,

> The WFI instruction can also be executed when interrupts are disabled.

I think that means that we can execute `wfi` if interrupts are enabled **or** disabled.

If we execute `wfi` with disabled interrupts and without pending interrupts, the hart will stop execution. Then, if an interrupt becomes pending, the hart will resume at `pc + 4`.

See this notes in the privileged spec on p. 52

> By allowing wakeup when interrupts are disabled, an alternate entry point to an interrupt
> handler can be called that does not require saving the current context, as the current
> context can be saved or discarded before the WFI is executed.

Correct me if I have overlooked something 😄.

h0lyalg0rithm, 2024-07-19:
Hi @robhau 
You are right, the specification allows the flexibility to resume the hart even if there are no interrupts(global or local).
This change changes the implementation to only resume if there are no local interrupts.This helps a lot of with  performance as the simulator doesnt have to simulate the wfi loop.

In some of experiments this reduced the simulation time for checkpointing(opensbi bootloader + booting linux) to less half.

robhau, 2024-07-19:
Hi @hortune,

as I understand the specification, it is optional to resume execution if interrupt is pending, but not locally enabled, right?

The resuming after a pending but not local enabled interrupt can be desired to avoid the context switch if you want to react on every interrupt anyway. Therefore, I think you should implement this as an option to let the user decide which behavior he wants.

I already did this a few month ago in my forked repository, maybe it helps for clarification what I mean.

https://github.com/robhau/gem5/commit/3083667bc48a8909e54dbf6ddf76c84a4370705b

h0lyalg0rithm, 2024-07-19:
Hi @robhau Your implementation makes sense, we were discussing internally if we should make it optional and I think that approach is better based on the use case you mentioned.
We should try to get your changes upstreamed or do you save something else in mind that is the blocker.

Thanks for the feedback

robhau, 2024-07-19:
Hi @h0lyalg0rithm,

I can create a pull request with my changes.

h0lyalg0rithm, 2024-07-19:
Thanks that would be great, I will remove my PR in that case

ivanaamit, 2024-08-12:
Hi @robhau, @h0lyalg0rithm, the linked PR has been merged. Could you please let me know if this resolves the issue, or if the PR was just a partial fix? Thank you.

h0lyalg0rithm, 2024-08-14:
@ivanaamit The Pr includes the change, I had created so I expect it to work the same. Will close this issue then

closed 2024-08-14 by h0lyalg0rithm