gem5/gem5 #1380: RISCV Vector code problem

huangxiansheng123, 2024-07-25:
[image]

Here on line 128 elen might be rewritten as vlen？

zfcfuture, 2024-07-25:
i meet the same problem, wait for a reply.

huangxiansheng123, 2024-07-25:
The path of the file is src/arch/riscv/isa/formats/vector_conf.isa

hnpl, 2024-07-26:
I believe that condition corresponds to this part of the spec,

https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc?plain=1#L330

Having said that, I'm puzzled why it was `sew > std::min(vflmul, 1.0f) * elen` instead of `sew > vflmul * elen`.

huangxiansheng123, 2024-07-29:
The sew here does not seem to be the smallest and does not match the SEW_{MIN} here in spec