gem5/gem5 #1393: configs,arch-riscv,stdlib: Core dump when booting Ubuntu 20.04 on RISCVMatchedBoard with riscvmatched-fs.py

robhau, 2024-07-29:
**Describe the bug**
The configuration script `configs/example/gem5_library/riscvmatched-fs.py` should boot Ubuntu 20.04 on RISCVMatchedBoard. After booting, the simulation should end via `m5_exit`. However, the simulation is aborted after about 1.86 simulated seconds.

**Affects version**
Development Branch, SHA: 679000f91dd50d67632cfbd46ff2886458c4f3ea

**gem5 Modifications**
No modification

**To Reproduce**
Steps to reproduce the behavior. Please assume starting from a clean repository:

1. Compile gem5: `scons build/ALL/gem5.fast --linker=mold -j(p)` (P - Number of available cores to speed up compilation)
2. Execute simulation: `build/ALL/gem5.fast -r configs/example/gem5_library/riscvmatched-fs.py`

**Terminal Output**

```shell
gem5 Simulator System.  https://www.gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version DEVELOP-FOR-24.1
gem5 compiled Jul 29 2024 07:02:38
gem5 started Jul 29 2024 07:03:59
gem5 executing on rh, pid 926797
command line: build/ALL/gem5.fast -r configs/example/gem5_library/riscvmatched-fs.py

info: Using default config
Global frequency set at 1000000000000 ticks per second
board.platform.terminal: Listening for connections on port 3456
src/arch/riscv/isa.cc:276: info: RVV enabled, VLEN = 256 bits, ELEN = 64 bits
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdUnitStrideLoad
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdUnitStrideStore
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdUnitStrideMaskLoad
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdUnitStrideMaskStore
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdStridedLoad
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdStridedStore
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdIndexedLoad
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdIndexedStore
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdWholeRegisterLoad
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdWholeRegisterStore
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdUnitStrideFaultOnlyFirstLoad
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdUnitStrideSegmentedLoad
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdUnitStrideSegmentedStore
src/arch/riscv/isa.cc:276: info: RVV enabled, VLEN = 256 bits, ELEN = 64 bits
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdUnitStrideLoad
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdUnitStrideStore
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdUnitStrideMaskLoad
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdUnitStrideMaskStore
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdStridedLoad
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdStridedStore
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdIndexedLoad
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdIndexedStore
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdWholeRegisterLoad
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdWholeRegisterStore
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdUnitStrideFaultOnlyFirstLoad
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdUnitStrideSegmentedLoad
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdUnitStrideSegmentedStore
src/arch/riscv/isa.cc:276: info: RVV enabled, VLEN = 256 bits, ELEN = 64 bits
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdUnitStrideLoad
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdUnitStrideStore
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdUnitStrideMaskLoad
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdUnitStrideMaskStore
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdStridedLoad
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdStridedStore
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdIndexedLoad
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdIndexedStore
src/cpu/minor/execute.cc:166: warn: No functional unit for OpClass SimdWholeRegisterLoad
[... 80 lines cut ...]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af)[0x7f2e2e3433af]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9d68)[0x7f2e2e1fc9c8]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af)[0x7f2e2e3433af]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x829e)[0x7f2e2e1faefe]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af)[0x7f2e2e3433af]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyEval_EvalCode+0xbe)[0x7f2e2e33e3de]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1bd96d)[0x7f2e2e33e96d]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1287b3)[0x7f2e2e2a97b3]
--- END LIBC BACKTRACE ---
For more info on how to address this issue, please visit https://www.gem5.org/documentation/general_docs/common-errors/ 

```

**Expected behavior**
Ubuntu 20.04 should be booted succesfull. The simulation then should be leaved via M5Op (`m5_exit`).

**Host Operating System**
Ubuntu 22.04.4 LTS

**Host ISA**
X86

**Compiler used**
g++ 11.4.0
Linker: mold (`--linker=mold`)

powerjg, 2024-08-09:
> src/cpu/minor/lsq.cc:1604: panic: Do not expect cross-cache-line atomic memory request

Yeah, something definitely went wrong here!

Harshil2107, 2024-08-19:
I ran the `riscvmatched-fs.py` on current develop and exits after boot, I think the bug was fixed in the commits after this issue was made.

robhau, 2024-08-21:
Hi @Harshil2107,

thank you for your reply 😄.

I deleted my build directory and compiled gem5 again. Now, it also boots successfully on my side.

powerjg, 2024-08-21:
Great news!

closed 2024-08-21 by powerjg