gem5/gem5 #1443: Stats Resetting

studyztp, 2024-08-09:
Currently, there does not exist a documentation or clear guidance about what stats in the stats.txt file that can be reset by the `m5.stats.reset()` function and what stats cannot.
For example, the stats `simInsts` does not reset when calling `m5.stats.reset()`.

BobbyRBruce, 2024-08-13:
Shouldn't it reset?

I'm wondering  if you're suggesting we document which stats are, and which are not, wiped, or if there are stats that should be reset but aren't.

My feeling is `m5.stats.reset()` should reset _all_ stats... no?

studyztp, 2024-08-17:
Sadly, `m5.stats.reset()` is not resetting all the stats in the stats file. 
I attached a simple run script that can show that. This script exits the simulation after simulating the workload for 143293729 Ticks, then dump and reset the stats using `m5.stats.reset()` and `m5.stats.dump()`. It then falls back to the simulation for only 1 Ticks then dump the stats before exit. The stats file should suggest that `simInsts` did not get reset.

We've been noticing `simInsts` is not reset, but maybe there are more. 
I suggest testing out what stats in the stats file that can or cannot be reset first, then depending on what stats are actually having issue, we can decide if we should document them, wipe them, or fix them.
If it's only `simInsts` that is being an issue, I think add the warning in its stat description should be fine.

closed 2024-10-22 by erin-le