gem5/gem5 #1452: Unable to simulate ARM O3 CPU FS with external DRAMSys Tool

atrah22, 2024-08-12:
@powerjg ,

Hello, I could run the simulation using the provided example arm-hello-dramsys.py script (configs/example/gem5_library/dramsys).
However, I failed to run FS ARM simulation with DRAMSys. I have Private L1,L2 and shared L3 cache (single core O3 cpu).

In devices.py script (configs/example/arm), I added configure_dramsys in SimpleSystem class like below:

```
class SimpleSystem(BaseSimpleSystem):
    """
    Meant to be used with the classic memory model
    """

    def __init__(self, caches, mem_size, platform=None, **kwargs):
        super().__init__(mem_size, platform, **kwargs)

        self.membus = MemBus()
        # CPUs->PIO
        self.iobridge = Bridge(delay="50ns")

        self._caches = caches
        if self._caches:
            self.iocache = IOCache(addr_ranges=self.mem_ranges)
        else:
            self.dmabridge = Bridge(delay="50ns", ranges=self.mem_ranges)

    def connect(self):
        self.iobridge.mem_side_port = self.iobus.cpu_side_ports
        self.iobridge.cpu_side_port = self.membus.mem_side_ports

        if self._caches:
            self.iocache.mem_side = self.membus.cpu_side_ports
            self.iocache.cpu_side = self.iobus.mem_side_ports
        else:
            self.dmabridge.mem_side_port = self.membus.cpu_side_ports
            self.dmabridge.cpu_side_port = self.iobus.mem_side_ports

        if hasattr(self.realview.gic, "cpu_addr"):
            self.gic_cpu_addr = self.realview.gic.cpu_addr
        self.realview.attachOnChipIO(self.membus, self.iobridge)
        self.realview.attachIO(self.iobus)
        self.system_port = self.membus.cpu_side_ports

    def attach_pci(self, dev):
        self.realview.attachPciDevice(dev, self.iobus)

    def configure_dramsys(self, dramsys_memory):
        self.dramsys = dramsys_memory
        self.mem_ranges = [AddrRange('4GB')]
        self.dramsys.set_memory_range(self.mem_ranges)
        self.membus.mem_side_ports = self.dramsys.get_mem_ports()[0][1]
```
In fs_bigLITTLE.py script, I tried to configure dramsys instead of using config_mem (gem5 DRAM config) in createSystem class like below:

```
def createSystem(
    options,
    caches,
    kernel,
    bootscript,
    machine_type="VExpress_GEM5",
    disks=[],
    mem_size=default_mem_size,
    bootloader=None,
):
    platform = ObjectList.platform_list.get(machine_type)
    m5.util.inform("Simulated platform: %s", platform.__name__)

    
    sys = devices.SimpleSystem(
        caches,
        mem_size,
        platform(),
        workload=ArmFsLinux(object_file=SysPaths.binary(kernel)),
        readfile=bootscript,
    )

    dramsys_memory = DRAMSysMem(
        configuration="/home/gem5/ext/dramsys/DRAMSys/configs/ddr4-example.json",
        recordable=True,
        resource_directory="/home/gem5/ext/dramsys/DRAMSys/configs",
        size="4GB",
    )

    #config_mem(options, sys)
    sys.configure_dramsys(dramsys_memory)

    sys.connect()

    # Attach disk images
    if disks:

        def cow_disk(image_file):
            image = CowDiskImage()
            image.child.image_file = SysPaths.disk(image_file)
            return image

        sys.disk_images = [cow_disk(f) for f in disks]
        sys.pci_vio_block = [
            PciVirtIO(vio=VirtIOBlock(image=img)) for img in sys.disk_images
        ]
        for dev in sys.pci_vio_block:
            sys.attach_pci(dev)

    sys.realview.setupBootLoader(sys, SysPaths.binary, bootloader)

    return sys
```

I get the following error when running FS ARM simulation:

```

■ ■ ■  DRAMSys5.0, Copyright (c) 2023
■ ■ ■  RPTU Kaiserslautern-Landau,
■ ■ ■  Fraunhofer IESE

===========================================================================
Memory Configuration:

 Memory type:           DDR4
 Memory size in bytes:  4294967296
 Channels:              1
 Ranks per channel:     1
 Bank groups per rank:  4
 Banks per rank:        16
 Rows per bank:         32768
 Columns per row:       1024
 Device width in bits:  8
 Device size in bits:   4294967296
 Device size in bytes:  536870912
 Devices per rank:      8

===========================================================================
Used Address Mapping:

 Bg  1: 0000000000000000000000000000000000100000000000000000000000000000
 Bg  0: 0000000000000000000000000000000000010000000000000000000000000000
 Ba  1: 0000000000000000000000000000000010000000000000000000000000000000
 Ba  0: 0000000000000000000000000000000001000000000000000000000000000000
 Ro 14: 0000000000000000000000000000000000001000000000000000000000000000
 Ro 13: 0000000000000000000000000000000000000100000000000000000000000000
 Ro 12: 0000000000000000000000000000000000000010000000000000000000000000
 Ro 11: 0000000000000000000000000000000000000001000000000000000000000000
 Ro 10: 0000000000000000000000000000000000000000100000000000000000000000
 Ro  9: 0000000000000000000000000000000000000000010000000000000000000000
 Ro  8: 0000000000000000000000000000000000000000001000000000000000000000
 Ro  7: 0000000000000000000000000000000000000000000100000000000000000000
 Ro  6: 0000000000000000000000000000000000000000000010000000000000000000
 Ro  5: 0000000000000000000000000000000000000000000001000000000000000000
 Ro  4: 0000000000000000000000000000000000000000000000100000000000000000
 Ro  3: 0000000000000000000000000000000000000000000000010000000000000000
 Ro  2: 0000000000000000000000000000000000000000000000001000000000000000
 Ro  1: 0000000000000000000000000000000000000000000000000100000000000000
 Ro  0: 0000000000000000000000000000000000000000000000000010000000000000
 Co  9: 0000000000000000000000000000000000000000000000000001000000000000
 Co  8: 0000000000000000000000000000000000000000000000000000100000000000
 Co  7: 0000000000000000000000000000000000000000000000000000010000000000
 Co  6: 0000000000000000000000000000000000000000000000000000001000000000
 Co  5: 0000000000000000000000000000000000000000000000000000000100000000
 Co  4: 0000000000000000000000000000000000000000000000000000000010000000
 Co  3: 0000000000000000000000000000000000000000000000000000000001000000
 Co  2: 0000000000000000000000000000000000000000000000000000000000100000
 Co  1: 0000000000000000000000000000000000000000000000000000000000010000
 Co  0: 0000000000000000000000000000000000000000000000000000000000001000
 By  2: 0000000000000000000000000000000000000000000000000000000000000100
 By  1: 0000000000000000000000000000000000000000000000000000000000000010
 By  0: 0000000000000000000000000000000000000000000000000000000000000001

===========================================================================
src/sim/kernel_workload.cc:46: info: kernel located at: /home/gem5/fs_images/aarch20220707/arm64-vmlinux-4.9_patched
src/base/loader/symtab.cc:95: warn: Cannot insert a new symbol table due to name collisions. Adding prefix to each symbol's name can resolve this issue.
src/mem/physical.cc:108: fatal: fatal condition addrMap.insert(m->getAddrRange(), m) == addrMap.end() occurred: Memory address range for system.realview.bootmem is overlapping
Memory Usage: 353344 KBytes

```
How to select memory address range for DRAMSys when running FS simulation?

powerjg, 2024-08-12:
I'm pretty sure this will work with the standard library. Unfortunately, the other example scripts are quite complex and it's impossible for the maintainers to make them work for every possible use case. However, the standard library has fixed APIs that are easier to maintain and extend.

Try using this example with a DRAMSys memory system: https://github.com/gem5/gem5/blob/stable/configs/example/gem5_library/arm-ubuntu-run.py

derchr, 2024-10-18:
Thank you for reporting this issue. We identified the issue and are working on a fix. In the meantime, as a workaround, you can comment out line 555 in src/libdramsys/DRAMSys/controller/Controller.cpp (https://github.com/tukl-msd/DRAMSys/blob/master/src/libdramsys/DRAMSys/controller/Controller.cpp#L555).

When the issue is fixed, I will update the integration in gem5.

erin-le, 2025-03-14:
A note to gem5 users who come across this: it appears that this issue is still ongoing.

powerjg, 2025-04-08:
Fixed by #2093