gem5/gem5 #1466: gem5-resources: Specifying a workload with an extra file to be passed as an argument

Sir-NoChill, 2024-08-15:
I am looking to develop a set of resources for our computer architecture course using the gem5 resources system. Some of these programs require that a file path is passed to the program, ex:

`./my_program 500 foo.bar`

In the gem5 resources, I see that we can specify a file resource (c.f. [llvm-ministat](http://resources.gem5.org/resources/riscv-llvm-minisat-run/raw?database=gem5-resources&version=1.0.0)) but in that case I only see how to use it when piped through stdin.

Is there currently any way to achieve passing a file-path into a program?

Harshil2107, 2024-08-15:
To pass a file path along with an argument to a binary, you can include the file path as part of the `arguments` field in the `additional_params`. However, unlike using `FileResource` via stdin, this method is not dynamic.

For example, you can specify the file path in the `arguments` like this:

```json
"additional_params": {
    "arguments": [
        "500",
        "./path/to/foo.bar"
    ]
}
```

If you want to do this programmatically, you would need to update the `set_se_binary_workload` function to handle this use case. This can be done by adding a new parameter, such as `arg_input_file: Optional[FileResource] = None`, to the `set_se_binary_workload` function.

Then, modify the function to append the file path to the arguments list if the file is specified. Specifically, update the line where the command arguments are set to include `arg_input_file.get_local_path()`.

For example, in the `se_binary_workload.py` file (found [here](https://github.com/gem5/gem5/blob/b1a44b89c7bae73fae2dc547bc1f871452075b85/src/python/gem5/components/boards/se_binary_workload.py#L71)), you would:

1. Add the `arg_input_file` parameter:
   ```python
   def set_se_binary_workload(..., arg_input_file: Optional[FileResource] = None):
   ```

2. Add the line where arguments are set (found [here](https://github.com/gem5/gem5/blob/b1a44b89c7bae73fae2dc547bc1f871452075b85/src/python/gem5/components/boards/se_binary_workload.py#L113)) to:
   ```python
   if arg_input_file:
       process.cmd += [arg_input_file.get_local_path()]
   ```

Once these changes are made, you can pass a file resource in the JSON with the key `arg_input_file`, and it will automatically set the corresponding parameter to the file resource you provided.
```json
    "resources": {
        "binary": {
            "id": "binary",
            "resource_version": "1.0.0"
        },
        "arg_input_file": {
            "id": "input",
            "resource_version": "1.0.0"
        }
    }
```

Sir-NoChill, 2024-08-15:
Do you think it would be feasible to add some wildcard entry into the arguments field so that we could specify file placement in the arguments list? Ideally, I think it would be nice to have a resource with the following information:

```json
    "additional_params": {
        "arguments": [
            "500",
            "%%FILE%%",
            ...,
            "%%FILE%%
        ]
    },
    "resources": {
        "binary": {
            "id": "binary",
            "resource_version": "1.0.0"
        },
        "arg_input_file": [
        {
            "id": "input",
            "resource_version": "1.0.0",
            "position": 1
        },
        {
            "id": "input2",
            "resource_version": "2.0.0",
            "position": 2
        }],
        ...
    }
```
    
Would this be something of interest to gem5? I think I could see a way to do it in the current resource framework, and I only see one example on gem5-resources using the `file` resource type, but I am not sure what compatibility this might break in some benchmarks.

Thank you for your response!

powerjg, 2024-08-16:
Hi @Sir-NoChill 

I don't think we would want to add this feature. One of the most important parts of gem5-resources is that it enables reproducible simulations. To do this, we require that there is a fixed input for the simulator. By adding wildcards or command-line parameters, we lose the ability to easily track and reproduce the inputs.

closed 2024-08-20 by ivanaamit