gem5/gem5 #1487: multiprefetcher

yichungchen84, 2024-08-21:
**Describe the bug**
multiprefecter failed when including 2+ prefetchers in the prefetcher list

**Affects version**
24.0.0.1

**gem5 Modifications**
None

**To Reproduce**

add a multiprefetcher
```python
    pf_list = [STeMSPrefetcher(prefetch_on_access=True), IndirectMemoryPrefetcher(prefetch_on_access=True)]
    prefetcher = MultiPrefetcher(prefetchers=pf_list)
```

**Terminal Output**
src/mem/cache/prefetch/base.hh
panic("PrefetchInfo::get called with a request with no data.");

```shell
#Terminal output here#
```

**Expected behavior**
running SE mode.

**Host Operating System**
Ubuntu 22.04

**Host ISA**
X86

erin-le, 2024-10-07:
Hello, could you provide a more thorough explanation of how to reproduce the issue? For instance, could you provide the configuration script you used, along with anything else that you think might be relevant?

yichungchen84, 2024-12-11:
Hi @erin-le, 

new update:

when you set up an simple example with multiprefetcher, you can easily reproduce the issue. Sorry that I cannot share the configuration with you due to IT restrictions. 

I found the problem, but I haven't dive into why this is the issue. 

Here is from 24.1.0.0
src/mem/cache/prefetch/base.cc line 68 "if ((!write && miss) || !pkt->hasData()) {"

in 23.1.0.0, it was "if (!write && miss) {"

if you remove the  "|| !pkt->hasData()", then multiprefetcher works. 

Hope this could help you.

erin-le, 2025-01-23:
I have reproduced the error and am in the process of testing the suggested fix. 

### Steps to reproduce

Add the following as `l2cache_multiprefetcher.py` in `src/python/gem5/components/cachehierarchies/classic/caches`

```python
# Copyright (c) 2021 The Regents of the University of California
# All rights reserved.
#
# Redistribution and use in source and binary forms, with or without
# modification, are permitted provided that the following conditions are
# met: redistributions of source code must retain the above copyright
# notice, this list of conditions and the following disclaimer;
# redistributions in binary form must reproduce the above copyright
# notice, this list of conditions and the following disclaimer in the
# documentation and/or other materials provided with the distribution;
# neither the name of the copyright holders nor the names of its
# contributors may be used to endorse or promote products derived from
# this software without specific prior written permission.
#
# THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS
# "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
# LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR
# A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT
# OWNER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL,
# SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT
# LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE,
# DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY
# THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
# (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
# OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.

from typing import Type

from m5.objects import (
    BasePrefetcher,
    Cache,
    Clusivity,
    MultiPrefetcher,
    STeMSPrefetcher,
    IndirectMemoryPrefetcher
)

from .....utils.override import *

class L2Cache(Cache):
    """
    A simple L2 Cache with default values.
    """

    def __init__(
        self,
        size: str,
        assoc: int = 16,
        tag_latency: int = 10,
        data_latency: int = 10,
        response_latency: int = 1,
        mshrs: int = 20,
        tgts_per_mshr: int = 12,
        writeback_clean: bool = False,
        clusivity: Clusivity = "mostly_incl",
        PrefetcherCls: Type[BasePrefetcher] = MultiPrefetcher,
    ):
        super().__init__()
        self.size = size
        self.assoc = assoc
        self.tag_latency = tag_latency
        self.data_latency = data_latency
        self.response_latency = response_latency
[... 356 lines cut ...]
        # exit event.
        ExitEvent.EXIT: exit_event_handler()
    },
)

simulator.run()

```

Run the simulation:

```bash
build/ALL/gem5.opt -re x86-ubuntu-run-with-kvm-no-perf-multiprefetcher.py
```

Output in simerr.txt:

```txt
src/sim/power_state.cc:105: warn: PowerState: Already in the requested power state, request ignored
src/sim/simulate.cc:199: info: Entering event queue @ 25173337164967.  Starting simulation...
src/mem/cache/prefetch/base.hh:227: panic: PrefetchInfo::get called with a request with no data.
Memory Usage: 3607124 KBytes
Program aborted at tick 25173471037293
--- BEGIN LIBC BACKTRACE ---
```

powerjg, 2025-03-11:
I'm looking into this... here's a simpler version of the code that uses a multiprefetcher.

```python
class MyCache(PrivateL1PrivateL2WalkCacheHierarchy):

    def _pre_instantiate(self, root):
        super()._pre_instantiate(root)

        # Find L2Cache objects and use multiprefetcher
        for child in self.descendants():
            if isinstance(child, L2Cache):
                pf_list = [
                    STeMSPrefetcher(prefetch_on_access=True),
                    IndirectMemoryPrefetcher(prefetch_on_access=True),
                ]
                child.prefetcher = MultiPrefetcher(prefetchers=pf_list)
```

You can add that to any script and use it as your cache hierarchy.

Unfortunately, I was unable to reproduce this bug. @erin-le was also unable to reproduce with the provided scripts above today (though they were able to reproduce before). So, maybe this bug has been fixed?

Let us know if you're still experiencing this issue with gem5-develop.

yichungchen84, 2025-03-11:
> I'm looking into this... here's a simpler version of the code that uses a multiprefetcher.
> 
> class MyCache(PrivateL1PrivateL2WalkCacheHierarchy):
> 
>     def _pre_instantiate(self, root):
>         super()._pre_instantiate(root)
> 
>         # Find L2Cache objects and use multiprefetcher
>         for child in self.descendants():
>             if isinstance(child, L2Cache):
>                 pf_list = [
>                     STeMSPrefetcher(prefetch_on_access=True),
>                     IndirectMemoryPrefetcher(prefetch_on_access=True),
>                 ]
>                 child.prefetcher = MultiPrefetcher(prefetchers=pf_list)
> You can add that to any script and use it as your cache hierarchy.
> 
> Unfortunately, I was unable to reproduce this bug. [@erin-le](https://github.com/erin-le) was also unable to reproduce with the provided scripts above today (though they were able to reproduce before). So, maybe this bug has been fixed?
> 
> Let us know if you're still experiencing this issue with gem5-develop.

I think @erin-le reproduced the problem and fixed this problem with #2003  .

closed 2025-03-11 by yichungchen84

powerjg, 2025-03-11:
FYI #2003 was not merged.