gem5/gem5 #1504: misc: bug with APIC when booting ubutnu 24.04 using `x86-linux-kernel-6.8.0-35-generic`

Harshil2107, 2024-08-22:
### Issue Description
Attempting to boot Ubuntu using the `x86-linux-kernel-6.8.0-35-generic` and `x86-ubuntu-24.04-img` for a full system simulation in gem5 results in the following kernel panic error:

```
[    0.000000] Processor #0 (Bootup-CPU)
[    0.000000] Kernel panic - not syncing: Boot CPU APIC not registered yet
[    0.000000] CPU: 0 PID: 0 Comm: swapper Not tainted 6.8.0-35-generic #35-Ubuntu
[    0.000000] Hardware name:  , BIOS  06/08/2008
```

However, booting Ubuntu with the `x86-linux-kernel-5.4.0-105-generic` kernel works correctly. This suggests a potential issue in gem5 regarding the initialization of the APIC for the newer kernel.

### Configuration Script for gem5 Simulation

```python
from gem5.coherence_protocol import CoherenceProtocol
from gem5.components.boards.x86_board import X86Board
from gem5.components.memory.single_channel import SingleChannelDDR3_1600
from gem5.components.processors.cpu_types import CPUTypes
from gem5.components.processors.simple_switchable_processor import SimpleSwitchableProcessor
from gem5.isas import ISA
from gem5.resources.resource import obtain_resource
from gem5.simulate.exit_event import ExitEvent
from gem5.simulate.simulator import Simulator
from gem5.utils.requires import requires
from gem5.components.cachehierarchies.ruby.mesi_two_level_cache_hierarchy import MESITwoLevelCacheHierarchy

requires(
    isa_required=ISA.X86,
    coherence_protocol_required=CoherenceProtocol.MESI_TWO_LEVEL,
    kvm_required=True,
)

cache_hierarchy = MESITwoLevelCacheHierarchy(
    l1d_size="16kB", l1d_assoc=8,
    l1i_size="16kB", l1i_assoc=8,
    l2_size="256kB", l2_assoc=16,
    num_l2_banks=1
)

memory = SingleChannelDDR3_1600(size="3GB")

processor = SimpleSwitchableProcessor(
    starting_core_type=CPUTypes.KVM,
    switch_core_type=CPUTypes.TIMING,
    isa=ISA.X86,
    num_cores=2
)

board = X86Board(
    clk_freq="3GHz",
    processor=processor,
    memory=memory,
    cache_hierarchy=cache_hierarchy
)

board.set_kernel_disk_workload(
    kernel=obtain_resource("x86-linux-kernel-6.8.0-35-generic"),
    disk_image=obtain_resource("x86-ubuntu-24.04-img"),
    kernel_args=[
        "earlyprintk=ttyS0", "console=ttyS0", "lpj=7999923", "root=/dev/sda2"
    ]
)

def exit_event_handler():
    print("First exit: kernel booted")
    yield False
    print("Second exit: Started `after_boot.sh` script")
    processor.switch()
    yield False
    print("Third exit: Finished `after_boot.sh` script")
    yield True

simulator = Simulator(
    board=board,
    on_exit_event={ExitEvent.EXIT: exit_event_handler()}
)

simulator.run()
```

Harshil2107, 2024-09-26:
The PR #1574, when run with this kernel and the 24.04 image moves past this error but still fails with the panic:

```
src/arch/x86/interrupts.cc:530: hack: Assuming logical destinations are 1 << id.
src/dev/pci/host.cc:171: panic: panic condition !pci_dev occurred: 00:1f.1: Write to config space on non-existent PCI device
Memory Usage: 4458472 KBytes
Program aborted at tick 664484273244
--- BEGIN LIBC BACKTRACE ---
./build/ALL/gem5.opt(_ZN4gem515print_backtraceEv+0x30)[0x556d09c06c50]
./build/ALL/gem5.opt(_ZN4gem512abortHandlerEi+0x4c)[0x556d09c2c6bc]
/lib/x86_64-linux-gnu/libc.so.6(+0x42520)[0x712a25e42520]
/lib/x86_64-linux-gnu/libc.so.6(pthread_kill+0x12c)[0x712a25e969fc]
/lib/x86_64-linux-gnu/libc.so.6(raise+0x16)[0x712a25e42476]
/lib/x86_64-linux-gnu/libc.so.6(abort+0xd3)[0x712a25e287f3]
./build/ALL/gem5.opt(+0x1e897d5)[0x556d090d47d5]
./build/ALL/gem5.opt(_ZN4gem514GenericPciHost5writeEPNS_6PacketE+0x1c2)[0x556d09147a42]
./build/ALL/gem5.opt(_ZThn64_N4gem57PioPortINS_9PioDeviceEE10recvAtomicEPNS_6PacketE+0x76)[0x556d090d4606]
./build/ALL/gem5.opt(_ZN4gem515NoncoherentXBar18recvAtomicBackdoorEPNS_6PacketEsPPNS_11MemBackdoorE+0x39c)[0x556d09f0a2ac]
./build/ALL/gem5.opt(_ZN4gem54ruby8RubyPort15MemResponsePort10recvAtomicEPNS_6PacketE+0x26d)[0x556d0a2e402d]
./build/ALL/gem5.opt(_ZN4gem515AtomicSimpleCPU8writeMemEPhjmNS_5FlagsImEEPmRKSt6vectorIbSaIbEE+0x6c4)[0x556d09959734]
./build/ALL/gem5.opt(_ZN4gem517SimpleExecContext8writeMemEPhjmNS_5FlagsImEEPmRKSt6vectorIbSaIbEE+0x59)[0x556d09975a89]
./build/ALL/gem5.opt(+0x37f660f)[0x556d0aa4160f]
./build/ALL/gem5.opt(_ZNK4gem510X86ISAInst2St7executeEPNS_11ExecContextEPNS_5trace10InstRecordE+0x168)[0x556d0aa7ceb8]
./build/ALL/gem5.opt(_ZN4gem515AtomicSimpleCPU4tickEv+0x39d)[0x556d099573ad]
./build/ALL/gem5.opt(_ZN4gem510EventQueue10serviceOneEv+0xc2)[0x556d09c1c992]
./build/ALL/gem5.opt(_ZN4gem59doSimLoopEPNS_10EventQueueE+0x68)[0x556d09c46988]
./build/ALL/gem5.opt(_ZN4gem58simulateEm+0x283)[0x556d09c46f83]
./build/ALL/gem5.opt(+0x2b99660)[0x556d09de4660]
./build/ALL/gem5.opt(+0x1e6dd54)[0x556d090b8d54]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x128023)[0x712a26d28023]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_Call+0x5c)[0x712a26ce1fec]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x4b16)[0x712a26c76776]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af)[0x712a26dc23af]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9d68)[0x712a26c7b9c8]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af)[0x712a26dc23af]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x829e)[0x712a26c79efe]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af)[0x712a26dc23af]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyEval_EvalCode+0xbe)[0x712a26dbd3de]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1bd96d)[0x712a26dbd96d]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1287b3)[0x712a26d287b3]
--- END LIBC BACKTRACE ---
For more info on how to address this issue, please visit https://www.gem5.org/documentation/general_docs/common-errors/ 

Aborted (core dumped)
```

The terminal output of the boot looks like the following:
```
[    0.000000] Linux version 6.8.0-35-generic (buildd@lcy02-amd64-020) (x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0, GNU ld (GNU Binutils for Ubuntu) 2.42) #35-Ubuntu SMP PREEMPT_DYNAMIC Mon May 20 15:51:52 UTC 2024 (Ubuntu 6.8.0-35.35-generic 6.8.4)
[    0.000000] Command line: earlyprintk=ttyS0 console=ttyS0 lpj=7999923 root=/dev/sda2
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000bfffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000ffff0000-0x00000000ffffffff] reserved
[    0.000000] printk: legacy bootconsole [earlyser0] enabled
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] APIC: Static calls initialized
[    0.000000] SMBIOS 2.5 present.
[    0.000000] DMI:  , BIOS  06/08/2008
[    0.000000] tsc: Fast TSC calibration failed
[    0.000000] last_pfn = 0xc0000 max_arch_pfn = 0x400000000
[    0.000000] MTRRs disabled by BIOS
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[... 153 lines cut ...]
[    0.402000] NetLabel:  domain hash size = 128
[    0.404000] NetLabel:  protocols = UNLABELED CIPSOv4 CALIPSO
[    0.407000] NetLabel:  unlabeled traffic allowed by default
[    0.410000] mctp: management component transport protocol core
[    0.413000] NET: Registered PF_MCTP protocol family
[    0.415000] PCI: Probing PCI hardware
[    0.417000] PCI host bridge to bus 0000:00
[    0.419000] pci_bus 0000:00: root bus resource [io  0x0000-0xffff]
[    0.422000] pci_bus 0000:00: root bus resource [mem 0x00000000-0xffffffffffff]
[    0.425000] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
[    0.429000] pci 0000:00:04.0: [8086:7111] type 00 class 0x010180 conventional PCI endpoint
[    0.434000] pci 0000:00:04.0: BAR 4 [io  0x0000-0x000f]
[    0.437000] pci 0000:00:04.0: ROM [mem 0x00000000-0x000007ff pref]
[... cut ...]
[    0.445000] pci 0000:00:04.0: BAR 2 [io  0x0170-0x0177]: legacy IDE quirk
[    0.448000] pci 0000:00:04.0: BAR 3 [io  0x0376]: legacy IDE quirk
[    0.452000] pci_bus 0000:00: busn_res: [bus 00-ff] end is updated to 00
[    0.455000] vgaarb: loaded
[    0.457000] clocksource: Switched to clocksource refined-jiffies
[    0.461999] VFS: Disk quotas dquot_6.6.0
[    0.463998] VFS: Dquot-cache hash table entries: 512 (order 0, 4096 bytes)
[    0.467998] AppArmor: AppArmor Filesystem Enabled
[    0.469998] pnp: PnP ACPI: disabled
```

abmerop, 2024-09-26:
Glad I am not the only one seeing this, I was starting to suspect it was GPU related.  I haven't figured out what this unknown PCI device is.

I have been using a workaround: https://github.com/abmerop/gem5/commit/eb1235f6ee41c2e70d62a16dd61f06834253b409

Harshil2107, 2024-10-01:
Using the change that you mentioned I was able to boot ubuntu with the new kernel. It seems like there is only one pci that is connected, when I run `lspci` on the guest after booting.

```
gem5@gem5:~$ lspci
00:04.0 IDE interface: Intel Corporation 82371AB/EB/MB PIIX4 IDE
```

closed 2024-10-17 by Harshil2107