gem5/gem5 #1507: gem5 crashes with panic: 'Tried to read unmapped address' on x86 when reading from argv

dakaidan, 2024-08-23:
**Describe the bug**
gem5 crashes with panic 'Tried to read unmapped address' when using `fscanf` or reading beyond `argc` on X86, instead of returning an error.

**Affects version**

gem5 version DEVELOP-FOR-24.1
gem5 compiled Jul 17 2024 14:27:05

**gem5 Modifications**
No modifications. I used the script hello-custom-binary-Ex.py modified from this script [https://github.com/BobbyRBruce/gem5-ssbse-challenge-2023](https://github.com/BobbyRBruce/gem5-ssbse-challenge-2023), to allow input. The script is available here: [https://github.com/karineek/SearchGEM5/blob/main/hello-custom-binary-Ex.py](https://github.com/karineek/SearchGEM5/blob/main/hello-custom-binary-Ex.py).

**To Reproduce**
Steps to reproduce the behavior. Please assume starting from a clean repository:

1. Compile gem5 with command 
```shell
git clone https://github.com/gem5/gem5.git
cd gem5/
pip install -r requirements.txt
scons build/X86/gem5.opt -j 4
```
2. Compile the code:
```shell
gcc -O3 test_panic_01.c -o test_panic_01.c.o
gcc -O3 test_panic_02.c -o test_panic_02.c.o
```
3. Prepare input files for running the code:
```shell
echo "$pwd"/test_panic_01.c.o >> test_panic_01.txt
echo 0 >> test_panic_01.txt

echo "$pwd"/test_panic_02.c.o >> test_panic_02.txt
echo 0 >> test_panic_02.txt
```
4. Run with:
```shell
./gem5/build/X86/gem5.opt ssbse-challenge-examples/hello-custom-binary-Ex.py --isa X86 --input test_panic_01.txt

./gem5/build/X86/gem5.opt ssbse-challenge-examples/hello-custom-binary-Ex.py --isa X86 --input test_panic_02.txt
```

The code for `test_panic_01.c` is:
```C
#include <stdio.h>
#define MAX_ARG 10

int main(int argc, char *argv[]) {
	for (int i = 1; i <= MAX_ARG; i++) {
		if (argv[i] == "-p") {
			printf("Pruning\n");
		} else if (argv[i] == "-f") {
			printf("Frontend implementation\n");
		} else if (argv[i] == "-l") {
			printf("Labeled statement\n");
		} else {
			printf("Invalid input: %s\n", argv[i]);
		}
	}
	
	return 0;
}
```

The code for `test_panic_02.c` is:
```C
#include <stdio.h>

int main(int argc, char* argv[]) {
	int num1;
	num1 = fscanf(argv[1], "%d", &num1);
[... 152 lines cut ...]

**Host ISA**
X86

**Compiler used**

GCC version used:
```shell
gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

SCons version used:
```shell
SCons by Steven Knight et al.:
	SCons: v4.7.0.265be6883fadbb5a545612265acc919595158366, Sun, 17 Mar 2024 17:33:54 -0700, by bdbaddog on M1Dog2021
	SCons path: ['/home/aidan/.local/lib/python3.10/site-packages/SCons']
Copyright (c) 2001 - 2024 The SCons Foundation
```

**Additional information**
In the case of `test_panic_01` the fault occurs on any inputs where there are less than 10 total inputs.
For `test_panic_02` the fault occurs under any input.

erin-le, 2025-01-16:
Since https://github.com/gem5/gem5/issues/1483 is very similar to this issue, it is being consolidated into this one.

#### To reproduce:

1. Build gem5 from source:
```bash
git clone https://github.com/gem5/gem5.git
cd gem5/
pip install -r requirements.txt
scons build/X86/gem5.opt -j 4
```

2. Compile the code:
```bash
gcc-11 -O3 test_abort_01.c -o test_abort_01.c.o
```
and create a text file to contain input:

```bash
:~$ more test_abort_01.txt
/home/ubuntu/test_abort_01.c.o
0
```
 
3. Run  

```bash
./gem5/build/X86/gem5.opt ssbse-challenge-examples/hello-custom-binary-Ex.py --isa X86 --input test_abort_01.txt
```

The code is:
```c
#include <stdio.h>
#include <stdlib.h>
#define MAXARGS 5

int main(int argc, char *argv[]) {
    int i, j;

    if (argc > MAXARGS) {
        printf("Error: too many arguments\n");
        exit(EXIT_FAILURE);
    }
    
    for (i = 0; i < argc - 1; i++) {
        sprintf(argv[i + 1], "%s%s", argv[i], argv[i]); // Constant folding to remove spaces and commas in the argument list
    }
    
    printf("%s\n", argv[argc - 1]);
    
    return 0;
}
```

Python config script used in Issue 1483 is the same as in this PR.

#### Terminal Output

Terminal output in Issue 1483 is similar to the outputs in this PR.

#### Expected Behavior

Running
```bash
$ ./test_abort_01.c.o 0
```
results in the following being printed:

```txt
./test_abort_01.c.o./test_abort_01.c.o
```
#### Host Operating System
Ubuntu 20.04, x86_64-pc-linux-gnu

#### Host ISA
X86

#### Compiler used
gcc 11.4.0

#### Additional Information

The abort happens also with other inputs such as "1" or "2 2 2" and so on. It might be related to how the simulated environment that the .o file runs on, looks like.

victoryang00, 2025-04-08:
Have this been resolved? I encounter the same issue here.

powerjg, 2025-04-17:
A panic is the expected result for these SE executions. In syscall emulation mode, there is no OS, so there is nothing to catch page faults. Therefore, when a page fault occurs (e.g., reading beyond argc), the result is a panic.

closed 2025-04-17 by powerjg