gem5/gem5 #1508: Panic error when executing `avx2` on X86 due to Unrecognized/invalid instruction executed

dakaidan, 2024-08-23:
**Describe the bug**
gem5 crashes with panic 'Unrecognized/invalid instruction executed' on X86 when executing `avx2` instructions. 

**Affects version**

gem5 version DEVELOP-FOR-24.1
gem5 compiled Jul 17 2024 14:27:05

**gem5 Modifications**
No modifications. I used the script hello-custom-binary-Ex.py modified from this script [https://github.com/BobbyRBruce/gem5-ssbse-challenge-2023](https://github.com/BobbyRBruce/gem5-ssbse-challenge-2023), to allow input. The script is available here: [https://github.com/karineek/SearchGEM5/blob/main/hello-custom-binary-Ex.py](https://github.com/karineek/SearchGEM5/blob/main/hello-custom-binary-Ex.py).

**To Reproduce**
Steps to reproduce the behavior. Please assume starting from a clean repository:

1. Compile gem5 with command 
```shell
git clone https://github.com/gem5/gem5.git
cd gem5/
pip install -r requirements.txt
scons build/X86/gem5.opt -j 4
```
2. Compile the code:
```shell
gcc -O3 test_vpbroadcastd.c -o test_vpbroadcastd.c.o
```
3. Prepare input files for running the code:
```shell
echo "$pwd"/test_vpbroadcastd.c.o >> test_vpbroadcastd.txt
echo 6 >> test_vpbroadcastd.txt
```
4. Run with:
```shell
./gem5/build/X86/gem5.opt ssbse-challenge-examples/hello-custom-binary-Ex.py --isa X86 --input test_vpbroadcastd.txt
```

The code for `test_panic_01.c` is:
```C
#include <stdio.h>
#include <immintrin.h>

int main() {
    int value = 1234;
    __m256i result;
    
    asm volatile (
        "vpbroadcastd %[src], %[dest]"
        : [dest] "=x" (result)   
        : [src] "r" (value)        
    );
    
    int output[8];
    _mm256_storeu_si256((__m256i*)output, result);

    printf("Broadcasted values:\n");
    for (int i = 0; i < 8; i++) {
        printf("%d ", output[i]);
    }
    printf("\n");

    return 0;
}

```

The __Python configuration script__ can be found here: [https://github.com/karineek/SearchGEM5/blob/main/hello-custom-binary-Ex.py](https://github.com/karineek/SearchGEM5/blob/main/hello-custom-binary-Ex.py)

**Terminal Output**

```shell
./gem5/build/X86/gem5.opt ssbse-challenge-examples/hello-custom-binary-Ex.py --isa X86 --input test_vpbroadcastd.txt
gem5 Simulator System.  https://www.gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version DEVELOP-FOR-24.1
gem5 compiled Jul 17 2024 14:27:05
gem5 started Aug 23 2024 14:38:20
gem5 executing on aidans-laptop, pid 1434528
command line: ../gem5-develop/build/X86/gem5.opt ../ASEgem5/hello-custom-binary-Ex.py --isa X86 --input vpbroadcastd.txt

Global frequency set at 1000000000000 ticks per second
warn: No dot file generated. Please install pydot to generate the dot file and pdf.
src/mem/dram_interface.cc:690: warn: DRAM device capacity (8192 Mbytes) does not match the address range assigned (32 Mbytes)
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
board.remote_gdb: Listening for connections on port 7000
src/sim/simulate.cc:199: info: Entering event queue @ 0.  Starting simulation...
src/sim/mem_state.cc:448: info: Increasing stack size by one page.
src/sim/syscall_emul.cc:74: warn: ignoring syscall mprotect(...)
src/sim/syscall_emul.cc:74: warn: ignoring syscall set_robust_list(...)
src/sim/syscall_emul.cc:74: warn: ignoring syscall rseq(...)
src/sim/syscall_emul.cc:74: warn: ignoring syscall mprotect(...)
src/sim/syscall_emul.cc:74: warn: ignoring syscall mprotect(...)
src/sim/syscall_emul.cc:74: warn: ignoring syscall mprotect(...)
src/arch/x86/faults.cc:131: panic: Unrecognized/invalid instruction executed:
 
{
	leg = 0,
	rex = 0,
	vex/xop = 0,
	op = {
		type = one byte,
		op = 0x62,
		},
	modRM = 0xf2,
	sib = 0,
	immediate = 0,
	displacement = 0
	dispSize = 0}
Memory Usage: 178872 KBytes
Program aborted at tick 6916148595
--- BEGIN LIBC BACKTRACE ---
../gem5-develop/build/X86/gem5.opt(_ZN4gem515print_backtraceEv+0x30)[0x62e413133130]
../gem5-develop/build/X86/gem5.opt(_ZN4gem512abortHandlerEi+0x4c)[0x62e41315768c]
/lib/x86_64-linux-gnu/libc.so.6(+0x42520)[0x705438442520]
/lib/x86_64-linux-gnu/libc.so.6(pthread_kill+0x12c)[0x7054384969fc]
/lib/x86_64-linux-gnu/libc.so.6(raise+0x16)[0x705438442476]
/lib/x86_64-linux-gnu/libc.so.6(abort+0xd3)[0x7054384287f3]
../gem5-develop/build/X86/gem5.opt(+0xa2f5c5)[0x62e411bb75c5]
../gem5-develop/build/X86/gem5.opt(_ZN4gem56X86ISA13InvalidOpcode6invokeEPNS_13ThreadContextERKNS_14RefCountingPtrINS_10StaticInstEEE+0x7a)[0x62e411c0a6ea]
../gem5-develop/build/X86/gem5.opt(_ZN4gem513BaseSimpleCPU9advancePCERKSt10shared_ptrINS_9FaultBaseEE+0xde)[0x62e412697ade]
../gem5-develop/build/X86/gem5.opt(_ZN4gem515TimingSimpleCPU11advanceInstERKSt10shared_ptrINS_9FaultBaseEE+0xc3)[0x62e41268d473]
../gem5-develop/build/X86/gem5.opt(_ZN4gem515TimingSimpleCPU14completeIfetchEPNS_6PacketE+0x2b8)[0x62e41268e498]
../gem5-develop/build/X86/gem5.opt(_ZN4gem510EventQueue10serviceOneEv+0x16d)[0x62e413147a3d]
../gem5-develop/build/X86/gem5.opt(_ZN4gem59doSimLoopEPNS_10EventQueueE+0x68)[0x62e413170498]
../gem5-develop/build/X86/gem5.opt(_ZN4gem58simulateEm+0x283)[0x62e413170a93]
../gem5-develop/build/X86/gem5.opt(+0x1f37e30)[0x62e4130bfe30]
../gem5-develop/build/X86/gem5.opt(+0xa41324)[0x62e411bc9324]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x128023)[0x705438d28023]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_Call+0x5c)[0x705438ce1fec]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x4b16)[0x705438c76776]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af)[0x705438dc23af]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9d68)[0x705438c7b9c8]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af)[0x705438dc23af]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x829e)[0x705438c79efe]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af)[0x705438dc23af]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyEval_EvalCode+0xbe)[0x705438dbd3de]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1bd96d)[0x705438dbd96d]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1287b3)[0x705438d287b3]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x69de)[0x705438c7863e]
/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af)[0x705438dc23af]
../gem5-develop/build/X86/gem5.opt(+0x1f1e2f7)[0x62e4130a62f7]
../gem5-develop/build/X86/gem5.opt(main+0x213)[0x62e411b9d193]
/lib/x86_64-linux-gnu/libc.so.6(+0x29d90)[0x705438429d90]
--- END LIBC BACKTRACE ---
For more info on how to address this issue, please visit https://www.gem5.org/documentation/general_docs/common-errors/ 

[1]    1434528 IOT instruction (core dumped)  ../gem5-develop/build/X86/gem5.opt ../ASEgem5/hello-custom-binary-Ex.py --isa
```

**Expected behavior**
For gem5 not to crash with panic, and execute the broadcast as expected. e.g. resulting in the output:
```shell
Broadcasted values:
1234 1234 1234 1234 1234 1234 1234 1234
```

**Host Operating System**
Ubuntu 22.04

**Host ISA**
X86

**Compiler used**

GCC version used:
```shell
gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

SCons version used:
```shell
SCons by Steven Knight et al.:
	SCons: v4.7.0.265be6883fadbb5a545612265acc919595158366, Sun, 17 Mar 2024 17:33:54 -0700, by bdbaddog on M1Dog2021
	SCons path: ['/home/aidan/.local/lib/python3.10/site-packages/SCons']
Copyright (c) 2001 - 2024 The SCons Foundation
```

**Additional information**
This works with many other instructions from the `avx2` family, but not from `avx` and `avx512`.

powerjg, 2024-09-11:
Just FYI, this is a known issue. There are many x86 instructions in AVX, AVX2, and some in SSE, etc. which we have not implemented. This would be a great way for someone like you to contribute to the project!

karineek, 2024-09-12:
Thanks for the feedback! We can turn it into a student final year project, but the requirement of the university is that there will be an organisation or a company to support the cause/case (not financially but for advising and guidance). Do you know who might be interested in x86 feature implementation?

arkhadem, 2024-10-18:
I'm having the same error with vbroadcastss. The auto-vectorizer creates it. Any workaround?

abmerop, 2025-07-07:
> Thanks for the feedback! We can turn it into a student final year project, but the requirement of the university is that there will be an organisation or a company to support the cause/case (not [...]

Just noticed this comment.  I would be willing to advise/guide depending on how "official" it needs to be.  Several colleagues at AMD have been asking about AVX2 support.

abmerop, 2025-07-07:
> I also have a similar problem when I try to simulate an HPC application that uses AVX512 instructions. I wish that gem5 would support avx512 in the future.

There have been at least two implementations of AVX512, but they have not been merged into the upstream.  I can't remember why, but I believe it was either that the two implementations were very different and a decision needed to be made which implementation to choose or that the code was provide "as-is" but there were changes requested which would require someone else to address the change requests.

mattsinc, 2025-07-08:
> There have been at least two implementations of AVX512, but they have not been merged into the upstream. I can't remember why, but I believe it was either that the two implementations were very [...]

The students in my class who did #1877 had tried to integrate those models (e.g., the one from UCLA) into the mainline of gem5 and found there were a ton of issues -- e.g., the mainline of gem5 had diverged so much the patches could not be applied cleanly, and it was so painful trying to figure out how to overcome this they felt starting over was better.  I do have their reports (and eventually merging #1877 in ...) to base a new effort off of, and in the Fall I may be able to get another student group to try again (perhaps with @abmerop giving more formal feedback) but I think the recurring problem has been that no one who is expert enough at gem5 has sufficient AVX expertise to get this support over the line.  Perhaps having you (@abmerop) help with this would be one way around this?

@arkhadem, @karineek  : agreed this would be a nice feature, if you are willing to take it on, that would be great!  But if it's something you want others to do, it's always tricky because others (like you) have many other responsibilities too.  Sadly, this has been the case with AVX support for a while.  With @abmerop I'm certainly happy to help how I can if either of you are willing to push this forward.