gem5/gem5 #1529: gem5 compilation fails when using GCC 12 due to pybind

erin-le, 2024-08-30:
**Describe the bug**
gem5 compilation fails when using GCC 12 due to an issue with pybind11. This occurs with both pybind version 2.10.3, which is currently used in gem5, and in version 2.13.5, the latest version of pybind. 

**Affects version**
gem5 develop, [29d6b46](https://github.com/gem5/gem5/commit/29d6b46f1f8d6e5d75a15fce5f6b03b3fbc974ec)

**gem5 Modifications**
None

**To Reproduce**
Steps to reproduce the behavior. Please assume starting from a clean repository:

1. Run the following command, replacing /path/to/gem5 with the path to your gem5 directory:
```
docker run -u $UID:$GID --volume /path/to/gem5:/gem5 --rm -it ghcr.io/gem5/gcc-version-12:latest
```

2.  Compile gem5 with 
```
scons build/ALL/gem5.fast -j (number of cores)
```

**Terminal Output**

```shell
 [     CXX] src/python/pybind11/core.cc -> pybind_update/build/ALL/python/pybind11/core.fo
In file included from /usr/include/c++/12/bits/specfun.h:45,
                 from /usr/include/c++/12/cmath:1935,
                 from /usr/include/c++/12/math.h:36,
                 from /usr/include/python3.12/pyport.h:195,
                 from /usr/include/python3.12/Python.h:38,
                 from ext/pybind11/include/pybind11/detail/common.h:274,
                 from ext/pybind11/include/pybind11/attr.h:13,
                 from ext/pybind11/include/pybind11/detail/class.h:12,
                 from ext/pybind11/include/pybind11/pybind11.h:13,
                 from ext/pybind11/include/pybind11/operators.h:12,
                 from src/python/pybind11/core.cc:42:
In static member function ‘static _Up* std::__copy_move<_IsMove, true, std::random_access_iterator_tag>::__copy_m(_Tp*, _Tp*, _Up*) [with _Tp = pybind11::detail::type_info* const; _Up = pybind11::detail::type_info*; bool _IsMove = false]’,
    inlined from ‘_OI std::__copy_move_a2(_II, _II, _OI) [with bool _IsMove = false; _II = pybind11::detail::type_info* const*; _OI = pybind11::detail::type_info**]’ at /usr/include/c++/12/bits/stl_algobase.h:501:30,
    inlined from ‘_OI std::__copy_move_a1(_II, _II, _OI) [with bool _IsMove = false; _II = pybind11::detail::type_info* const*; _OI = pybind11::detail::type_info**]’ at /usr/include/c++/12/bits/stl_algobase.h:528:42,
    inlined from ‘_OI std::__copy_move_a(_II, _II, _OI) [with bool _IsMove = false; _II = pybind11::detail::type_info* const*; _OI = pybind11::detail::type_info**]’ at /usr/include/c++/12/bits/stl_algobase.h:535:31,
    inlined from ‘_OI std::copy(_II, _II, _OI) [with _II = pybind11::detail::type_info* const*; _OI = pybind11::detail::type_info**]’ at /usr/include/c++/12/bits/stl_algobase.h:626:7,
    inlined from ‘static _ForwardIterator std::__uninitialized_copy<true>::__uninit_copy(_InputIterator, _InputIterator, _ForwardIterator) [with _InputIterator = pybind11::detail::type_info* const*; _ForwardIterator = pybind11::detail::type_info**]’ at /usr/include/c++/12/bits/stl_uninitialized.h:147:27,
    inlined from ‘_ForwardIterator std::uninitialized_copy(_InputIterator, _InputIterator, _ForwardIterator) [with _InputIterator = pybind11::detail::type_info* const*; _ForwardIterator = pybind11::detail::type_info**]’ at /usr/include/c++/12/bits/stl_uninitialized.h:185:15,
    inlined from ‘_ForwardIterator std::__uninitialized_copy_a(_InputIterator, _InputIterator, _ForwardIterator, allocator<_Tp>&) [with _InputIterator = pybind11::detail::type_info* const*; _ForwardIterator = pybind11::detail::type_info**; _Tp = pybind11::detail::type_info*]’ at /usr/include/c++/12/bits/stl_uninitialized.h:372:37,
    inlined from ‘void std::vector<_Tp, _Alloc>::_M_assign_aux(_ForwardIterator, _ForwardIterator, std::forward_iterator_tag) [with _ForwardIterator = pybind11::detail::type_info* const*; _Tp = pybind11::detail::type_info*; _Alloc = std::allocator<pybind11::detail::type_info*>]’ at /usr/include/c++/12/bits/vector.tcc:339:35,
    inlined from ‘std::vector<_Tp, _Alloc>& std::vector<_Tp, _Alloc>::operator=(std::initializer_list<_Tp>) [with _Tp = pybind11::detail::type_info*; _Alloc = std::allocator<pybind11::detail::type_info*>]’ at /usr/include/c++/12/bits/stl_vector.h:787:21,
    inlined from ‘pybind11::detail::generic_type::initialize(const pybind11::detail::type_record&)::<lambda(pybind11::detail::internals&)>’ at ext/pybind11/include/pybind11/pybind11.h:1434:75,
    inlined from ‘decltype (cb(pybind11::detail::get_internals())) pybind11::detail::with_internals(const F&) [with F = generic_type::initialize(const pybind11::detail::type_record&)::<lambda(internals&)>]’ at ext/pybind11/include/pybind11/detail/internals.h:641:14,
    inlined from ‘void pybind11::detail::generic_type::initialize(const pybind11::detail::type_record&)’ at ext/pybind11/include/pybind11/pybind11.h:1426:23:
/usr/include/c++/12/bits/stl_algobase.h:434:30: error: ‘void* __builtin_memcpy(void*, const void*, long unsigned int)’ forming offset 8 is out of the bounds [0, 8] of object ‘<anonymous>’ with type ‘pybind11::detail::type_info* const [1]’ [-Werror=array-bounds]
  434 |             __builtin_memmove(__result, __first, sizeof(_Tp) * _Num);
      |             ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
ext/pybind11/include/pybind11/pybind11.h: In member function ‘void pybind11::detail::generic_type::initialize(const pybind11::detail::type_record&)’:
ext/pybind11/include/pybind11/pybind11.h:1434:75: note: ‘<anonymous>’ declared here
 1434 |             internals.registered_types_py[(PyTypeObject *) m_ptr] = {tinfo};
      |                                                                           ^
cc1plus: all warnings being treated as errors
scons: *** [pybind_update/build/ALL/python/pybind11/core.fo] Error 1
scons: building terminated because of errors.
```

**Expected behavior**
For gem5 to finish compiling.

**Host Operating System**
Ubuntu 24.04

**Host ISA**
X86
**Compiler used**
GCC 12

BobbyRBruce, 2024-09-04:
I made a patch for this bug here: https://github.com/gem5/gem5/pull/1501.

I don't like modifying Pybind like this but it does work.

BobbyRBruce, 2024-09-08:
I created a PR to Pybind11 to implement this fix in their repo: https://github.com/pybind/pybind11/pull/5355. Hopefully they merge it and we can use the next the release of PyBind as-is. I'd advise merging #1501 for now

abmerop, 2024-09-09:
Thanks Bobby! I was running into this issue on a new CI machine which has GCC 12 as the default, so this is super helpful.

ivanaamit, 2024-10-18:
Closing this issue since both PRs (#1501 and https://github.com/pybind/pybind11/pull/5355) have been merged. Feel free to reopen it if I’m mistaken. Thanks!

powerjg, 2024-10-18:
Unless I'm wrong, this should stay open until we integrate the new pybind (or we should have an issue for that). It is not good to diverge our pybind from upstream.

Has https://github.com/pybind/pybind11/pull/5355 been merged in https://github.com/pybind/pybind11/releases/tag/v2.13.6?

ivanaamit, 2024-10-28:
It hasn’t been included in the latest release. I'll make a note to keep an eye on the pybind releases, and once it’s merged, I’ll update this issue.