gem5/gem5 #1536: No error reported but fail to link or generate an .opt file.

liuyuxivapor, 2024-09-04:
**Describe the bug**
No error reported but fail to link or generate an .opt file.

**Affects version**
24.0

**gem5 Modifications**
Nothing changed compared to the repo.

**To Reproduce**
Steps to reproduce the behavior. Please assume starting from a clean repository:

1. Compile gem5 with command `scons build/X86/gem5.opt`

**Terminal Output**

```shell
<lots of outputs>
 [SO Param] m5.objects.Sequencer, RubyPort -> X86/python/_m5/param_RubyPort.cc
 [     CXX] src/base/date.cc -> X86/base/date.o
scons: done building targets.
*** Summary of Warnings ***
Warning: Header file <capstone/capstone.h> not found.
         This host has no capstone library installed.
Warning: Couldn't find HDF5 C++ libraries. Disabling HDF5 support.
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Host Operating System**
Ubuntu 20.04

**Host ISA**
X86

**Additional information**
It seems that I have installed all requirements, but it still appears no error!

Harshil2107, 2024-09-04:
It looks like you are building gem5 with just 1 thread so it will take a long time. I would recommend trying to build gem5 with more than one thread using the `-j` command. you can use `nproc` to get the total number of threads you have on your system. To build gem5 you can use `scons build/X86/gem5.opt -j16` for example.

liuyuxivapor, 2024-09-05:
> It looks like you are building gem5 with just 1 thread so it will take a long time. I would recommend trying to build gem5 with more than one thread using the `-j` command. you can use `nproc` to [...]

I have tried this with `-j10`, but got the same problem. :( Thanks a lot anyway.

giactra, 2024-09-08:
Have you tried to redirect stderr to file to check if there is a linkage error line you have missed in the terminal?

`scons build/X86/gem5.opt -j10 2> error.log`

Mr-Cong, 2024-09-14:
The same error.Nothing generated.No error.

Harshil2107, 2024-09-16:
It looks like the issue was solved. I am closing the issue as resolved.

liuyuxivapor, 2024-09-21:
> Have you tried to redirect stderr to file to check if there is a linkage error line you have missed in the terminal?
> 
> `scons build/X86/gem5.opt -j10 2> error.log`

Acctually I have done this before, but no error occurred.

abmerop, 2024-09-21:
I was also seeing this issue, mainly when changing branches a lot, but wasn't able to find a fix. I basically had to delete the build directory and recompile. I am commenting to acknowledge that this issue exists for others too.

erin-le, 2024-10-31:
Duplicate

closed 2024-10-31 by erin-le