gem5/gem5 #1547: Missing Locale support within gem5 simulator on ARM

dakaidan, 2024-09-07:
**Describe the bug**
Gem5 cannot set the locale to `C.UTF-8`, gem5 appears to only support `C` locale, and also defaults to it, while native hardware appears to default to `C.UTF-8`.

**Affects version**
gem5 version 24.0.0.1
gem5 compiled Sep  6 2024 03:51:59

**gem5 Modifications**
No modifications. I used the script hello-custom-binary-Ex.py modified from this script [https://github.com/BobbyRBruce/gem5-ssbse-challenge-2023](https://github.com/BobbyRBruce/gem5-ssbse-challenge-2023), to allow input. The script is available here: [https://github.com/karineek/SearchGEM5/blob/main/hello-custom-binary-Ex.py](https://github.com/karineek/SearchGEM5/blob/main/hello-custom-binary-Ex.py), which has an ARM ISA option.

**To Reproduce**
Steps to reproduce the behavior:

1. Compile gem5 with command:
```shell
git clone https://github.com/gem5/gem5.git
cd gem5/
pip install -r requirements.txt
scons build/ARM/gem5.opt
```
2. Compile the code with:
```shell 
gcc -O3 test_locale.c -o test_locale.o
```
3. Create a text file for the input:
```shell
echo "$pwd"/test_locale.o >> test_locale.txt
echo 20ac >> test_locale.txt
```
5. Run with:
```shell
./gem5/build/ARM/gem5.opt ./hello-custom-binary-Ex.py --isa ARM --input test_locale.txt
```

The programs code is:

```c
#include <stdio.h>
#include <uchar.h>
#include <stdlib.h>
#include <locale.h>
#include <string.h>

int main(int argc, char *argv[]) {
    char *locale = setlocale(LC_ALL, "C.UTF-8");
    if (locale == NULL) {
        printf("Error: Unable to set locale.\n");
        return EXIT_FAILURE;
    }
    
    printf("Current Locale: %s\n", locale);
      
    return EXIT_SUCCESS;
}
```

Or in this case:
```C
#include <stdio.h>
#include <uchar.h>
#include <stdlib.h>
#include <locale.h>

int main(int argc, char *argv[]) {
    setlocale(LC_ALL, "");
    
    if (argc != 2) {
        printf("Usage: %s <UTF-16 character>\n", argv[0]);
        return EXIT_FAILURE;
    }
[... 162 lines cut ...]
```

**Expected behavior**
For the program to default to a similar encoding within the simulator as is done outside of it (`C.UTF-8`), and for the simulator to support explicitly setting locales other than `C`.

**Host Operating System**
Ubuntu 22.04.4 LTS 

**Host ISA**
ARMv8

**Compiler used**
```shell
> gcc --version
gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

> scons --version
SCons by Steven Knight et al.:
	SCons: v4.0.1.c289977f8b34786ab6c334311e232886da7e8df1, 2020-07-17 01:50:03, by bdbaddog on ProDog2020
	SCons path: ['/usr/lib/python3/dist-packages/SCons']
Copyright (c) 2001 - 2020 The SCons Foundation
```

giactra, 2024-09-08:
Hi @dakaidan , just to clarify, you are also running the simulator on the same Arm host?

dakaidan, 2024-09-10:
Hi @giactra, yes, the simulator, and the native run were done on the same host.

giactra, 2024-09-12:
> Hi @giactra, yes, the simulator, and the native run were done on the same host.

What happens if you run gem5 with --debug-flags=SyscallBase ?
This will give you a strace of the program.

In particular I am interested to know which file gem5 is trying to access to get the list of locales.

dakaidan, 2024-09-18:
> What happens if you run gem5 with --debug-flags=SyscallBase ? This will give you a strace of the program.

I get the following output:
```shell
gem5 Simulator System.  https://www.gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version 24.0.0.1
gem5 compiled Sep 18 2024 14:12:04
gem5 started Sep 18 2024 14:25:56
gem5 executing on node0.gem5test.kclsystemfuzz-pg0.utah.cloudlab.us, pid 51378
command line: ./build/ARM/gem5.opt --debug-flags=SyscallBase configs/example/arm/starter_se.py /users/AidanDak/test_locale.o

info: 1. command and arguments: ['/users/AidanDak/test_locale.o']
Global frequency set at 1000000000000 ticks per second
warn: No dot file generated. Please install pydot to generate the dot file and pdf.
src/mem/dram_interface.cc:690: warn: DRAM device capacity (8192 Mbytes) does not match the address range assigned (1024 Mbytes)
src/mem/dram_interface.cc:690: warn: DRAM device capacity (8192 Mbytes) does not match the address range assigned (1024 Mbytes)
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
system.remote_gdb: Listening for connections on port 7000
src/sim/simulate.cc:199: info: Entering event queue @ 0.  Starting simulation...
src/arch/arm/insts/pseudo.cc:174: warn: 	instruction 'bti' unimplemented
 580750: system.cpu_cluster.cpus: T0 : syscall Calling brk(0)...
 580750: system.cpu_cluster.cpus: T0 : syscall Returned 73728.
 910500: system.cpu_cluster.cpus: T0 : syscall Calling uname(0x7ffffef838)...
 910500: system.cpu_cluster.cpus: T0 : syscall Returned 0.
1386000: system.cpu_cluster.cpus: T0 : syscall Calling faccessat(-100, 274878060128, 4)...
1386000: system.cpu_cluster.cpus: T0 : syscall Returned -2.
src/sim/mem_state.cc:448: info: Increasing stack size by one page.
1578750: system.cpu_cluster.cpus: T0 : syscall Calling openat(-100, 274878051456, 524288, 0)...
1578750: system.cpu_cluster.cpus: T0 : syscall Returned 3.
1583250: system.cpu_cluster.cpus: T0 : syscall Calling fstatat64(3, 274878056584, 0x7ffffeede0)...
1583250: system.cpu_cluster.cpus: T0 : syscall Returned 0.
1588500: system.cpu_cluster.cpus: T0 : syscall Calling mmap2(0, 33839, 1, 2, 3, 0)...
1588500: system.cpu_cluster.cpus: T0 : syscall Returned 274877870080.
1591000: system.cpu_cluster.cpus: T0 : syscall Calling close(3)...
1591000: system.cpu_cluster.cpus: T0 : syscall Returned 0.
1933750: system.cpu_cluster.cpus: T0 : syscall Calling openat(-100, 274878159712, 524288, 0)...
1933750: system.cpu_cluster.cpus: T0 : syscall Returned 3.
1939000: system.cpu_cluster.cpus: T0 : syscall Calling read(3, 549755744144, 832)...
1939000: system.cpu_cluster.cpus: T0 : syscall Returned 832.
2011750: system.cpu_cluster.cpus: T0 : syscall Calling fstatat64(3, 274878056584, 0x7ffffeee20)...
2011750: system.cpu_cluster.cpus: T0 : syscall Returned 0.
2054500: system.cpu_cluster.cpus: T0 : syscall Calling mmap2(0, 8192, 3, 34, -1, 0)...
2054500: system.cpu_cluster.cpus: T0 : syscall Returned 274877861888.
2215250: system.cpu_cluster.cpus: T0 : syscall Calling mmap2(0, 1805928, 0, 34, -1, 0)...
2215250: system.cpu_cluster.cpus: T0 : syscall Returned 274876055552.
2221500: system.cpu_cluster.cpus: T0 : syscall Calling mmap2(274876071936, 1740392, 5, 2066, 3, 0)...
2221500: system.cpu_cluster.cpus: T0 : syscall Returned 274876071936.
2228000: system.cpu_cluster.cpus: T0 : syscall Calling munmap(274876055552, 16384)...
2228000: system.cpu_cluster.cpus: T0 : syscall Returned 0.
2235750: system.cpu_cluster.cpus: T0 : syscall Calling munmap(274877812736, 48744)...
2235750: system.cpu_cluster.cpus: T0 : syscall Returned 0.
2243250: system.cpu_cluster.cpus: T0 : syscall Calling mprotect)...
src/sim/syscall_emul.cc:74: warn: ignoring syscall mprotect(...)
2243250: system.cpu_cluster.cpus: T0 : syscall Returned 0.
2256250: system.cpu_cluster.cpus: T0 : syscall Calling mmap2(274877739008, 24576, 3, 2066, 3, 1601536)...
2256250: system.cpu_cluster.cpus: T0 : syscall Returned 274877739008.
2320250: system.cpu_cluster.cpus: T0 : syscall Calling mmap2(274877763584, 48744, 3, 50, -1, 0)...
2320250: system.cpu_cluster.cpus: T0 : syscall Returned 274877763584.
2422000: system.cpu_cluster.cpus: T0 : syscall Calling close(3)...
2422000: system.cpu_cluster.cpus: T0 : syscall Returned 0.
3660750: system.cpu_cluster.cpus: T0 : syscall Calling set_tid_address(274877865328)...
3660750: system.cpu_cluster.cpus: T0 : syscall Returned 100.
3665500: system.cpu_cluster.cpus: T0 : syscall Calling set_robust_list)...
src/sim/syscall_emul.cc:74: warn: ignoring syscall set_robust_list(...)
3665500: system.cpu_cluster.cpus: T0 : syscall Returned 0.
3674500: system.cpu_cluster.cpus: T0 : syscall Calling rseq)...
src/sim/syscall_emul.cc:85: warn: ignoring syscall rseq(...)
      (further warnings will be suppressed)
[... 51 lines cut ...]

> In particular I am interested to know which file gem5 is trying to access to get the list of locales.

For reference there seems to only be one locale present on this system:
```shell
> ls /usr/lib/locale
C.utf8
> ls /usr/lib/locale/C.utf8/
LC_ADDRESS  LC_CTYPE           LC_MEASUREMENT  LC_MONETARY  LC_NUMERIC  LC_TELEPHONE
LC_COLLATE  LC_IDENTIFICATION  LC_MESSAGES     LC_NAME      LC_PAPER    LC_TIME
```

[... cut ...]
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=2996, ...}, AT_EMPTY_PATH) = 0
openat(AT_FDCWD, "/usr/lib/locale/C.UTF-8/LC_CTYPE", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/locale/C.utf8/LC_CTYPE", O_RDONLY|O_CLOEXEC) = 3
```

So it appears to find the correct locale files.
I have also attached the strace for the native run here for reference, which also tries `locale-archive` first, then `locale.alias`, `C.UTF-8`, then `C.utf8` [here](https://github.com/user-attachments/files/17050338/strace_gem5_locale_native.txt)

erin-le, 2025-04-17:
@dakaidan Hello, thank you for submitting this issue! According to @BobbyRBruce and @powerjg, this error is most likely due to an unimplemented syscall in SE mode, of which there are many. If you have a change that resolves this, please feel free to open a PR.

closed 2025-04-17 by erin-le