gem5/gem5 #1550: Sample Script for Arm Cortex-A15/Cortex-A7

hoseinaghaei, 2024-09-09:
Hi,

I need to simulate an arm Cortex-A15/Cortex-A7 processor. However, I could not find any built-in scripts. I know I need to use O3 processor types with two levels of caches, but I don't know how to completely configure all the details.
I read the below link, too:
https://www.gem5.org/documentation/learning_gem5/part1/extending_configs

I appreciate any help you could provide.

hoseinaghaei, 2024-09-09:
The slack link does not work. I appreciate if you could generate a new one.

ivanaamit, 2024-09-09:
> The slack link does not work. I appreciate if you could generate a new one.
@hoseinaghaei, could you explain what you mean by this? The Slack link on our website works for me https://join.slack.com/t/gem5-workspace/shared_invite/zt-2e2nfln38-xsIkN1aRmofRlAHOIkZaEA

powerjg, 2024-09-11:
> I need to simulate an arm Cortex-A15/Cortex-A7 processor. However, I could not find any built-in scripts. I know I need to use O3 processor types with two levels of caches, but I don't know how to completely configure all the details.

Unfortunately, we don't provide any components or scripts which out-of-the-box model any specific hardware. At least, not yet.

I would suggest using the standard library and creating a new `Processor` which uses an O3 core with the configuration values from the datasheets. You can find example of how to do this with [caches](https://bootcamp.gem5.org/#02-Using-gem5/05-cache-hierarchies) and [processors](https://bootcamp.gem5.org/#02-Using-gem5/04-cores) in the gem5 bootcamp from this year. I would start with an overview of the [standard library](https://bootcamp.gem5.org/#02-Using-gem5/01-stdlib). There are videos of many of these presentations available on [YouTube](https://www.youtube.com/@gem5).

closed 2024-09-11 by powerjg