gem5/gem5 #1562: Why can gem5 SE mode X86 architecture run multithreading?

TonyXinPlus, 2024-09-13:
Why can gem5 SE mode X86 architecture run multithreading
Isn't there no thread scheduler in SE mode, so can only run single-threaded programs? Why can multithreaded programs run under x86 architecture
Compiler command line gcc test_mem_S-L.c -g -o ./bin_x86/test_mem_S-L_x86 -lpthread
The test_mem_S-L.c file contains pthread_create and pthread_join functions
The code is as follows
//#include "/usr/local/riscv64-unknown-elf/include/pthread.h"
#include <stdio.h>
#include <pthread.h>
//#include </opt/riscv64/riscv64-unknown-elf/include/pthread.h>
#include <stdatomic.h>
#include <stdlib.h>

atomic_int x = 0;
atomic_int y = 0;

void* thread1(void* arg) {
    atomic_store_explicit(&x, 1, memory_order_release);
    int r1 = atomic_load_explicit(&y, memory_order_acquire);
    printf("T1: x = %d  r1 = %d       ", x, r1);
    return NULL;
}

void* thread2(void* arg) {
    atomic_store_explicit(&y, 1, memory_order_release);
    int r2 = atomic_load_explicit(&x, memory_order_acquire);
    printf("  T2: y = %d  r2 = %d       ", y, r2);
    return NULL;
}
int main() {
		int i, n=20;
		pthread_t t1, t2;
		for (i = 0; i < n; i++) {
			
      if (pthread_create(&t1, NULL, thread1, NULL)) {
				perror("pthread_create:p0");
				exit(1);
			}
			if (pthread_create(&t2, NULL, thread2, NULL)) {
				perror("pthread_create:p1");
				exit(1);
			}
      
			if (pthread_join(t1, NULL)) {
				perror("pthread_join:p0");
				exit(1);
			}
			if (pthread_join(t2, NULL)) {
				perror("pthread_join:p1");
				exit(1);
			}
			x = 0;
			y = 0;
			printf("%d\n", i);
		}
    return 0;
}

The gem5 running command is as follows
build/X86/gem5.opt configs/deprecated/example/se.py -c /media/shz37/F8304A82304A47B8/tests_MCM/bin_x86/test_mem_S-L_x86 --num-cpus=3 --cpu-type=O3CPU --caches --l1d_size=32kB --l1d_assoc=2 --l1i_size=32kB --l1i_assoc=2 --l2_size=256kB --l2_assoc=2 --cpu-clock=1GHz
Why can gem5 SE mode X86 architecture run multithreading

Harshil2107, 2024-09-19:
Hi, I am not an expert at this, but this is the answer I got by ask gem5gpt this question, hope this is helpful. 

The question is about how gem5's **SE (System Emulation) mode** can run multithreaded programs on the **X86 architecture**, even though SE mode is designed without a full-fledged thread scheduler, as in a real operating system. Let's break down the behavior and answer the key points.

### Why Multithreaded Programs Run in gem5 SE Mode

#### 1. **How Threads Are Handled in SE Mode**:
   In SE mode, gem5 simulates user-level code, and the kernel services (like thread scheduling) are not fully modeled. However, gem5 **can simulate multithreaded programs** due to how it intercepts system calls and maps them to simulated behavior, including thread creation.

   - When a multithreaded program makes a system call to create a new thread (via `pthread_create`), gem5 handles this by simulating the process of creating new threads.
   - Each thread runs on a separate simulated CPU in gem5 SE mode, which gives the appearance that threads are running concurrently.
   - gem5 intercepts these thread-related system calls (`pthread_create`, `pthread_join`, etc.) and handles them internally, providing basic support for multithreading within the context of the simulation.

   In your case, even though SE mode does not provide a complete operating system environment, it handles **thread creation and synchronization** in a way that allows multithreaded programs to run. This is why you're able to use `pthread_create` and `pthread_join` in your program successfully.

#### 2. **Thread Scheduling in SE Mode**:
   In a real system, thread scheduling is managed by the operating system. However, in gem5 SE mode, there's no actual OS kernel to schedule threads. Instead:
   - **Thread switching happens when gem5 switches between simulated CPUs**.
   - Each simulated CPU is essentially handling its own execution stream (corresponding to different threads).
   - gem5 doesn't perform "real" thread scheduling in SE mode but rather simulates the concurrent execution of threads across different CPUs based on how you configure the simulation (e.g., `--num-cpus=3`).

   Therefore, while there isn't an OS-level thread scheduler in SE mode, the gem5 CPU model allows for **simulated concurrent execution** of threads, which is why multithreaded programs can still run.

#### 3. **Why Multithreading Works on X86 in SE Mode**:
   - X86 is a well-supported architecture in gem5, with a robust implementation of system calls and basic threading support.
   - The simulation is able to handle system calls related to threading, allowing multithreaded programs to run, even in SE mode.

### How gem5 Executes the Program
In your program, you are creating two threads (`thread1` and `thread2`) using `pthread_create`. The threads modify and read the atomic variables `x` and `y`. Here's a breakdown of what happens when this runs in gem5 SE mode:

1. **Thread Creation (`pthread_create`)**:
   - gem5 intercepts the `pthread_create` system call and creates two simulated threads, mapping each to a different simulated CPU.
   - Since you’ve configured gem5 with `--num-cpus=3`, and the program uses two threads, gem5 can assign each thread to a different CPU, effectively simulating parallel execution.

2. **Thread Synchronization (`pthread_join`)**:
   - When the program calls `pthread_join`, gem5 handles this by waiting for the created threads to complete before continuing with the main program execution.
   
3. **Atomic Operations**:
   - The atomic operations (`atomic_store_explicit` and `atomic_load_explicit`) in your program are executed correctly in gem5, as these are standard CPU instructions and do not depend on OS-level support.

### Explanation of the Simulation Command
The command you provided is:

```bash
build/X86/gem5.opt configs/deprecated/example/se.py -c /media/shz37/F8304A82304A47B8/tests_MCM/bin_x86/test_mem_S-L_x86 --num-cpus=3 --cpu-type=O3CPU --caches --l1d_size=32kB --l1d_assoc=2 --l1i_size=32kB --l1i_assoc=2 --l2_size=256kB --l2_assoc=2 --cpu-clock=1GHz
```

Here's what the parameters mean:
- `--num-cpus=3`: This specifies that the simulation should run with 3 simulated CPUs. Since you have two threads in your program, they can each be assigned to a different CPU, allowing for concurrent execution.
- `--cpu-type=O3CPU`: This specifies that an out-of-order CPU model is used, which can better handle instruction-level parallelism.
- `--caches`: This includes a cache model in the simulation.
- `--l1d_size=32kB`, `--l1i_size=32kB`: These specify the sizes of the L1 data and instruction caches.
- `--l2_size=256kB`: This specifies the size of the L2 cache.
- `--cpu-clock=1GHz`: This sets the simulated CPU clock speed to 1 GHz.

### Summary
- **Multithreaded programs** can run in gem5 SE mode because gem5 can simulate the creation and execution of multiple threads, even though there's no OS-level thread scheduler.
- **gem5 intercepts system calls** like `pthread_create` and `pthread_join`, allowing thread creation and synchronization to work within the simulated environment.
- The **X86 architecture** in gem5 supports these operations well, which is why multithreaded programs like yours can run in SE mode.

Thus, while gem5 SE mode doesn't have a full OS kernel with a thread scheduler, it simulates multithreading by running each thread on a separate simulated CPU.

Harshil2107, 2024-10-02:
I am converting this issue to a discussion as it looks like a Q/A than a bug.