gem5/gem5 #1589: RISC-V SE mode system call 258 not implemented

hnpl, 2024-09-22:
**Describe the bug**
Using gem5 SE mode, running a RISC-V workload compiled by a more recent RISC-V compiler results in an error due to system call 258 not implement.

**Affects version**
Current gem5 develop branch.

**gem5 Modifications**
N/A

**To Reproduce**
Running any RISC-V workload in SE mode compiled by a more recent RISC-V compiler. I'm using the latest version of gcc compiled from https://github.com/riscv-collab/riscv-gnu-toolchain.

**Terminal Output**

```shell
~/scr/gem5/> build/RISCV/gem5.opt riscv-hello.py
...
src/sim/simulate.cc:199: info: Entering event queue @ 0.  Starting simulation...
src/sim/syscall_desc.hh:209: fatal: Syscall 258 out of range
Memory Usage: 165896 KBytes
```

**Additional information**
The Linux kernel added the new RISC-V system call number 258 awhile ago [1] namedly `hwprobe`.
This system call is a solution to detect extensions implemented by the processor.
It looks like libc's `__libc_memcpy_ifunc` is calling hwprobe system call.

```
7062264: board.processor.cores.core: T0 : 0x162a6 @__libc_memcpy    : c_beqz a1, 70              : IntAlu : 
7062264: board.processor.cores.core: T0 : 0x162a8 @__libc_memcpy+2    : c_addi sp, -32             : IntAlu :  D=0x7ffffffffffffe90
7062597: board.processor.cores.core: T0 : 0x162aa @__libc_memcpy+4    : c_li a6, 5                 : IntAlu :  D=0x0000000000000005
7062597: board.processor.cores.core: T0 : 0x162ac @__libc_memcpy+6    : c_sdsp ra, 24(sp)          : MemWrite :  D=0x0000000000010c76 A=0x7ffffffffffffea8
7063263: board.processor.cores.core: T0 : 0x162ae @__libc_memcpy+8    : c_mv a5, a1                : IntAlu :  D=0x0000000000016f30
7063263: board.processor.cores.core: T0 : 0x162b0 @__libc_memcpy+10    : c_li a4, 0                 : IntAlu :  D=0x0000000000000000
7063596: board.processor.cores.core: T0 : 0x162b2 @__libc_memcpy+12    : c_li a3, 0                 : IntAlu :  D=0x0000000000000000
7063596: board.processor.cores.core: T0 : 0x162b4 @__libc_memcpy+14    : c_li a2, 0                 : IntAlu :  D=0x0000000000000000
7063929: board.processor.cores.core: T0 : 0x162b6 @__libc_memcpy+16    : c_li a1, 1                 : IntAlu :  D=0x0000000000000001
7063929: board.processor.cores.core: T0 : 0x162b8 @__libc_memcpy+18    : c_mv a0, sp                : IntAlu :  D=0x7ffffffffffffe90
7064262: board.processor.cores.core: T0 : 0x162ba @__libc_memcpy+20    : c_sdsp a6, 0(sp)           : MemWrite :  D=0x0000000000000005 A=0x7ffffffffffffe90
7064595: board.processor.cores.core: T0 : 0x162bc @__libc_memcpy+22    : c_jalr a5                  : IntAlu :  D=0x00000000000162be
7148178: board.processor.cores.core: T0 : 0x16f30 @__riscv_hwprobe    : auipc a5, 99               : IntAlu :  D=0x0000000000079f30
7218108: board.processor.cores.core: T0 : 0x16f34 @__riscv_hwprobe+4    : ld a5, -808(a5)            : MemRead :  D=0x0000000000078d38 A=0x79c08
7260732: board.processor.cores.core: T0 : 0x16f38 @__riscv_hwprobe+8    : c_ld a5, 0(a5)             : MemRead :  D=0x0000000000000000 A=0x78d38
7260732: board.processor.cores.core: T0 : 0x16f3a @__riscv_hwprobe+10    : c_addi16sp sp, -48         : IntAlu :  D=0x7ffffffffffffe60
7261065: board.processor.cores.core: T0 : 0x16f3c @__riscv_hwprobe+12    : c_sdsp s0, 32(sp)          : MemWrite :  D=0x00000000000101e8 A=0x7ffffffffffffe80
7261398: board.processor.cores.core: T0 : 0x16f3e @__riscv_hwprobe+14    : c_sdsp s1, 24(sp)          : MemWrite :  D=0x0000000000000001 A=0x7ffffffffffffe78
7262064: board.processor.cores.core: T0 : 0x16f40 @__riscv_hwprobe+16    : c_sdsp s2, 16(sp)          : MemWrite :  D=0x7fffffffffffff38 A=0x7ffffffffffffe70
7262397: board.processor.cores.core: T0 : 0x16f42 @__riscv_hwprobe+18    : c_sdsp s3, 8(sp)           : MemWrite :  D=0x00000000ffffffff A=0x7ffffffffffffe68
7262730: board.processor.cores.core: T0 : 0x16f44 @__riscv_hwprobe+20    : c_sdsp s4, 0(sp)           : MemWrite :  D=0x0000000000010200 A=0x7ffffffffffffe60
7275717: board.processor.cores.core: T0 : 0x16f46 @__riscv_hwprobe+22    : c_sdsp ra, 40(sp)          : MemWrite :  D=0x00000000000162be A=0x7ffffffffffffe88
7275717: board.processor.cores.core: T0 : 0x16f48 @__riscv_hwprobe+24    : c_mv s4, a0                : IntAlu :  D=0x7ffffffffffffe90
7276050: board.processor.cores.core: T0 : 0x16f4a @__riscv_hwprobe+26    : c_mv s0, a1                : IntAlu :  D=0x0000000000000001
7276050: board.processor.cores.core: T0 : 0x16f4c @__riscv_hwprobe+28    : c_mv s1, a2                : IntAlu :  D=0x0000000000000000
7276383: board.processor.cores.core: T0 : 0x16f4e @__riscv_hwprobe+30    : c_mv s2, a3                : IntAlu :  D=0x0000000000000000
7276383: board.processor.cores.core: T0 : 0x16f50 @__riscv_hwprobe+32    : c_mv s3, a4                : IntAlu :  D=0x0000000000000000
7276716: board.processor.cores.core: T0 : 0x16f52 @__riscv_hwprobe+34    : c_beqz a5, 32              : IntAlu : 
7349976: board.processor.cores.core: T0 : 0x16f72 @__riscv_hwprobe+66    : slli a4, s3, 32            : IntAlu :  D=0x0000000000000000
7349976: board.processor.cores.core: T0 : 0x16f76 @__riscv_hwprobe+70    : addi a7, zero, 258         : IntAlu :  D=0x0000000000000102
7350309: board.processor.cores.core: T0 : 0x16f7a @__riscv_hwprobe+74    : c_mv a0, s4                : IntAlu :  D=0x7ffffffffffffe90
7350309: board.processor.cores.core: T0 : 0x16f7c @__riscv_hwprobe+76    : c_mv a1, s0                : IntAlu :  D=0x0000000000000001
7350642: board.processor.cores.core: T0 : 0x16f7e @__riscv_hwprobe+78    : c_mv a2, s1                : IntAlu :  D=0x0000000000000000
7392933: board.processor.cores.core: T0 : 0x16f80 @__riscv_hwprobe+80    : c_mv a3, s2                : IntAlu :  D=0x0000000000000000
7392933: board.processor.cores.core: T0 : 0x16f82 @__riscv_hwprobe+82    : c_srli a4, 32              : IntAlu :  D=0x0000000000000000
```

[1] https://docs.kernel.org/arch/riscv/hwprobe.html
[2] https://patchew.org/linux/20230314183220.513101-1-evan@rivosinc.com/

hnpl, 2024-09-22:
Fix: #1525

ivanaamit, 2024-10-18:
Closing this issue since the PR has been merged. Feel free to reopen it if I’m mistaken. Thanks!

closed 2024-10-18 by ivanaamit

amar6905, 2024-11-04:
Hello @hnpl, 
I am a gem5 beginner, and I basically want to run a gem5 simulation of a machine learning benchmark named "mlbench" on a riscv cpu model with a memory system (including L1 and L2). I built gem5 for riscv and and the simulation command I am running is:
./build/RISCV/gem5.opt ./configs/learning_gem5/part1/simple-riscv.py

simple-riscv.py is a configuration script in which I added L1 and L2 caches and the simulation mode is set to SE. The error I get is similar to yours:
[image]
 
And I was wondering if you could dive into how you solved your error and what the root cause was?

hnpl, 2024-11-04:
Hi there,
The PR #1525 implements this syscall.
You can use gem5's develop branch, or cherry-pick [this commit](https://github.com/gem5/gem5/pull/1525/commits/b7b6f5643812b8917a7a902725c727681e367792) to fix this problem.

wslqk, 2024-12-04:
Hello @hnpl,
I replaced linux.hh and se_workload with the modified versions from #1525, but the same issue persists. Are there any additional steps I need to take after the replacement?
Thanks.

hnpl, 2024-12-05:
I'm not sure what happened but it appears to be a syntactic error.

CupinDusk, 2024-12-09:
> Hello @hnpl, I replaced linux.hh and se_workload with the modified versions from #1525. When I use scons to rebuild(scons build/RISCV/gem5.opt -j8), the following error occurs: ![image](https://private-user-images.githubusercontent.com/149781196/392246526-0d0a0648-8fc4-4568-993c-134a9ffa6434.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM3NDI4MDksIm5iZiI6MTczMzc0MjUwOSwicGF0aCI6Ii8xNDk3ODExOTYvMzkyMjQ2NTI2LTBkMGEwNjQ4LThmYzQtNDU2OC05OTNjLTEzNGE5ZmZhNjQzNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwOVQxMTA4MjlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zMDI1MTQzNDIwNDllMWQ4NDljYTQzODA3MGI2ZjAwODU1MDAxZTZmMmUxNDdkMzIxNzY2OTQ3NDNkNmM5MGYxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.xYV5g7KXr9J1S5QTtRu2O1jNzHRxhPQHYyT4zkDk-FA) ![image](https://private-user-images.githubusercontent.com/149781196/392246712-f5db6cd8-2591-44af-87e9-883e8bd17c01.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM3NDI4MDksIm5iZiI6MTczMzc0MjUwOSwicGF0aCI6Ii8xNDk3ODExOTYvMzkyMjQ2NzEyLWY1ZGI2Y2Q4LTI1OTEtNDRhZi04N2U5LTg4M2U4YmQxN2MwMS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwOVQxMTA4MjlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wMmZlYzYyMGQzZTFiNjFjMDQyYjcyZDA3Y2ZiYWQ4MzVlNzQ5N2JjMGEwYjJjNTgxZGI3OGNkNWIwM2NkNmJhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Es-8vGWv6ZiwTgHstRleDtkaYnlHVcG057s0s9PoxtE) ![image](https://private-user-images.githubusercontent.com/149781196/392246786-ab6f9816-82f3-4774-88fe-6691455bfff5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM3NDI4MDksIm5iZiI6MTczMzc0MjUwOSwicGF0aCI6Ii8xNDk3ODExOTYvMzkyMjQ2Nzg2LWFiNmY5ODE2LTgyZjMtNDc3NC04OGZlLTY2OTE0NTViZmZmNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwOVQxMTA4MjlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yNTljY2ZhMzQwOWI3YzhmOTk0N2EzYjY2NWMwZTY3MDE4ZmRlYWMxZTQ1NjMwOTkwNjgyMzMyMzcxOWNjNDI0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.OHGydboO1QABPiYIYT8fW801fZkKk5uAV4Rhn8FjZek) I would like to know if there are any steps I might have misunderstood or done incorrectly. Thanks.

Hello，I have the same errors as yours.  And I think it's because of gem5 version. My project is based on gem5 23, so only replacing the linux.hh and se_workload.cc is predictably not enough, and it's the result after my try. I guess you also use the old version gem5? Now I don't want to upgrade my gem5, so I use the old riscv64-gcc, and it works. The truth is that I used old riscv64-gcc to build my whole project, but recently I'm using a new pc, so I try the newest riscv64-gcc, then I  get the syscall 258 error. Maybe you can also try to use an old riscv64-gcc like 2023-10-version. Please let me know if you have some different results. Thx

wslqk, 2024-12-11:
> Hello，I have the same errors as yours. And I think it's because of gem5 version. My project is based on gem5 23, so only replacing the linux.hh and se_workload.cc is predictably not enough, and it's [...]

Thank you very much for your reply. I updated the versions of gem5 and riscv32-gcc (the compiler I am using) to match yours, and the "258 out of range" error has disappeared. However, I am now encountering a "syscall 291 out of range" error. I simply followed the tutorial in this link: https://medium.com/@ayushdixithere/gem5-with-risc-v-a-tinkerers-guide-to-installation-f4cf1a5ec6d1, using riscv32-gcc to compile a pi.c program and running build/RISCV/gem5.opt configs/learning_gem5/part1/simple-riscv.py.

Do you happen to know how to resolve this issue? Once again, thank you for your reply.

CupinDusk, 2024-12-11:
> Thank you very much for your reply. I updated the versions of gem5 and riscv32-gcc (the compiler I am using) to match yours, and the "258 out of range" error has disappeared. However, I am now [...]

hello. My advice is that if your project is in the beginning, you can upgrade the gem5 to the latest version, gem5-24, and use the latest riscv-gcc. I think you can solve most errors. I can't use the latest version because my whole project is based on gem5-23 and it took my several months. So if it's not too difficult, you can upgrade your gem5 and riscv-gcc. By the way, I guess you come from China? And I 'm Chinese lol

wslqk, 2024-12-11:
> hello. My advice is that if your project is in the beginning, you can upgrade the gem5 to the latest version, gem5-24, and use the latest riscv-gcc. I think you can solve most errors. I can't use [...]

Yes, I’m from China. I tried the latest versions of gem5 and riscv32-gcc, but I encountered a fatal 258 error. I also tried the latest develop branch and ran into a fatal 291 error, which is the same issue I faced when using the same versions of gem5 and riscv32-gcc as you. I’m not sure which step is invoking function 291, as there’s no function numbered 291 in the se_workload. If you’d like, we can connect through other means; my email is liugankun4@gmail.com.

CupinDusk, 2024-12-12:
> > > > > Hello @hnpl, I replaced linux.hh and se_workload with the modified versions from #1525. When I use scons to rebuild(scons build/RISCV/gem5.opt -j8), the following error occurs: ![image](https://private-user-images.githubusercontent.com/149781196/392246526-0d0a0648-8fc4-4568-993c-134a9ffa6434.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM3NDI4MDksIm5iZiI6MTczMzc0MjUwOSwicGF0aCI6Ii8xNDk3ODExOTYvMzkyMjQ2NTI2LTBkMGEwNjQ4LThmYzQtNDU2OC05OTNjLTEzNGE5ZmZhNjQzNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwOVQxMTA4MjlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zMDI1MTQzNDIwNDllMWQ4NDljYTQzODA3MGI2ZjAwODU1MDAxZTZmMmUxNDdkMzIxNzY2OTQ3NDNkNmM5MGYxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.xYV5g7KXr9J1S5QTtRu2O1jNzHRxhPQHYyT4zkDk-FA) ![image](https://private-user-images.githubusercontent.com/149781196/392246712-f5db6cd8-2591-44af-87e9-883e8bd17c01.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM3NDI4MDksIm5iZiI6MTczMzc0MjUwOSwicGF0aCI6Ii8xNDk3ODExOTYvMzkyMjQ2NzEyLWY1ZGI2Y2Q4LTI1OTEtNDRhZi04N2U5LTg4M2U4YmQxN2MwMS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwOVQxMTA4MjlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wMmZlYzYyMGQzZTFiNjFjMDQyYjcyZDA3Y2ZiYWQ4MzVlNzQ5N2JjMGEwYjJjNTgxZGI3OGNkNWIwM2NkNmJhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Es-8vGWv6ZiwTgHstRleDtkaYnlHVcG057s0s9PoxtE) ![image](https://private-user-images.githubusercontent.com/149781196/392246786-ab6f9816-82f3-4774-88fe-6691455bfff5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM3NDI4MDksIm5iZiI6MTczMzc0MjUwOSwicGF0aCI6Ii8xNDk3ODExOTYvMzkyMjQ2Nzg2LWFiNmY5ODE2LTgyZjMtNDc3NC04OGZlLTY2OTE0NTViZmZmNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjA5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwOVQxMTA4MjlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yNTljY2ZhMzQwOWI3YzhmOTk0N2EzYjY2NWMwZTY3MDE4ZmRlYWMxZTQ1NjMwOTkwNjgyMzMyMzcxOWNjNDI0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.OHGydboO1QABPiYIYT8fW801fZkKk5uAV4Rhn8FjZek) I would like to know if there are any steps I might have misunderstood or done incorrectly. Thanks.
> > > > 
> > > > 
> > > > Hello，I have the same errors as yours. And I think it's because of gem5 version. My project is based on gem5 23, so only replacing the linux.hh and se_workload.cc is predictably not enough, and it's the result after my try. I guess you also use the old version gem5? Now I don't want to upgrade my gem5, so I use the old riscv64-gcc, and it works. The truth is that I used old riscv64-gcc to build my whole project, but recently I'm using a new pc, so I try the newest riscv64-gcc, then I get the syscall 258 error. Maybe you can also try to use an old riscv64-gcc like 2023-10-version. Please let me know if you have some different results. Thx
> > > 
> > > 
> > > Thank you very much for your reply. I updated the versions of gem5 and riscv32-gcc (the compiler I am using) to match yours, and the "258 out of range" error has disappeared. However, I am now encountering a "syscall 291 out of range" error. I simply followed the tutorial in this link: https://medium.com/@ayushdixithere/gem5-with-risc-v-a-tinkerers-guide-to-installation-f4cf1a5ec6d1, using riscv32-gcc to compile a pi.c program and running build/RISCV/gem5.opt configs/learning_gem5/part1/simple-riscv.py.
> > > Do you happen to know how to resolve this issue? Once again, thank you for your reply.
> > 
> > 
> > hello. My advice is that if your project is in the beginning, you can upgrade the gem5 to the latest version, gem5-24, and use the latest riscv-gcc. I think you can solve most errors. I can't use the latest version because my whole project is based on gem5-23 and it took my several months. So if it's not too difficult, you can upgrade your gem5 and riscv-gcc. By the way, I guess you come from China? And I 'm Chinese lol
> 
> Yes, I’m from China. I tried the latest versions of gem5 and riscv32-gcc, but I encountered a fatal 258 error. I also tried the latest develop branch and ran into a fatal 291 error, which is the same issue I faced when using the same versions of gem5 and riscv32-gcc as you. I’m not sure which step is invoking function 291, as there’s no function numbered 291 in the se_workload. If you’d like, we can connect through other means; my email is [liugankun4@gmail.com](mailto:liugankun4@gmail.com).
> 
> The fatal 291 issue has been resolved. I chose to use gem5 24 and riscv32 gnu compiler toolchain 23, compiling with riscv32-unknown-elf-gcc. If you wish to use the latest version of gem5 while avoiding the fatal 258 issue, you can opt for the gem5 develop branch.
好的，所以291也解决了是吗？我以前遇到过类似的，c内的函数有点新，se模式没有支持，解决办法是换了个系统调用。我们可以邮箱加个wechat