gem5/gem5 #1596: RISCV CPU bug when booting up 24.04 disk image

studyztp, 2024-09-23:
**Describe the bug**
When using 8 RISCV ATOMIC CPUs to boot up a 24.04 ubuntu disk image created by following the [riscv-fs](https://github.com/gem5/gem5-resources/tree/stable/src/riscv-fs/riscv-ubuntu-22.04-24.04) guide, the kernel encounters CPU related bug after [gem5_init.sh](https://github.com/gem5/gem5-resources/blob/stable/src/riscv-fs/riscv-ubuntu-22.04-24.04/files/gem5_init.sh).

**Affects version**
both stable 24.0.0.1 and develop to commit# 2fc44a5

**gem5 Modifications**
None

**To Reproduce**
1. Build a disk image using the `riscv-fs` guide on an 24.04 ubuntu host.
2. Input the path to the disk image to the following script and run it with stable 24.0.0.1 or develop(2fc44a5) gem5
```
from m5.objects.DRAMInterface import DDR4_2400_16x4
from gem5.components.memory.memory import ChanneledMemory
from gem5.components.boards.riscv_board import RiscvBoard
from gem5.components.cachehierarchies.classic.no_cache import NoCache
from gem5.components.processors.simple_processor import SimpleProcessor
from gem5.isas import ISA
from gem5.components.processors.cpu_types import CPUTypes

from gem5.simulate.simulator import Simulator

from gem5.resources.resource import WorkloadResource, obtain_resource,DiskImageResource

# input the path to the disk image here
disk_image_path = ""

processor = SimpleProcessor(
    cpu_type = CPUTypes.ATOMIC,
    isa=ISA.RISCV,
    num_cores=8
)
cache = NoCache()
memory = ChanneledMemory(DDR4_2400_16x4, 1, 128, size="8GiB")
board = RiscvBoard(
    clk_freq="2GHz",
    processor=processor,
    cache_hierarchy=cache,
    memory=memory,
)

workload = WorkloadResource(
    function = "set_kernel_disk_workload",
    parameters = {
        "kernel" : obtain_resource(resource_id="riscv-linux-6.6.33-kernel"),
        "disk_image" : DiskImageResource(local_path=disk_image_path,
            root_partition="1"),
        "bootloader" : obtain_resource(resource_id="riscv-bootloader-opensbi-1.3.1")
    }
)

board.set_workload(workload)

simulator = Simulator(
    board=board
)

simulator.run()

print("all done!")

```

**Terminal Output**

board.terminal:
```
[    0.264981] devtmpfs: mounted
[    0.265440] Freeing unused kernel image (initmem) memory: 2200K
[    0.265460] Run /sbin/init as init process
Kernel booted, In gem5 init...
Starting systemd...
[    0.304812] systemd[1]: System time before build time, advancing clock.
[    0.311767] systemd[1]: systemd 255.4-1ubuntu8 running in system mode (+PAM +AUDIT +SELINUX +APPARMOR +IMA +SMACK +SECCOMP +GCRYPT -GNUTLS +OPENSSL +ACL +BLKID +CURL +ELFUTILS +FIDO2 +IDN2 -IDN +IPTC +KMOD +LIBCRYPTSETUP +LIBFDISK +PCRE2 -PWQUALITY +P11KIT +QRENCODE +TPM2 +BZIP2 +LZ4 +XZ +ZLIB +ZSTD -BPF_FRAMEWORK -XKBCOMMON +UTMP +SYSVINIT default-hierarchy=unified)
[    0.311804] systemd[1]: Detected architecture riscv64.

Welcome to [1mUbuntu 24.04 LTS[0m!

[    0.312429] systemd[1]: Hostname set to <ubuntu>.
[    0.394446] BUG: spinlock wrong owner on CPU#6, swapper/6/0
[    0.394456]  lock: 0xffffffff81487480, .magic: dead4ead, .owner: swapper/2/0, .owner_cpu: -1
[    0.394470] CPU: 6 PID: 0 Comm: swapper/6 Not tainted 6.6.33 #1
[    0.394481] Hardware name: riscv-virtio (DT)
[    0.394488] Call Trace:
[    0.394494] [<ffffffff80005a92>] dump_backtrace+0x1c/0x24
[    0.394506] [<ffffffff808a5362>] show_stack+0x2c/0x38
[    0.394519] [<ffffffff808b150c>] dump_stack_lvl+0x3c/0x54
[    0.394533] [<ffffffff808b1538>] dump_stack+0x14/0x1c
[    0.394546] [<ffffffff8005ea94>] spin_bug+0x80/0xb4
[    0.394560] [<ffffffff8005ecf2>] do_raw_spin_unlock+0xe4/0xfe
[    0.394574] [<ffffffff808ba80e>] _raw_spin_unlock_irqrestore+0x16/0x2e
[    0.394589] [<ffffffff80077618>] rcu_report_qs_rnp+0x86/0x100
[    0.394602] [<ffffffff8007772a>] rcu_report_qs_rdp+0x98/0xcc
[    0.394614] [<ffffffff8007b50c>] rcu_core+0x15a/0x400
[    0.394627] [<ffffffff8007b8ac>] rcu_core_si+0xc/0x14
[    0.394639] [<ffffffff800168c6>] handle_softirqs+0xfc/0x246
[    0.394654] [<ffffffff80016dda>] irq_exit_rcu+0x78/0xa2
[    0.394668] [<ffffffff808b15a4>] handle_riscv_irq+0x64/0x74
[    0.394682] [<ffffffff808b202a>] do_irq+0x56/0x7e
```
simerr:
```
src/mem/dram_interface.cc:690: warn: DRAM device capacity (32768 Mbytes) does not match the address range assigned (8192 Mbytes)
board.platform.terminal: Listening for connections on port 3458
src/arch/riscv/isa.cc:278: info: RVV enabled, VLEN = 256 bits, ELEN = 64 bits
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
src/arch/riscv/isa.cc:278: info: RVV enabled, VLEN = 256 bits, ELEN = 64 bits
src/arch/riscv/isa.cc:278: info: RVV enabled, VLEN = 256 bits, ELEN = 64 bits
src/arch/riscv/isa.cc:278: info: RVV enabled, VLEN = 256 bits, ELEN = 64 bits
src/arch/riscv/isa.cc:278: info: RVV enabled, VLEN = 256 bits, ELEN = 64 bits
src/arch/riscv/isa.cc:278: info: RVV enabled, VLEN = 256 bits, ELEN = 64 bits
src/arch/riscv/isa.cc:278: info: RVV enabled, VLEN = 256 bits, ELEN = 64 bits
src/arch/riscv/isa.cc:278: info: RVV enabled, VLEN = 256 bits, ELEN = 64 bits
board.remote_gdb: Listening for connections on port 7002
src/arch/riscv/linux/fs_workload.cc:162: info: Loaded bootloader '/home/ztpc/.cache/gem5/riscv-bootloader-opensbi-1.3.1' at 0x0
src/arch/riscv/linux/fs_workload.cc:180: info: Loaded kernel '/home/ztpc/.cache/gem5/riscv-linux-6.6.33-kernel' at 0x80200000
src/arch/riscv/linux/fs_workload.cc:199: info: Loaded DTB 'develop-m5outs/device.dtb' at 0x87e00000
src/sim/simulate.cc:199: info: Entering event queue @ 0.  Starting simulation...
src/arch/riscv/isa.cc:655: warn: Ignoring write to miscreg HPMCOUNTER03.
src/arch/riscv/isa.cc:655: warn: Ignoring write to miscreg HPMCOUNTER03.
src/arch/riscv/isa.cc:655: warn: Ignoring write to miscreg HPMCOUNTER03.
src/arch/riscv/isa.cc:655: warn: Ignoring write to miscreg HPMCOUNTER03.
src/arch/riscv/isa.cc:655: warn: Ignoring write to miscreg HPMCOUNTER03.
src/arch/riscv/isa.cc:655: warn: Ignoring write to miscreg HPMCOUNTER03.
src/arch/riscv/isa.cc:655: warn: Ignoring write to miscreg HPMCOUNTER03.
src/arch/riscv/isa.cc:655: warn: Ignoring write to miscreg HPMCOUNTER03.
src/arch/riscv/isa.cc:655: warn: Ignoring write to miscreg HPMCOUNTER03.
src/arch/riscv/isa.cc:655: warn: Ignoring write to miscreg HPMCOUNTER03.
src/arch/riscv/isa.cc:655: warn: Ignoring write to miscreg HPMCOUNTER03.
src/arch/riscv/isa.cc:655: warn: Ignoring write to miscreg HPMCOUNTER03.
src/arch/riscv/isa.cc:655: warn: Ignoring write to miscreg HPMCOUNTER03.
src/arch/riscv/isa.cc:655: warn: Ignoring write to miscreg HPMCOUNTER03.
src/arch/riscv/isa.cc:655: warn: Ignoring write to miscreg HPMCOUNTER03.
src/arch/riscv/isa.cc:655: warn: Ignoring write to miscreg HPMCOUNTER03.
src/dev/riscv/vio_mmio.cc:216: warn: Ignoring queue resize hint. Requested size: 16
src/sim/simulate.cc:199: info: Entering event queue @ 470844665500.  Starting simulation...
```

**Expected behavior**
After encountering the `gem5_init.sh`, the CPU bug will show up in the board terminal output.

**Host Operating System**
Ubuntu 24.04

**Host ISA**
X86

Harshil2107, 2024-09-24:
It looks like you have not added the exit event handlers, this disk will throw 3 m5 exits. It looks like the simulation ended as the first exit was triggered.

For more info, check: https://github.com/orgs/gem5/discussions/1540

studyztp, 2024-09-24:
> It looks like you have not added the exit event handlers, this disk will throw 3 m5 exits. It looks like the simulation ended as the first exit was triggered.
> 
> For more info, check: https://github.com/orgs/gem5/discussions/1540

I have a version that defined an exit event, but after taking the checkpoint, it ran into the cpu bug:
```
from m5.objects.DRAMInterface import DDR4_2400_16x4
from gem5.components.memory.memory import ChanneledMemory
from gem5.components.boards.riscv_board import RiscvBoard
from gem5.components.cachehierarchies.classic.no_cache import NoCache
from gem5.components.processors.simple_processor import SimpleProcessor
from gem5.isas import ISA
from gem5.components.processors.cpu_types import CPUTypes

from gem5.simulate.simulator import Simulator
from gem5.simulate.exit_event import ExitEvent

import m5
import argparse
from pathlib import Path

import sys
import os
sys.path.append(str(Path(__file__).resolve().parent.parent))
print(str(Path(__file__).resolve().parent.parent))

from workload.workload import get_workload

parser = argparse.ArgumentParser()
parser.add_argument("--checkpoint", type=str, default="m5-cpt", help="Checkpoint put here")
args = parser.parse_args()

checkpoint_output_dir = Path(args.checkpoint)
checkpoint_output_dir.mkdir(parents=True, exist_ok=True)

processor = SimpleProcessor(
    cpu_type = CPUTypes.ATOMIC,
    isa=ISA.RISCV,
    num_cores=8
)
cache = NoCache()
memory = ChanneledMemory(DDR4_2400_16x4, 1, 128, size="8GiB")
board = RiscvBoard(
    clk_freq="2GHz",
    processor=processor,
    cache_hierarchy=cache,
    memory=memory,
)
board.set_workload(get_workload())

def exit_handle():
    checkpoint_id = 0
    while True:
        print("get to exit")
        m5.checkpoint(Path(checkpoint_output_dir / f"{checkpoint_id}").as_posix())
        checkpoint_id += 1
        yield False

simulator = Simulator(
    board=board,
    on_exit_event={
        ExitEvent.EXIT:exit_handle()
    }
)

simulator.run()

print("all done!")
```
It took one checkpoint, then the simulation stopped due to the cpu bug.

Harshil2107, 2024-10-15:
Just an update on this bug. I was able to reproduce this issue but when I run the simulation with 1 core the simulation runs to completion. It might my a bug in the way we handle multiple cores in riscv.

Harshil2107, 2025-05-02:
I tried reproducing this bug with the updated disk image with hypercalls and the simulation booted to completion. I am closing this issue as completed.

closed 2025-05-02 by Harshil2107