gem5/gem5 #1602: M5 ops addr mode does not work with checkpointing

hjzccc, 2024-09-24:
**Describe the bug**
When taking a checkpoint in work_begin_addr handler in python and run with the checkpoint, it cannot trigger future ops like exit or work_end in the same binary

**Affects version**
gem5 version 24.0.0.1

**gem5 Modifications**
No

**To Reproduce**
Compile gem5 and m5 as usual

for taking checkpoint 
```python
# should be aligned with ../run_ubuntu.py
from gem5.coherence_protocol import CoherenceProtocol
from gem5.components.boards.x86_board import X86Board
from gem5.components.memory.single_channel import SingleChannelDDR3_1600
from gem5.components.processors.cpu_types import CPUTypes
from gem5.components.processors.simple_processor import (
    SimpleProcessor,
)
from gem5.isas import ISA
from gem5.resources.resource import obtain_resource, Resource, DiskImageResource
from gem5.simulate.exit_event import ExitEvent
from gem5.simulate.simulator import Simulator
from gem5.utils.requires import requires
from gem5.coherence_protocol import CoherenceProtocol
import m5
# This runs a check to ensure the gem5 binary is compiled to X86 and to the
# MESI Two Level coherence protocol.
requires(
    isa_required=ISA.X86,
    kvm_required=True,
)

from gem5.components.memory import SingleChannelDDR4_2400
memory = SingleChannelDDR4_2400(size="3GB")

# Here we setup the processor.
from gem5.components.processors.simple_switchable_processor import SimpleSwitchableProcessor
processor = SimpleProcessor(
    cpu_type = CPUTypes.KVM,
    num_cores = 1,
    isa = ISA.X86
)
from gem5.components.cachehierarchies.classic.no_cache import NoCache
board = X86Board(
    clk_freq="3GHz",
    processor=processor,
    memory=memory,
    cache_hierarchy=NoCache(),
)

command = "cd ~/experiment/testops;" + \
"echo '12345' | sudo -E -S ./01-annotate-this;" 
board.set_kernel_disk_workload(
    kernel=Resource("x86-linux-kernel-5.4.0-105-generic",),
    disk_image=DiskImageResource(local_path="imagebuild/disk-image-ubuntu-24-041/x86-ubuntu-24-04-gapbs"),
    readfile_contents=command,
    kernel_args=[
            "earlyprintk=ttyS0",
            "console=ttyS0",
            "lpj=7999923",
            "root=/dev/sda2",
            "no_systemd=true",
        ],
)
num_run = 0
def workbegin_handler():
    print("Work begin")
    m5.checkpoint("cpt01")
    return True
def workend_handler():
    print("Done with the workload")
    return False
def exit_event_handler():
    print('first exit event: Kernel booted')
    yield False
    print('second exit event: In after boot')
    yield False
    yield False
simulator = Simulator(
    board=board,
    on_exit_event= {
        ExitEvent.WORKBEGIN: workbegin_handler,
        ExitEvent.EXIT: exit_event_handler(),
        ExitEvent.WORKEND: workend_handler
    },
#
)

simulator.run()

```

for running the checkpoint
```python
# should be aligned with ../run_ubuntu.py
from gem5.coherence_protocol import CoherenceProtocol
from gem5.components.boards.x86_board import X86Board
from gem5.components.memory.single_channel import SingleChannelDDR3_1600
from gem5.components.processors.cpu_types import CPUTypes
from gem5.components.processors.simple_processor import (
    SimpleProcessor,
)
from gem5.isas import ISA
from gem5.resources.resource import obtain_resource, Resource, DiskImageResource
from gem5.simulate.exit_event import ExitEvent
from gem5.simulate.simulator import Simulator
from gem5.utils.requires import requires
from gem5.coherence_protocol import CoherenceProtocol
import m5
# This runs a check to ensure the gem5 binary is compiled to X86 and to the
# MESI Two Level coherence protocol.
requires(
    isa_required=ISA.X86,
    kvm_required=True,
)

from gem5.components.memory import SingleChannelDDR4_2400
memory = SingleChannelDDR4_2400(size="3GB")

processor = SimpleProcessor(
    cpu_type = CPUTypes.TIMING,
    num_cores = 1,
    isa = ISA.X86
)

from gem5.components.cachehierarchies.classic.no_cache import NoCache
board = X86Board(
    clk_freq="3GHz",
    processor=processor,
    memory=memory,
    cache_hierarchy=NoCache(),
)
command = "cd ~/experiment;" + \
"echo '12345' | sudo -E -S ./01-annotate-this;"
board.set_kernel_disk_workload(
    kernel=Resource("x86-linux-kernel-5.4.0-105-generic",),
    disk_image=DiskImageResource(local_path="imagebuild/disk-image-ubuntu-24-041/x86-ubuntu-24-04-gapbs"),# actually still 24.04
    readfile_contents=command,
    kernel_args=[
            "earlyprintk=ttyS0",
            "console=ttyS0",
            "lpj=7999923",
            "root=/dev/sda2",
            "no_systemd=true",
        ],
)
num_run = 0
def workbegin_handler():
    print("Work begin")
    return False
def workend_handler():
    print("Done with the workload")
    print("Dump the current stats")
    m5.stats.dump()
    return True
def exit_event_handler():
    print('first exit event: Kernel booted')
    yield False
    print('second exit event: In after boot')
    yield False
    yield True
simulator = Simulator(
    board=board,
    on_exit_event= {
        ExitEvent.WORKBEGIN :workbegin_handler,
        ExitEvent.EXIT: exit_event_handler(),
        ExitEvent.WORKEND: workend_handler
    },
    checkpoint_path="cpt01"
)

simulator.run()
```
01-annotate-this code
```cpp
#include <iostream>
#include <unistd.h>
#include <dirent.h>

// Include the gem5 m5ops header file
#include <gem5/m5ops.h>
//

// Include the gem5 m5_mmap header file
#include <m5_mmap.h>
//

int computation(int a, int b){
    for(int i=0;i<b;i++){
        a+=b;
    }
    return a;
}
int main() {

    m5op_addr = 0XFFFF0000;

    map_m5_mem();

    m5_work_begin_addr(0, 0);

    int result=computation(10,10000);

    m5_work_end_addr(0, 0);

    struct dirent *d;
    DIR *dr;
    dr = opendir(".");
    if (dr!=NULL) {
        std::cout<<"List of Files & Folders:\n";
        for (d=readdir(dr); d!=NULL; d=readdir(dr)) {
            std::cout<<d->d_name<< ", ";
            std::cout<<result<< ", ";
        }
        closedir(dr);
    }
    else {
        std::cout<<"\nError Occurred!";
    }
    std::cout<<std::endl;

    unmap_m5_mem();

    return 0;
}

```
If possible, please include the Python configuration script used and state clearly any parameters passed.

**Terminal Output**
when taking checkpoint it successfully triggers the work_begin handler

```shell
Work begin
Writing checkpoint
```
when run with checkpoint, it ignores work end and output the contents after it
```shell
cript from gem5-bridge stored in /tmp/script
sudo: unable to resolve host (none): Name or service not known
[sudo] password for gem5: List of Files & Folders:
01-annotate-this.cpp, 100000010, Makefile, 100000010, ., 100000010, .., 100000010, 01-annotate-this, 100000010, 
```
**Expected behavior**
Trigger work_end when run with checkpoint

**Host Operating System**
Ubuntu 22.04

**Host ISA**
X86

**Compiler used**
gcc version 11.4.0 (Ubuntu 11.4.0-1ubuntu1~22.04) 

**Additional information**
Add any other information which does not fit in the previous sections but may be of use in fixing this bug.

hly2019, 2024-09-24:
Meet similar issue. Appreciate for any help, thanks!

Harshil2107, 2024-09-24:
One thing that might cause an issue is that the `workbegin_handler` is not passed as a function. Can you try by updating `ExitEvent.WORKBEGIN :workbegin_handler` to `ExitEvent.WORKBEGIN :workbegin_handler()` and similarly for the work_end. This could lead to the handler not being called correctly. 

link to an issue for more context: #1492

hjzccc, 2024-09-24:
> One thing that might cause an issue is that the `workbegin_handler` is not passed as a function. Can you try by updating `ExitEvent.WORKBEGIN :workbegin_handler` to `ExitEvent.WORKBEGIN [...]

Still have the same issue when using generator function, one observation is if the checkpoint is taken before the binary(ie. in the second m5 exit event) it can run successfully, my guess is maybe the m5_map_mem is not working well with checkpointing.

Harshil2107, 2024-09-24:
That would also be my first guess that you need to call `map_m5_mem()` again, to maybe initialize the addresses when running from the checkpoint?

hjzccc, 2024-09-25:
> That would also be my first guess that you need to call `map_m5_mem()` again, to maybe initialize the addresses when running from the checkpoint?

So I've tried to use something like this:
```cpp
    map_m5_mem();
    m5_exit_addr(0);
    map_m5_mem();
    m5_work_begin_addr(0, 0);

    int result=computation(10,10000);

    m5_work_end_addr(0, 0);
    
```

checkpoint is set in the m5_exit event here, and this is the output I get:
```shell
sudo: unable to resolve host (none): Name or service not known
[sudo] password for gem5: m5 mem already mapped.
[   33.786081] ata1: lost interrupt (Status 0x50)
[   33.786081] ata1.00: exception Emask 0x0 SAct 0x0 SErr 0x0 action 0x6 frozen
[   33.790081] ata1.00: failed command: READ DMA
[   33.790081] ata1.00: cmd c8/00:d8:28:4b:14/00:00:00:00:00/e0 tag 0 dma 110592 in
[   33.790081]          res 40/00:00:00:00:00/00:00:00:00:00/00 Emask 0x4 (timeout)
[   33.794081] ata1.00: status: { DRDY }
[   33.798081] ata1: soft resetting link
[   33.958091] ata1.00: configured for UDMA/33
[   33.962092] ata1.00: device reported invalid CHS sector 0
[   33.962092] sd 0:0:0:0: [sda] tag#0 FAILED Result: hostbyte=DID_OK driverbyte=DRIVER_SENSE
[   33.966092] sd 0:0:0:0: [sda] tag#0 Sense Key : Illegal Request [current] 
[   33.970092] sd 0:0:0:0: [sda] tag#0 Add. Sense: Unaligned write command
[   33.974092] sd 0:0:0:0: [sda] tag#0 CDB: Read(10) 28 00 00 14 4b 28 00 00 d8 00
[   33.978093] blk_update_request: I/O error, dev sda, sector 1329960 op 0x0:(READ) flags 0x80700 phys_seg 3 prio class 0
[   33.982093] ata1: EH complete
Done running script from gem5-bridge, exiting.
[?2004hgem5@(none):~$ 
```

firstly it says m5 mem already mapped means it successfully executes from the checkpoint but then there are some errors.

Harshil2107, 2024-09-25:
I will try to reproduce the error with the npb 24.04 disk. To narrow it down if it is a gem5 issue or a disk issue.

Harshil2107, 2024-09-26:
I tried to reproduce the issue but `m5 work_end`  was triggered when I did. This are the scripts that I used for the saving and taking checkpoints.

Saving
```python
import argparse
import time

import m5
from m5.objects import Root
from m5.stats.gem5stats import get_simstat
from m5.util import warn

from gem5.coherence_protocol import CoherenceProtocol
from gem5.components.boards.x86_board import X86Board
from gem5.components.memory import DualChannelDDR4_2400
from gem5.components.processors.cpu_types import CPUTypes
from gem5.components.processors.simple_processor import (
    SimpleProcessor,
)
from gem5.isas import ISA
from gem5.resources.resource import obtain_resource
from gem5.simulate.simulator import (
    ExitEvent,
    Simulator,
)
from gem5.utils.requires import requires

requires(
    isa_required=ISA.X86,
    coherence_protocol_required=CoherenceProtocol.MESI_TWO_LEVEL,
    kvm_required=True,
)

# Following are the list of benchmark programs for npb.

# We are restricting classes of NPB to A, B and C as the other classes (D and
# F) require main memory size of more than 3 GB. The X86Board is currently
# limited to 3 GB of memory. This limitation is explained later in line 136.

# The resource disk has binaries for class D. However, only `ep` benchmark
# works with class D in the current configuration. More information on the
# memory footprint for NPB is available at https://arxiv.org/abs/2010.13216

parser = argparse.ArgumentParser(
    description="An example configuration script to run the npb benchmarks."
)

npb_suite = obtain_resource(
    "x86-ubuntu-24.04-npb-suite", resource_version="1.0.0"
)
# The only positional argument accepted is the benchmark name in this script.

parser.add_argument(
    "--benchmark",
    type=str,
    required=True,
    help="Input the benchmark program to execute.",
    choices=[workload.get_id() for workload in npb_suite],
)

parser.add_argument(
    "--ticks",
    type=int,
    help="Optionally put the maximum number of ticks to execute during the "
    "ROI. It accepts an integer value.",
)

args = parser.parse_args()

# The simulation may fail in the case of using size "c" and size "d" of the
[... 253 lines cut ...]
        ExitEvent.WORKEND: handle_workend(),
        ExitEvent.EXIT: exit_event_handler(),
    },
    checkpoint_path="path/to/checkpoint",
)

# We start the simulation.
simulator.run()
```

I ran the save checkpoint with 

`./build/ALL/gem5.opt -d m5out/npd-save-checkpoint -re npb-checkpoint-tests/npb-save-cpt.py --benchmark x86-ubuntu-24.04-npb-ua-s`

and restore with `./build/ALL/gem5.opt -d m5out/npd-restore-checkpoint -re npb-checkpoint-tests/npb-restore-cpt.py --benchmark x86-ubuntu-24.04-npb-ua-s`

In the restore I hard coded the code to run the benchmark as the `set_workload` function doesnt have a checkpoint argument.

I get the following in the simout for the restore
```
      0: board.pc.south_bridge.cmos.rtc: Real-time clock set to Sun Jan  1 00:00:00 2012
Dump stats at the end of the ROI!
```

which indicates that the workend handler was triggered.

Harshil2107, 2024-10-02:
I am closing the issue as it looks like it have been completed. Please feel free to open the issue if the issue still persists or there are new details pertaining to the issue.

closed 2024-10-02 by Harshil2107