gem5/gem5 #1611: ARM binary out of range syscall

hoseinaghaei, 2024-09-26:
**Describe the bug**
I need to run gem5 with the ARM O3 processor.
When I use the binary located at `/tests/test-progs/hello/bin/arm/linux/hello` everything works fine but when compile another code 
`#include <stdio.h>
int main() {
    printf("Hello, ARM!\n");
    return 0;
}
`
I get this error:
`build/ARM/base/statistics.hh:277: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
gem5 Simulator System.  http://gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version 21.1.0.0
gem5 compiled Sep 24 2024 17:30:34
gem5 started Sep 26 2024 16:53:01
gem5 executing on user01, pid 28484
command line: build/ARM/gem5.opt -d m5out-cortex-2 configs/common/cores/arm/cortex-a15.py

Global frequency set at 1000000000000 ticks per second
warn: No dot file generated. Please install pydot to generate the dot file and pdf.
0: system.remote_gdb: listening for remote gdb on port 7000
Starting simulation with O3CPU (Out-of-Order) Cortex-A15 and L1/L2 cache configuration...
build/ARM/sim/simulate.cc:107: info: Entering event queue @ 0.  Starting simulation...
build/ARM/sim/syscall_emul.cc:73: warn: ignoring syscall set_robust_list(...)
build/ARM/sim/syscall_desc.hh:204: fatal: Syscall 398 out of range
Memory Usage: 8515608 KBytes
`
I compile the code using `arm-linux-gnueabi-gcc -marm -static -g -o hello-arm hello.c` and it's information mach with the binary in the test folder.
`hello-arm: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), statically linked, BuildID[sha1]=11741fdbc30cd57860546f706974c28508da1467, for GNU/Linux 3.2.0, with debug_info, not stripped
`

**Affects version**
21.1.0.0

**gem5 Modifications**
No

**To Reproduce**
Run gem5 with this command:
`build/ARM/gem5.opt -d m5out-cortex-2 configs/common/cores/arm/cortex-a15.py`

If writing code, or a terminal command, use code blocks. Either an inline code block, `scons build/ALL/gem5.opt` (enclosed in two '`') or a multi-line codeblock:

```python
import m5
from m5.objects import *
from m5.util import *
from O3_ARM_v7a import *
# Define a system to simulate
system = System()

# Set clock frequency for the system
system.clk_domain = SrcClockDomain()
system.clk_domain.clock = '2.5GHz'  # Adjust according to your needs
system.clk_domain.voltage_domain = VoltageDomain()
system.clk_domain.voltage_domain.voltage = "1.1V"

# Define the memory configuration
system.mem_mode = 'timing'  # Use timing memory mode for accurate simulation
system.mem_ranges = [AddrRange('8GB')]  # Set memory size

# Create a simple memory bus
system.membus = SystemXBar()

# Define the CPU - Out-of-Order CPU (O3CPU model) simulating ARM Cortex-A15
system.cpu = O3_ARM_v7a_3()  # O3CPU simulates an out-of-order ARMv7 Cortex-A15-like CPU
system.cpu.numPhysFloatRegs = 32

# L1 Instruction Cache (I-Cache)
system.cpu.icache = O3_ARM_v7a_ICache(size='32kB', assoc=2)  # Set L1 I-cache size and associativity
system.cpu.icache.cpu_side = system.cpu.icache_port
# system.cpu.icache.mem_side = system.membus.cpu_side_ports

# L1 Data Cache (D-Cache)
system.cpu.dcache = O3_ARM_v7a_DCache(size='32kB', assoc=2)  # Set L1 D-cache size and associativity
system.cpu.dcache.cpu_side = system.cpu.dcache_port
# system.cpu.dcache.mem_side = system.membus.cpu_side_ports

system.l2bus = L2XBar()

system.cpu.dcache.mem_side = system.l2bus.cpu_side_ports
system.cpu.icache.mem_side = system.l2bus.cpu_side_ports

# L2 Cache (shared by both L1 I-cache and D-cache)
system.l2cache = O3_ARM_v7aL2(size='1MB', assoc=16)  # Set L2 cache size and associativity
system.l2cache.cpu_side = system.l2bus.mem_side_ports
system.l2cache.mem_side = system.membus.cpu_side_ports  # Connect L2 cache to memory bus
# system.l2cache.cpu_side = system.cpu.icache.mem_side
# system.l2cache.mem_side = system.membus.cpu_side_ports

# Connect L2 cache to the memory bus
# system.l2cache.mem_side = system.membus.cpu_side_ports

# Setup memory controller
system.cpu.createInterruptController()

system.mem_ctrl = MemCtrl()
system.mem_ctrl.dram = DDR3_1600_8x8()
system.mem_ctrl.dram.range = system.mem_ranges[0]
system.mem_ctrl.port = system.membus.mem_side_ports

system.system_port = system.membus.cpu_side_ports

# Create a process for running a binary
binary = '/home/user01/workspace/gem5/tests/test-progs/hello/bin/arm/linux/hello'
binary = '/home/user01/workspace/arm/hello-arm'
#binary = '/home/user01/workspace/paper/hello'
system.workload = SEWorkload.init_compatible(binary)

process = Process()
process.cmd = [binary]
system.cpu.workload = process
system.cpu.createThreads()

# Define the root object for the simulation
root = Root(full_system=False, system=system)

# Instantiate the simulation
m5.instantiate()

print("Starting simulation with O3CPU (Out-of-Order) Cortex-A15 and L1/L2 cache configuration...")
exit_event = m5.simulate()

print(f"Simulation finished with exit cause: {exit_event.getCause()}")

```

If possible, please include the Python configuration script used and state clearly any parameters passed.

**Terminal Output**
If applicable, add the terminal output here. If long, only include the relevant lines.
Please put the terminal output in code blocks. I.e.:

```shell
#Terminal output here#
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Host Operating System**
Ubuntu 22.04

**Host ISA**
X86

**Compiler used**
State which compiler was used to compile gem5. Please include the compiler version.

**Additional information**
Add any other information which does not fit in the previous sections but may be of use in fixing this bug.

erin-le, 2024-10-01:
Hello, I was able to get this to work by using `aarch64-linux-gnu-gcc` as the compiler and by using gem5 version 24.0.0.1, which is the latest version. Could you try it and see if that works for you?

hoseinaghaei, 2024-10-04:
Hi. Thanks for your response.
I get another error.
```shell
user01@user01:~/workspace/gem5$ build/ARM/gem5.opt -d m5out-cortex-2 configs/common/cores/arm/cortex-a15.py
build/ARM/base/statistics.hh:277: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
gem5 Simulator System.  http://gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version 21.1.0.0
gem5 compiled Sep 24 2024 17:30:34
gem5 started Oct  4 2024 15:36:00
gem5 executing on user01, pid 1019844
command line: build/ARM/gem5.opt -d m5out-cortex-2 configs/common/cores/arm/cortex-a15.py

Global frequency set at 1000000000000 ticks per second
warn: No dot file generated. Please install pydot to generate the dot file and pdf.
0: system.remote_gdb: listening for remote gdb on port 7000
Starting simulation with O3CPU (Out-of-Order) Cortex-A15 and L1/L2 cache configuration...
build/ARM/sim/simulate.cc:107: info: Entering event queue @ 0.  Starting simulation...
build/ARM/sim/syscall_emul.cc:73: warn: ignoring syscall set_robust_list(...)
build/ARM/sim/syscall_desc.hh:204: fatal: Syscall 293 out of range
Memory Usage: 8515668 KBytes
```
Is this relevant to the 64 bits?
I can not change the version because I have to convert gem5 outputs to McPAT inputs, and no project is available for the last version of gem5.

hoseinaghaei, 2024-10-04:
`aarch64-none-linux-gnu-gcc -static hello.c -o hello-arm` worked.

closed 2024-10-04 by hoseinaghaei