gem5/gem5 #1622: Implement GPU stdlib component

abmerop, 2024-10-02:
Add a GPU stdlib component as a new feature. The idea is to have a self contained component that other stdlib scripts can use to simply "get" a GPU. Ideally something like:

```
from gem5.components.devices.amdgpu import MI300X

...

requires(
    isa_required=ISA.X86,
    coherence_protocol_required=CoherenceProtocol.GPU_VIPER,
)

...

gpu1 = MI300X(...)

...

board = SomeX86Board(
    processor=processor,
    cache_hierarchy=cache_hierarchy,
    memory=memory,
    pci_devices=[gpu1],
)
```

A proof-of-concept "bash-style" script can be seen here: https://github.com/abmerop/gem5/commit/ad6ab42d8299f9bdf9c78634acbcbabb926abc11  . This confirms multiple RubySystems is working by setting different cache line sizes for CPU and GPU RubySystem side and running applications to completion with verification.

The proof-of-concept is not yet completely self-contained because it must return a list of CPU-side Ruby DMA objects that the CPU RubySystem must connect to its network.

This should require following steps:
- [x] Implement multiple RubySystems #1349 .
- [x] Confirm multiple RubySystems is working by testing independent CPU and GPU RubySystems with obvious different configurations.
- [x] Create isolated GPU config in classic config style.
- [ ] Move isolated GPU config to stdlib.
- [ ] Implement "pci_devices" param in base stdlib board class.
- [ ] Create stdlib config creating a standard CPU side and use GPU object.
- [ ] Test completed stdlib config.

powerjg, 2024-10-02:
A few very quick thoughts. I hope to dig into this, and your other changes, sometime today.

1. Thanks for taking this on!
2. I'm super excited!
3. To expedite getting a minimum thing ready to merge, I suggest creating a new board (e.g., `X86GPUBoard`) instead of having a list of PCI devices. The list of PCI devices is a cool idea, but it may cause refactoring that could delay getting something working merged.

abmerop, 2024-10-03:
> 3. To expedite getting a minimum thing ready to merge, I suggest creating a new board (e.g., `X86GPUBoard`) instead of having a list of PCI devices. The list of PCI devices is a cool idea, but it may cause refactoring that could delay getting something working merged.

Sounds reasonable.  I think the first thing I will do is to get a MOESI_AMD_Base protocol working.  From there, adding GPU device should be straightforward.

abmerop, 2024-10-06:
Implemented: #1636

closed 2025-01-15 by erin-le