gem5/gem5 #1630: unable to compile gem5

MGThePro, 2024-10-04:
**Describe the bug**
Errors out during compilation

**Affects version**
Latest commit on both stable branch (b1a44b89c7bae73fae2dc547bc1f871452075b85) and develop (6a24b69a972684544385ef7abd1e96c3644c1bb4)

**gem5 Modifications**
none

**To Reproduce**

1. Compile gem5 on an up to date linux distro (in my case Arch Linux) with command `scons build/X86/gem5.opt`

**Terminal Output**

```shell
 [     CXX] src/arch/x86/cpuid.cc -> X86/arch/x86/cpuid.o
Generating LALR tables
In file included from src/arch/x86/regs/int.hh:44,
                 from src/arch/x86/emulenv.hh:41,
                 from src/arch/x86/emulenv.cc:38:
src/cpu/reg_class.hh:168:7: error: ‘class gem5::RegClassOps’ has virtual functions and accessible non-virtual destructor [-Werror=non-virtual-dtor]
  168 | class RegClassOps
      |       ^~~~~~~~~~~
src/arch/x86/regs/int.hh:105:7: error: base class ‘class gem5::RegClassOps’ has accessible non-virtual destructor [-Werror=non-virtual-dtor]
  105 | class FlatIntRegClassOps : public RegClassOps
      |
```
With a lot more affected classes, but always the same error message (Probably too long for github, but I could also send the whole thing if necessary).

**Host Operating System**
Arch Linux

**Host ISA**
X86-64

**Compiler used**
gcc 14.2.1. Also tried with gcc 13.3.0 as well as clang 18.1.8 with the same issue

erin-le, 2024-10-04:
Hello, could you send the entire error message?

MGThePro, 2024-10-04:
```
In file included from src/base/stats/group.hh:46,
                 from src/sim/sim_object.hh:53,
                 from src/arch/generic/decoder.hh:37,
                 from src/arch/generic/decoder.cc:28:
src/base/stats/units.hh:120:7: error: ‘class gem5::statistics::units::Base’ has virtual functions and accessible non-virtual destructor [-Werror=non-virtual-dtor]
  120 | class Base
      |       ^~~~
src/base/stats/units.hh:126:7: error: base class ‘class gem5::statistics::units::Base’ has accessible non-virtual destructor [-Werror=non-virtual-dtor]
  126 | class Cycle : public Base
      |       ^~~~~
src/base/stats/units.hh:126:7: error: ‘class gem5::statistics::units::Cycle’ has virtual functions and accessible non-virtual destructor [-Werror=non-virtual-dtor]
src/base/stats/units.hh:143:7: error: base class ‘class gem5::statistics::units::Base’ has accessible non-virtual destructor [-Werror=non-virtual-dtor]
  143 | class Tick : public Base
      |       ^~~~
src/base/stats/units.hh:143:7: error: ‘class gem5::statistics::units::Tick’ has virtual functions and accessible non-virtual destructor [-Werror=non-virtual-dtor]
src/base/stats/units.hh:160:7: error: base class ‘class gem5::statistics::units::Base’ has accessible non-virtual destructor [-Werror=non-virtual-dtor]
  160 | class Second : public Base
      |       ^~~~~~
src/base/stats/units.hh:160:7: error: ‘class gem5::statistics::units::Second’ has virtual functions and accessible non-virtual destructor [-Werror=non-virtual-dtor]
src/base/stats/units.hh:177:7: error: base class ‘class gem5::statistics::units::Base’ has accessible non-virtual destructor [-Werror=non-virtual-dtor]
  177 | class Bit : public Base
      |       ^~~
src/base/stats/units.hh:177:7: error: ‘class gem5::statistics::units::Bit’ has virtual functions and accessible non-virtual destructor [-Werror=non-virtual-dtor]
src/base/stats/units.hh:194:7: error: base class ‘class gem5::statistics::units::Base’ has accessible non-virtual destructor [-Werror=non-virtual-dtor]
  194 | class Byte : public Base
      |       ^~~~
src/base/stats/units.hh:194:7: error: ‘class gem5::statistics::units::Byte’ has virtual functions and accessible non-virtual destructor [-Werror=non-virtual-dtor]
src/base/stats/units.hh:211:7: error: base class ‘class gem5::statistics::units::Base’ has accessible non-virtual destructor [-Werror=non-virtual-dtor]
  211 | class Watt : public Base
      |       ^~~~
src/base/stats/units.hh:211:7: error: ‘class gem5::statistics::units::Watt’ has virtual functions and accessible non-virtual destructor [-Werror=non-virtual-dtor]
src/base/stats/units.hh:229:7: error: base class ‘class gem5::statistics::units::Base’ has accessible non-virtual destructor [-Werror=non-virtual-dtor]
  229 | class Joule : public Base
      |       ^~~~~
src/base/stats/units.hh:229:7: error: ‘class gem5::statistics::units::Joule’ has virtual functions and accessible non-virtual destructor [-Werror=non-virtual-dtor]
src/base/stats/units.hh:246:7: error: base class ‘class gem5::statistics::units::Base’ has accessible non-virtual destructor [-Werror=non-virtual-dtor]
  246 | class Volt : public Base
      |       ^~~~
src/base/stats/units.hh:246:7: error: ‘class gem5::statistics::units::Volt’ has virtual functions and accessible non-virtual destructor [-Werror=non-virtual-dtor]
src/base/stats/units.hh:263:7: error: base class ‘class gem5::statistics::units::Base’ has accessible non-virtual destructor [-Werror=non-virtual-dtor]
  263 | class DegreeCelsius : public Base
      |       ^~~~~~~~~~~~~
src/base/stats/units.hh:263:7: error: ‘class gem5::statistics::units::DegreeCelsius’ has virtual functions and accessible non-virtual destructor [-Werror=non-virtual-dtor]
src/base/stats/units.hh:285:7: error: base class ‘class gem5::statistics::units::Base’ has accessible non-virtual destructor [-Werror=non-virtual-dtor]
  285 | class Count : public Base
      |       ^~~~~
src/base/stats/units.hh:285:7: error: ‘class gem5::statistics::units::Count’ has virtual functions and accessible non-virtual destructor [-Werror=non-virtual-dtor]
src/base/stats/units.hh:302:7: error: base class ‘class gem5::statistics::units::Base’ has accessible non-virtual destructor [-Werror=non-virtual-dtor]
  302 | class Ratio : public Base
      |       ^~~~~
src/base/stats/units.hh:302:7: error: ‘class gem5::statistics::units::Ratio’ has virtual functions and accessible non-virtual destructor [-Werror=non-virtual-dtor]
src/base/stats/units.hh:319:7: error: base class ‘class gem5::statistics::units::Base’ has accessible non-virtual destructor [-Werror=non-virtual-dtor]
  319 | class Unspecified : public Base
      |       ^~~~~~~~~~~
src/base/stats/units.hh:319:7: error: ‘class gem5::statistics::units::Unspecified’ has virtual functions and accessible non-virtual destructor [-Werror=non-virtual-dtor]
cc1plus: all warnings being treated as errors
scons: *** [build/X86/arch/generic/decoder.o] Error 1
In file included from src/base/stats/group.hh:46,
                 from src/sim/sim_object.hh:53,
                 from src/arch/generic/mmu.hh:46,
                 from src/arch/generic/mmu.cc:44:
src/base/stats/units.hh:120:7: error: ‘class gem5::statistics::units::Base’ has virtual functions and accessible non-virtual destructor [-Werror=non-virtual-dtor]
  120 | class Base
      |       ^~~~
[... cut ...]
src/base/stats/units.hh:302:7: error: ‘class gem5::statistics::units::Ratio’ has virtual functions and accessible non-virtual destructor [-Werror=non-virtual-dtor]
src/base/stats/units.hh:319:7: error: base class ‘class gem5::statistics::units::Base’ has accessible non-virtual destructor [-Werror=non-virtual-dtor]
  319 | class Unspecified : public Base
      |       ^~~~~~~~~~~
src/base/stats/units.hh:319:7: error: ‘class gem5::statistics::units::Unspecified’ has virtual functions and accessible non-virtual destructor [-Werror=non-virtual-dtor]
cc1plus: all warnings being treated as errors
scons: *** [build/X86/python/_m5/param_BaseTLB.o] Error 1
```

erin-le, 2024-10-07:
I'm having trouble reproducing this error. I used GCC 13.3.0 on Ubuntu 24.04 on an x86 system and was able to compile successfully. I compiled X86/gem5.fast and X86/gem5.opt on the stable branch. The build command was `scons build/X86/gem5.opt -j 6`. Can you give any more information on how to reproduce this?

If you urgently need to use gem5, you can try using one of the gem5 Docker images. Instructions can be found here: https://www.gem5.org/documentation/general_docs/building

abmerop, 2024-10-07:
I am able to reproduce this (any many many more errors) on gcc 14.2 on Ubuntu 22.04 (no apt package, I built gcc from source).  It looks like gcc 14 is much more strict about things moving into C++20 support.

erin-le, 2024-10-09:
As a note, I was able to get gem5 to build on gcc-14.0.1, which is the version provided by Ubuntu packages. Unfortunately, I am having some difficulty with building gcc 14.2 from source and getting a gem5 build to start using it. Could someone provide a Docker image that can be used to reproduce this error?