gem5/gem5 #1645: riscv simulation raises panic condition

psharath1, 2024-10-08:
**Describe the bug**
running vlseg4e32 instruction causes panic condition

**Affects version**
commit b1a44b89c

**gem5 Modifications**

building the vec-fft benchmark from https://github.com/ucb-bar/saturn-vectors/tree/master/benchmarks/vec-fft

1. Compile gem5 with command ...  scons build/RISCV/gem5.opt
2. Execute the simulation with...
riscv64-unknown-elf-gcc -DPREALLOCATE=1 -march=rv64imafdcv -mabi=lp64d -O3 -ftree-vectorize  -ffast-math -fno-common fft2_main.c vec-fft2.c  -o FFT

build/RISCV/gem5.opt ../Test/riscv.py

```python
import m5
from m5.objects import *

class L1Cache(Cache):
    assoc = 2
    tag_latency = 1
    data_latency = 1
    response_latency = 1
    mshrs = 4
    tgts_per_mshr = 20

    def connectCPU(self, cpu):
        raise NotImplementedError

    def connectBus(self, bus):
        self.mem_side = bus.cpu_side_ports

class L1ICache(L1Cache):
    size = '512kB'

    def connectCPU(self, cpu):
        self.cpu_side = cpu.icache_port

class L1DCache(L1Cache):
    size = '1024kB'

    def connectCPU(self, cpu):
        self.cpu_side = cpu.dcache_port

class L2Cache(Cache):
    size = '1024kB'
    assoc = 8
    tag_latency = 1
    data_latency = 1
    response_latency = 1
    mshrs = 20
    tgts_per_mshr = 12

    def connectCPUSideBus(self, bus):
        self.cpu_side = bus.mem_side_ports

    def connectMemSideBus(self, bus):
        self.mem_side = bus.cpu_side_ports

system = System()

system.clk_domain = SrcClockDomain()
system.clk_domain.clock = "800MHz"
system.clk_domain.voltage_domain = VoltageDomain()

system.mem_mode = "timing"
system.mem_ranges = [AddrRange("1024MB")]
system.cpu = RiscvO3CPU()
#system.cpu = RiscvMinorCPU()
system.cpu.isa = RiscvISA(vlen=256, elen=32)

# Create the L1 caches
system.cpu.icache = L1ICache()
system.cpu.dcache = L1DCache()

# Connect the caches to the CPU
system.cpu.icache.connectCPU(system.cpu)
system.cpu.dcache.connectCPU(system.cpu)

# Connect the CPU to the L2 bus
system.l2bus = L2XBar()

# Connect the L1 caches to the L2 bus
system.cpu.icache.connectBus(system.l2bus)
system.cpu.dcache.connectBus(system.l2bus)

# Connect the L2 cache to the CPU side bus
system.l2cache = L2Cache()
system.l2cache.connectCPUSideBus(system.l2bus)

# Connect the L2 cache to the memory bus
system.membus = SystemXBar()
system.l2cache.connectMemSideBus(system.membus)

# Connect the CPU to the memory bus
system.cpu.createInterruptController()

system.mem_ctrl = MemCtrl()
system.mem_ctrl.dram = DDR3_1600_8x8()
system.mem_ctrl.dram.range = system.mem_ranges[0]
system.mem_ctrl.port = system.membus.mem_side_ports

system.system_port = system.membus.cpu_side_ports

thispath = os.path.dirname(os.path.realpath(__file__))
binary = "/vec-fft/FFT"

system.workload = SEWorkload.init_compatible(binary)

process = Process()
process.cmd = [binary]
system.cpu.workload = process
system.cpu.createThreads()
root = Root(full_system=False, system=system)
m5.instantiate()

print(f"Beginning simulation!")
exit_event = m5.simulate()
  

print(f"Exiting @ tick {m5.curTick()} because {exit_event.getCause()}")
```

If possible, please include the Python configuration script used and state clearly any parameters passed.

**Terminal Output**
Beginning simulation!
src/sim/simulate.cc:199: info: Entering event queue @ 0.  Starting simulation...
build/RISCV/arch/riscv/generated/exec-ns.cc.inc:10499: panic: panic condition (pow(2, vlmul) * this->numFields) > 8 occurred: LMUL value is illegal for vlseg inst

**Expected behavior**
Spike simulation goes through without any issues. 

**Host Operating System**
Ubuntu 22.04 on WSL.

**Host ISA**
RISC-V

psharath1, 2024-10-09:
The issue is coming from the below check

    const int64_t vlmul = vtype_vlmul(machInst.vtype8);

    panic_if((pow(2, vlmul) * this->numFields) > 8, "LMUL value is illegal for vlseg inst");

in vseg templates https://github.com/gem5/gem5/blob/stable/src/arch/riscv/isa/templates/vector_mem.isa

I am not sure what is being computed here
(pow(2, vlmul) * this->numFields)

the limitation should be on vlmul not to exceed 8 ?

powerjg, 2024-10-09:
@aarmejach or @saul44203 can you take a look at this?

saul44203, 2024-10-10:
The issue here seems to be that the mentioned vlseg instruction, when decoded, is getting the LMUL from an old vset instruction, and not the immediately preceding one (since it hasn't executed yet), and it so happens to be an invalid one that triggers the assertion (which is correct per spec imo).

So I think there's no direct solution here if the speculative vset behavior stays the same. I suppose we could remove the assertion and let the instruction squash once the new vset is executed, but I don't know if it's a good idea. This is related to #885.

psharath1, 2024-10-10:
@saul44203 does this mean that any LMUL dependent instruction decoded before the preceding vset is executed potentially has this issue? If indeed this is the case it is something that needs to be fixed

5265325, 2024-10-12:
This was due to vtype misprediction. I have fixed it and implemented the other missing vector instructions. I will submit a pull request.

5265325, 2024-10-14:
Fix: #1669

5265325, 2024-10-24:
New fix: #1711

erin-le, 2025-01-13:
It seems like there was some debate about whether PR 1711 fixes this issue or not. Could someone clarify if this issue has been fixed and should be closed?

saul44203, 2025-01-14:
Yes @erin-le, I think it can be closed.

From what I understand, the panic that was triggered on decode due to misprediction is now an illegal instruction fault, which should only halt execution on commit. However, (in this case) this won't happen because the instruction should be squashed before reaching that point.

closed 2025-01-14 by erin-le