gem5/gem5 #1658: Clang fails to compile latest develop branch

pranith, 2024-10-10:
I am seeing the following build errors with clang:

In file included from src/mem/ruby/network/garnet/VirtualChannel.cc:32:
In file included from src/mem/ruby/network/garnet/VirtualChannel.hh:38:
In file included from src/mem/ruby/network/garnet/flitBuffer.hh:39:
In file included from src/mem/ruby/network/garnet/flit.hh:39:
src/mem/ruby/slicc_interface/Message.hh:140:23: error: private field 'p_ruby_system' is not used [-Werror,-Wunused-private-field]
    const RubySystem *p_ruby_system = nullptr;
                      ^
1 error generated.
 [     CXX] src/mem/cache/replacement_policies/random_rp.cc -> ARM/mem/cache/replacement_policies/random_rp.do
 [     CXX] src/arch/arm/htm.cc -> ARM/arch/arm/htm.do
 [     CXX] src/mem/cache/replacement_policies/mru_rp.cc -> ARM/mem/cache/replacement_policies/mru_rp.do
 [     CXX] src/mem/tport.cc -> ARM/mem/tport.do
scons: *** [build/ARM/mem/ruby/network/garnet/VirtualChannel.do] Error 1
src/mem/ruby/common/DataBlock.cc:63:18: error: assigning field to itself [-Werror,-Wself-assign-field]
    m_block_size = m_block_size;
                 ^
1 error generated.

There also seem to be some overloaded virtual function being hidden errors.

pranith, 2024-10-10:
git hash: a8f88abfb12f772f496aad2b9866476f7ebb575c

Clang version:

```
Ubuntu clang version 14.0.0-1ubuntu1.1
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```

powerjg, 2024-10-11:
Fixed by #1650

pranith, 2024-10-11:
Still seeing other errors:

`rc/systemc/ext/tlm_utils/../tlm_core/2/sockets/initiator_socket.hh:127:5: error: 'tlm::tlm_base_initiator_socket<512, tlm::tlm_fw_transport_if<>, tlm::tlm_bw_transport_if<>, 1, sc_core::SC_ONE_OR_MORE_BOUND>::b
ind' hides overloaded virtual functions [-Werror,-Woverloaded-virtual]
    bind(base_type &s)                          
    ^                                                                                                    
src/systemc/ext/tlm_utils/../tlm_core/2/sockets/../../../core/sc_port.hh:135:18: note: hidden overloaded virtual function 'sc_core::sc_port_b<tlm::tlm_fw_transport_if<>>::bind' declared here: type mismatch at 1s
t parameter ('tlm::tlm_fw_transport_if<> &' vs 'tlm::tlm_base_initiator_socket<512, tlm::tlm_fw_transport_if<>, tlm::tlm_bw_transport_if<>, 1, sc_core::SC_ONE_OR_MORE_BOUND>::base_type &' (aka 'tlm_base_initiato
r_socket_b<512U, tlm::tlm_fw_transport_if<>, tlm::tlm_bw_transport_if<>> &'))
    virtual void bind(IF &i) { sc_port_base::bind(i); }
                 ^                                                                                       
src/systemc/ext/tlm_utils/../tlm_core/2/sockets/../../../core/sc_port.hh:136:18: note: hidden overloaded virtual function 'sc_core::sc_port_b<tlm::tlm_fw_transport_if<>>::bind' declared here: type mismatch at 1s
t parameter ('sc_port_b<tlm::tlm_fw_transport_if<>> &' vs 'tlm::tlm_base_initiator_socket<512, tlm::tlm_fw_transport_if<>, tlm::tlm_bw_transport_if<>, 1, sc_core::SC_ONE_OR_MORE_BOUND>::base_type &' (aka 'tlm_ba
se_initiator_socket_b<512U, tlm::tlm_fw_transport_if<>, tlm::tlm_bw_transport_if<>> &'))
    virtual void bind(sc_port_b<IF> &p) { sc_port_base::bind(p); }
                 ^
In file included from src/systemc/tlm_bridge/gem5_to_tlm.cc:59:
In file included from src/systemc/tlm_bridge/gem5_to_tlm.hh:73:
In file included from src/systemc/ext/tlm_utils/simple_initiator_socket.h:27:
src/systemc/ext/tlm_utils/../tlm_core/2/sockets/initiator_socket.hh:141:18: error: 'tlm::tlm_base_initiator_socket<512, tlm::tlm_fw_transport_if<>, tlm::tlm_bw_transport_if<>, 1, sc_core::SC_ONE_OR_MORE_BOUND>::
bind' hides overloaded virtual functions [-Werror,-Woverloaded-virtual]
    virtual void bind(bw_interface_type &ifs) { (get_base_export())(ifs); }
                 ^
`

powerjg, 2024-10-11:
SystemC errors seem unrelated to the recent Ruby changes. Maybe #1637?

pranith, 2024-10-11:
Yes, that needs to be extended to include clang too. PR #1662 is up fixing this.

PR #1662 (2024-10-18): systemc: Disable 'overloaded-virtual' warn for clang
We need to extend the warning disable even for clang compiler.

Fixes #1658
commit ae0cee66 (2024-10-18): systemc: Disable 'overloaded-virtual' warn for clang (#1662)

We need to extend the warning disable even for clang compiler.

Fixes #1658
closed 2024-10-18 by pranith