gem5/gem5 #1668: Incorrect riscv vector unit-stride segment load/store instructions

5265325, 2024-10-12:
**Describe the bug**
The execution result of the riscv vector unit-stride segment load/store instruction is not as expected. The picture of trace here shows the bug. (vlen=128).

The results in trace do not conform to the description in the [specification](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#sec-aos).

[image]

The correct result of vlseg2e8.v(0x106b4) shoule be:
```
vlseg2e8_v_micro v0, (a0), zero : SimdUnitStrideSegmentedLoad :  D=[00020406_080a0c0e_10121416_181a1c1e]
vlseg2e8_v_micro v1, (a0), zero : SimdUnitStrideSegmentedLoad :  D=[01030507_090b0d0f_11131517_191b1d1f]
```

**Affects version**
Develop branch. Commit [a8f88abfb12f772f496aad2b9866476f7ebb575c](https://github.com/gem5/gem5/tree/a8f88abfb12f772f496aad2b9866476f7ebb575c).

**gem5 Modifications**
If you have modified gem5 in some way please state, to the best of your ability, how it has been modified.

**To Reproduce**
[Here](https://github.com/user-attachments/files/17349831/vlseg.tar.gz) is a simple test. Use the `vlseg` elf file and `se.py` script in `gz` package file to simulate. And dump the Exec trace.

**Terminal Output**
If applicable, add the terminal output here. If long, only include the relevant lines.
Please put the terminal output in code blocks. I.e.:

```shell
/data00/home/hongzhibo/share/benchmark/vlseg/vlseg
gem5 Simulator System.  https://www.gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 version DEVELOP-FOR-24.1
gem5 compiled Oct 12 2024 14:28:11
gem5 started Oct 12 2024 18:34:40
gem5 executing on n37-094-005, pid 385420
command line: ./build/RISCV/gem5.opt --debug-flags=Exec --debug-start=45690500 --debug-file=trace.out configs/deprecated/example/se.py --cmd /data00/home/hongzhibo/share/benchmark/vlseg/vlseg

warn: The se.py script is deprecated. It will be removed in future releases of  gem5.
Global frequency set at 1000000000000 ticks per second
src/mem/dram_interface.cc:690: warn: DRAM device capacity (8192 Mbytes) does not match the address range assigned (512 Mbytes)
src/arch/riscv/isa.cc:278: info: RVV enabled, VLEN = 128 bits, ELEN = 64 bits
src/base/statistics.hh:279: warn: One of the stats is a legacy stat. Legacy stat is a stat that does not belong to any statistics::Group. Legacy stat is deprecated.
system.remote_gdb: Listening for connections on port 7000
**** REAL SIMULATION ****
src/sim/simulate.cc:199: info: Entering event queue @ 0.  Starting simulation...
src/sim/syscall_emul.cc:97: warn: ignoring syscall set_robust_list(...)
      (further warnings will be suppressed)
src/sim/mem_state.cc:448: info: Increasing stack size by one page.
src/sim/syscall_emul.cc:86: warn: ignoring syscall mprotect(...)
Exiting @ tick 52690500 because exiting with last active thread context
```

**Expected behavior**
You can see the above picture from the trace file. The results of the vlseg2e8.v instruction do not meet the riscv vector specifications.

**Host Operating System**
Debian 12.

**Host ISA**
X86

**Compiler used**
gcc 14

erin-le, 2025-01-16:
@5265325: Hello, thank you for your contributions! Could you clarify if any of PRs 1709 through 1712 resolve this issue?

5265325, 2025-01-20:
> [@5265325](https://github.com/5265325): Hello, thank you for your contributions! Could you clarify if any of PRs 1709 through 1712 resolve this issue?

No.

closed 2025-06-04 by erin-le