gem5/gem5 #1707: Return address is wrong during Call instruction missprediction

atrah22, 2024-10-24:
It is a bug in _BPredUnit::squash_ function of src/cpu/pred/bpred_unit.cc file.
During correction of RAS due to Call instruction miss-prediction, return_addr is built using corr_target of Call instruction. However, the return_addr should be built with Call instruction's PC (return_addr = Call instruction PC + instruction size).

```
// A call has not pushed a return address to the RAS.
if (hist->call) {
    // In case of a call build the return address and
    // push it to the RAS.
    auto return_addr = hist->inst->buildRetPC(
                                    corr_target, corr_target);

    DPRINTF(Branch, "[tid:%i] [squash sn:%llu] "
            "Incorrectly predicted call: [sn:%llu,PC:%#x] "
            " Push return address %s onto RAS\n", tid,
            squashed_sn, hist->seqNum, hist->pc,
            *return_addr);
    ras->push(tid, *return_addr, hist->rasHistory);
}
```
A quick fix is shown below:
```
// A call has not pushed a return address to the RAS.
if (hist->call) {
    // In case of a call build the return address and
    // push it to the RAS.
    auto return_addr = hist->inst->buildRetPC(
                                    corr_target, corr_target);
    //bug fix
    if (hist->inst->size()) {
            return_addr->set(hist->pc + hist->inst->size());
    }
    DPRINTF(Branch, "[tid:%i] [squash sn:%llu] "
            "Incorrectly predicted call: [sn:%llu,PC:%#x] "
            " Push return address %s onto RAS\n", tid,
            squashed_sn, hist->seqNum, hist->pc,
            *return_addr);
    ras->push(tid, *return_addr, hist->rasHistory);
}
```

Also I suggest to add stats to show, which miss-predictions occurred due to BTB miss and miss-prediction by Predictor. It is important metric to optimize or understand the branch miss-prediction bottleneck specially when requiresBTBHit=true which is normally the case for modern CPU frontend. 
Due to the above mentioned bug, there was high return branch instruction miss-predictions due to Predictor (RAS). For a micro-benchmark, return branch instruction miss-prediction due to Predictor was 4.1k+ whereas return branch instruction miss-prediction due to BTB miss was 4k+. If the RAS size is big enough like 32 entries, the return address miss prediction due to RAS should be minimal (close to 0) for most benchmarks or atleast for micro-benchmarks. After the above mentioned fix, it is close to 0 (24 miss-predictions to be exact instead of 4.1k+).

I added the stats of type Vector2D in bpred_unit.hh. I added the following code in _BPredUnit::squash_ function to get stats related to miss-predictions occurred due to BTB miss and  miss-predictions by Predictor.

```
if (hist->target &&
(hist->target->instAddr() != corr_target.instAddr())) {
    stats.targetWrong[tid][hist->targetProvider]++;
    target_mismatch = true;

}

// stats for identifying miss-prediction due to BTB or predictor
if (hist->btbHit){
    stats.mispredictDueToPredictor[tid][hist->type]++;
}
else{
   stats.mispredictDueToBTBMiss[tid][hist->type]++;
}

```

erin-le, 2024-12-06:
@atrah22 Hello, thank you for documenting this issue. Since you've already done some work to fix it, would you be willing to open a PR with your changes?

atrah22, 2025-03-06:
Thanks for the PR @dhschall.
Actually I did something like this for myself regarding "moving the counting to the point where we commit branches" few months back.
For the time being, I think this PR is good enough to atleast fix the bug.

PR #2063 (2025-03-07): cpu: Fix incorrect return address after flush
An incorrect return address is pushed onto the RAS during misprediction of a call instruction. Currently, the corr_target (which is the target address of the call instruction) is pushed instead of the call PC + instruction size (instruction after the call). This issue was raised in #1707
The commit also adds new statistics to separate mispredictions due to predictor mispredictions and BTB misses.

Fixes #1707
commit 55d22fd2 (2025-03-07): cpu: Fix incorrect return address after flush (#2063)

An incorrect return address is pushed onto the RAS during misprediction
of a call instruction. Currently, the corr_target (which is the target
address of the call instruction) is pushed instead of the call PC +
instruction size (instruction after the call). This issue was raised in
#1707
The commit also adds new statistics to separate mispredictions due to
predictor mispredictions and BTB misses.

Fixes #1707
closed 2025-03-12 by erin-le