gem5/gem5 #1720: Compilation of RISCV/gem5 failing on MacOS with `use of undeclared identifier 'cpu_set_t'` error

BobbyRBruce, 2024-10-27:
**Describe the bug**

Compilation of RISCV/gem5 fails on MacOS fails with `error: use of undeclared identifier 'cpu_set_t'` due to the existance of https://github.com/gem5/gem5/blob/c9f94f4e06391d956b3f43b2f7940bf6cbcc7347/src/arch/riscv/linux/se_workload.cc#L260

This was in introduced by PR #1525.

**Affects version**

Verified to exist on revision c9f94f4e06391d956b3f43b2f7940bf6cbcc7347 (on the develop branch at the time of writing). Introduced with b7b6f5643812b8917a7a902725c727681e367792 via PR #1525.

**To Reproduce**

On a Mac OS system:

```sh
scons build/RISCV/gem5.opt -j`nproc`
```

**Terminal Output**

```console
riscv/generated/decode-method.cc.inc, ALL/arch/riscv/generated/decoder.hh, ALL/arch/riscv/generated/decoder-g.hh.inc, ALL/arch/riscv/generated/decoder-ns.hh.inc, ALL/arch/riscv/generated/exec-g.cc.inc, ALL/arch/riscv/generated/exec-ns.cc.inc, ALL/arch/riscv/generated/decoder.cc, ALL/arch/riscv/generated/inst-constrs.cc, ALL/arch/riscv/generated/generic_cpu_exec.cc
src/arch/riscv/linux/se_workload.cc:260:25: error: use of undeclared identifier 'cpu_set_t'
  260 |             CPU_SET(i, (cpu_set_t *)&cpu_online_mask->bits);
      |                         ^
src/arch/riscv/linux/se_workload.cc:260:36: error: expected expression
  260 |             CPU_SET(i, (cpu_set_t *)&cpu_online_mask->bits);
      |                                    ^
 [VER TAGS]  -> ALL/sim/tags.cc
 [     CXX] ALL/arch/riscv/generated/generic_cpu_exec.cc -> .o
2 errors generated.
scons: *** [build/ALL/arch/riscv/linux/se_workload.o] Error 1
scons: building terminated because of errors.
*** Summary of Warnings ***
Warning: You can get a 12% performance improvement by installing tcmalloc (libgoogle-perftools-dev package
         on Ubuntu or RedHat).
```

**Host Operating System**

MacOS.

**Host ISA**

ARM

**Compiler used**

Clang 19.1.2

MarcoBonino, 2024-11-15:
Any update on this issue ? @powerjg  @ivanaamit  I see you approved / merged the pull request (I'm not able to tag the author of the commit: Zhibo Hong) could you look into this ?

powerjg, 2024-11-15:
This is almost certainly a problem where the `cpu_set_t` is available in a linux header that's not on macos. You should be able to see what we've done with similar types throughout the file and fix this in the same way. I don't have a mac, so I can't test it.

PR #1835 (2024-12-04): arch-riscv: Remove CPU_SET use for non-linux host
For non-Linux systems, we use cpu_set_cpu. CPU_SET is a macro that is not available on all non-Linux systems.

Fixes #1720
commit 2625d238 (2024-12-04): arch-riscv: Remove CPU_SET use for non-linux host (#1835)

For non-Linux systems, we use cpu_set_cpu. CPU_SET is a macro that is
not available for non-Linux systems.

Fixes #1720
commit dee42f18 (2024-12-05): arch-riscv: Remove CPU_SET use for non-linux host (#1835)

For non-Linux systems, we use cpu_set_cpu. CPU_SET is a macro that is
not available for non-Linux systems.

Fixes #1720
closed 2024-12-07 by BobbyRBruce