gem5/gem5 #1722: Build gem5 successfully with no errors, but no .opt file is generated.

ivanaamit, 2024-10-28:
It appears others have encountered this issue, though we haven’t been able to reproduce it ourselves.

#1536 
#1706 

@liuyuxivapor @soniab Please let us know if you're still experiencing this, or if you have found a solution.

If it’s still an issue, could you provide details about your host system? This will help us to create a docker environment and to try to reproduce the issue.

If you have resolved it, sharing your solution would be very helpful for others in the future. Thank you!

JianYugen, 2024-11-02:
Hi. I also recently got this problem when setting up gem5. I'm posting compilation message and my system info here.

Compilation: 
```
scons: Reading SConscript files ...

You're missing the pre-commit/commit-msg hooks. These hook help to ensure your
code follows gem5's style rules on git commit and your commit messages follow
our commit message requirements. This script will now install these hooks in
your .git/hooks/ directory.
Press enter to continue, or ctrl-c to abort:
 Cannot find 'pre-commit'. Please ensure all Python requirements are
installed. This can be done via 'pip install -r requirements.txt'.
It is strongly recommended you install the pre-commit hooks before working with gem5. Do you want to continue compilation (y/n)?
y
Mkdir("/home/jianyugen/gem5/build/X86/gem5.build")
Checking for linker -Wl,--as-needed support... (cached) yes
Checking for compiler -gz support... (cached) yes
Checking for linker -gz support... (cached) yes
Info: Using Python config: python3-config
Checking for C header file Python.h... (cached) yes
Checking Python version... (cached) 3.8.10
Checking for accept(0,0,0) in C++ library None... (cached) yes
Checking for zlibVersion() in C++ library z... (cached) yes
Checking for C library tcmalloc_minimal... (cached) yes
Building in /home/jianyugen/gem5/build/X86
"build_tools/kconfig_base.py" "/home/jianyugen/gem5/build/X86/gem5.build/Kconfig" "/home/jianyugen/gem5/src/Kconfig"
Checking size of struct kvm_xsave ... (cached) yes
Checking for C header file fenv.h... (cached) yes
Checking for C header file png.h... (cached) no
Warning: Header file <png.h> not found.
         This host has no libpng library.
         Disabling support for PNG framebuffers.
Checking for clock_nanosleep(0,0,NULL,NULL) in C library None... (cached) yes
Checking for C header file valgrind/valgrind.h... (cached) yes
Checking for pkg-config package hdf5-serial... (cached) no
Checking for pkg-config package hdf5... (cached) no
Checking for H5Fcreate("", 0, 0, 0) in C library hdf5... (cached) no
Warning: Couldn't find HDF5 C++ libraries. Disabling HDF5 support.
Checking for C header file capstone/capstone.h... (cached) no
Warning: Header file <capstone/capstone.h> not found.
         This host has no capstone library installed.
Checking for C header file linux/kvm.h... (cached) yes
Checking for timer_create(CLOCK_MONOTONIC, NULL, NULL) in C library None... (cached) no
Checking for timer_create(CLOCK_MONOTONIC, NULL, NULL) in C library rt... (cached) yes
Checking for member exclude_host in struct perf_event_attr...(cached) yes
Checking for C header file linux/if_tun.h... (cached) yes
Checking for shm_open("/test", 0, 0) in C library None... (cached) yes
Checking for pkg-config package protobuf... (cached) yes
Checking for backtrace_symbols_fd((void *)1, 0, 0) in C library None... (cached) yes
Checking whether __i386__ is declared... (cached) no
Checking whether __x86_64__ is declared... (cached) yes
Checking for compiler -Wno-self-assign-overloaded support... (cached) yes
Checking for linker -Wno-free-nonheap-object support... (cached) no
scons: done reading SConscript files.
scons: Building targets ...
 [VER TAGS]  -> X86/sim/tags.cc
scons: done building targets.
*** Summary of Warnings ***
Warning: Header file <png.h> not found.
         This host has no libpng library.
         Disabling support for PNG framebuffers.
Warning: Couldn't find HDF5 C++ libraries. Disabling HDF5 support.
Warning: Header file <capstone/capstone.h> not found.
         This host has no capstone library installed.
```

Host system:
```
LSB Version:    core-11.1.0ubuntu2-noarch:security-11.1.0ubuntu2-noarch
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.6 LTS
Release:        20.04
Codename:       focal
```

Hopefully you can figure out how to solve this.

powerjg, 2024-11-02:
@JianYugen what command did you run to compile gem5?

To try to get around this, others have been able to get it to work if they delete the build directory and rebuild from scratch.

liuyuxivapor, 2024-11-02:
@JianYugen I thought i have solved this by enlarging RAM for Ubuntu, and the **link** step will automatically appear.

JianYugen, 2024-11-03:
@powerjg I just followed the steps on https://www.gem5.org/getting_started/ with command `scons build/X86/gem5.opt -j$(nproc)`. And you are right. The problem is solved by cloning the clean source code and rebuilding it again. Probably there was something wrong when installing the dependencies.

@liuyuxivapor I'm also not sure that this will be affected by the RAM size. I built gem5 on a machine with 1T of memory.

soniab, 2024-11-05:
Hi, Sorry for late reply. 

I could not solve it yet. I tried with a clean repository(git clone and gemm5-24.0.tar) but ended up having no .opt generating 

Host system:
LSB Version:	n/a
Distributor ID:	SUSE
Description:	SUSE Linux Enterprise Server 15 SP5
Release:	15.5
Codename:	n/a
Linux version 5.14.21-150500.55.31_13.0.53-cray_shasta_c

Please let me know if you need more details.

giactra, 2024-11-28:
> Description: SUSE Linux Enterprise Server 15 SP5

What happens if you compile with scons --verbose?

iaIm14, 2024-12-18:
I solved this issue by reinstalling the `gcc` dependencies and creating a symbolic link to `gcc-10`. Not sure if others are facing the same cause

liuyuxivapor, 2025-01-12:
@iaIm14 Thanks for your solution but it still refused to work with me :(. I have solved it by reinstalling dependencies.

closed 2025-05-07 by github-actions[bot]

peterhchen, 2025-07-26:
I can build X86, X86_MSI but have problem for ARM_MSI and RISCV. The error messages are same as 
[JianYugen](https://github.com/JianYugen)
[on Nov 2, 2024](https://github.com/gem5/gem5/issues/1722#issuecomment-2452961000)

Below are commands I used:
# Build ARM MESI
scons defconfig build/ARM_MSI build_opts/ARM
scons setconfig build/ARM_MSI RUBY_PROTOCOL_MSI=y SLICC_HTML=y
sudo apt-get install libpng-dev  # png library
sudo apt update
sudo apt install libhdf5-dev hdf5-tools   # Install hdf5 library
pip install capstone   # Install Python capstone library
scons build/ARM_MSI/gem5.opt

peterhchen, 2025-07-26:
Any pre-built binary for gem5/ARM, ARM_MSI, RISCV, ... in docker hub?