gem5/gem5 #1815: Cryptic error when a resource has an unknown category name

franciscod, 2024-11-27:
**Describe the bug**
If a resource has a bad category name, there is a cryptic AssertionError.

**Affects version**
24.0.0.0

**gem5 Modifications**
None

**To Reproduce**
1. Create a `my_resources.json` with the following contents :
```json
[
    {
        "category": "unknown-category",
        "gem5_versions": "24.0",
        "id": "foo",
        "resource_version": "1"
    }
]
 ```

2. Create a `badcategory.py` file with a call to obtain that resource:
```py
from gem5.resources.resource import obtain_resource
obtain_resource("foo")
```
4. Run `GEM5_RESOURCE_JSON_APPEND=my_resources.json gem5 badcategory.py`

**Terminal Output**
```
info: Appending resources from ./my_resources.json
AssertionError: <EMPTY MESSAGE>

At:
  src/python/gem5/resources/resource.py(994): obtain_resource
  badcategory.py(4): <module>
  src/python/m5/main.py(669): main
  ```

**Expected behavior**
Maybe the error could list the valid category names. This could help catching typos.

**Host Operating System**
Ubuntu 24.04.1

**Host ISA**
Unrelated?

**Compiler used**
I don't know, probably not related.

BobbyRBruce, 2024-11-27:
@Harshil2107 : Seems like a simple enough error message to implement?

Harshil2107, 2024-11-27:
It looks like I simple change, I will push a PR for this change,

closed 2025-01-15 by erin-le